fwupd/fwupd #3223: CircleCI Windows build failing

superm1, 2021-05-05:
I looked a little more closely and compared to local runs.  When run on CircleCI with strace here is what happens when you run wine:

```
faccessat2(AT_FDCWD, "/usr/bin/", F_OK, AT_EACCESS) = -1 EPERM (Operation not permitted)
write(2, "wine: ", 6wine: )                   = 6
write(2, "cannot get path to ntdll.so\n", 28cannot get path to ntdll.so
) = 28
exit_group(1)                           = ?
```

Comparing it to local run
```
faccessat2(AT_FDCWD, "/usr/bin", F_OK, AT_EACCESS) = 0
```

hughsie, 2021-05-05:
Ohh, good debugging. Do you think this is because of the trailing slash on `"/usr/bin/"` or something else entirely?

superm1, 2021-05-05:
I don't think it's a trailing slash thing, it feels more like a permissions thing, maybe running docker without --privileged locally might repro it too.

superm1, 2021-05-05:
The trailing slash was a mistake when I transcribed the error from local run.  It's there in the local one too.

superm1, 2021-05-05:
Tried using a local run without `--privileged` and still don't hit it there.

Circle lets you connect to SSH on a failed run.  Maybe you can poke around and see if you can come up with the reason it's happening.

superm1, 2021-05-05:
So this is fixed with reverting back to Fedora 33 instead of "latest" (34).

@hughsie can you talk to some fedora peeps about this failure?  It would be nice to leave it at `fedora:latest` so we're always picking up new deps and we don't have to worry about 33's impending EOL dates.

hughsie, 2021-05-05:
> hughsie can you talk to some fedora peeps about this failure

I'll poke around the differences myself and then file a bug downstream. Thanks dude.

bgilbert, 2021-05-28:
See e.g. [RHBZ 1900021](https://bugzilla.redhat.com/show_bug.cgi?id=1900021).  glibc 2.33 opportunistically implements `faccessat()` using `faccessat2()`, and falls back to compatibility code if it gets `-ENOSYS`.  But older versions of several container runtimes install seccomp filters that return `-EPERM` on unrecognized system calls.

hughsie, 2021-06-04:
Super useful, thanks @bgilbert!

closed 2021-07-15 by stale[bot]