fwupd/fwupd #3228: UEFI Update: (null)/fwupdx64.efi cannot be found

parmort, 2021-05-06:
**Describe the bug**
When running on an XPS 13 9305 with Arch Linux, upgrading the "System Firmware" from 1.0.1 to 1.0.5 failed with the error:
```
(null)/fwupdx64.efi cannot be found
```

This happened after the second "Authenticating..." bar fully completed.

**Steps to Reproduce**
I used `fwupdmgr get-devices` to list the devices, installed udisks2, and rebooted. I then ran `fwupdmgr refresh` and then `fwupdmgr update`. It installed a firmware for my NVME SSD fine, but then failed with the system firmware.

**Expected behavior**
I expected the application to update the system firmware.

**fwupd version information**
```shell
client version:	1.6.0
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.6.0
```

Installed by pacman from the community repository.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
XPS 13 9305
│
├─11th Gen Intel(R) Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000068
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba
│                         1df99df5-22f4-5505-98cc-22dcc53b132a
│     Device Flags:       • Internal device
│   
├─KBG40ZNS256G NVMe KIOXIA 256GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express Solid State Drive
│     Current version:    10410106
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     GUIDs:              35de42d4-bd35-5bcd-89fa-6e71cbd4d0ed
│                         5264a089-86d6-e811-9f8b-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          85469c3b8d4e10990242945a1d1f15f9bed695bf
│ │   Current version:    1.0.1
│ │   Minimum Version:    1.0.1
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Failed
│ │   Update Error:       (null)/fwupdx64.efi cannot be found
│ │   Last modified:      2021-05-06 20:11
│ │   GUIDs:              3b159f6e-2c39-458b-9928-32aa3c68a0ab
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       c27f0b52-d1a3-5326-8f21-cdad1777e15e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
[... 439 lines cut ...]
│       ├── videotest.mod
│       ├── wrmsr.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux

9 directories, 355 files
```

**Additional questions**
- **Operating system and version:** Arch Linux (`5.12.1-arch1-1`)
- **Have you tried rebooting?** Yes, a few times at various stages throughout the process.
- **Is this a regression?** No
- **Are you using an NVMe disk?** Yes
- **Is secure boot enabled?** No
- **Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?** No

hughsie, 2021-05-06:
Does this help? https://github.com/fwupd/fwupd/commit/197d3dc3b9acb93be30c4a7f49e8c0ad3f554818 -- if that's indeed the ix then I should probably tag a 1.6.1 fairly soon...

parmort, 2021-05-06:
Yes, that fixed the issue. At least I assume it was that commit--I just switched to the `fwupd-git` package on the AUR.

closed 2021-05-06 by parmort

hughsie, 2021-05-07:
I'll do a 1.6.1 in a week or so I guess. Thanks for checking.

uPvtkBlpiucvDUf, 2021-05-25:
I'm on Gentoo and had the same error message as @parmort . But after manually applying the patch from <https://github.com/fwupd/fwupd/commit/197d3dc3b9acb93be30c4a7f49e8c0ad3f554818> I now just get a different error message:
```
Upgrade available for System Firmware from 0.1.49 to 0.1.50
20HR002MGE must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
/usr/libexec/fwupd/efi/fwupdx64.efi cannot be found
```

There is no fwupdx64.efi installed anywhere. So maybe this is a Gentoo packaging bug then?

superm1, 2021-05-25:
There is a separate package starting with fwupd 1.6.0 where fwupd-efi comes separately.

For compatibility purposes it can be installed via subproject as well, but that's a packaging maintainer decision. I would check with your packaging maintainer on this.