fwupd/fwupd #3231: Update is not getting applied (yet again)

LinuxOnTheDesktop, 2021-05-07:
**Describe the bug**
As has happened to me and others before, an update is not getting applied.

**Steps to Reproduce**

Try to install this update:

• UEFI Device Firmware has the latest available firmware version
Upgrade available for UEFI Device Firmware from 0.71.12 to 0.73.20

My PC: Lenovo ThinkPad X1CG6.

**Expected behavior**
If multiple reboots-to-desktop are required then the user should be told about this. If the situation is a bug then the bug should be fixed.

**fwupd version information**

```shell
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20KH001UKP
│
├─Thunderbolt Controller:
│     Device ID:           9e4ed2585a7c7ad2de1e4593bb8b631301584cea
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.8.8341
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               963522d7-bc73-51fa-a70c-cf615f42bcf6
│                          b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                          ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.50
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
[... 784 lines cut ...]
│       └── zfs.mod
├── initrd.img -> initrd.img-5.4.0-72-generic
├── initrd.img-5.4.0-72-generic
├── initrd.img-5.8.0-49-generic
├── initrd.img-5.8.0-50-generic
├── initrd.img.old -> initrd.img-5.8.0-50-generic
├── System.map-5.4.0-72-generic
├── System.map-5.8.0-49-generic
├── System.map-5.8.0-50-generic
├── vmlinuz -> vmlinuz-5.4.0-72-generic
├── vmlinuz-5.4.0-72-generic
├── vmlinuz-5.8.0-49-generic
├── vmlinuz-5.8.0-50-generic
└── vmlinuz.old -> vmlinuz-5.8.0-50-generic

21 directories, 653 files
```

**Additional questions**
- Operating system and version: Mint 20.1 Cinnamon x64
- Have you tried rebooting? Several times. Originally there were two updates to be installed. Now there is only one. Perhaps at some stage the remaining update will actually get installed. Just another, say, six reboots to go!
- Is this a regression? I don't know.
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I don't know. I will check.

LinuxOnTheDesktop, 2021-05-08:
Perhaps. [The last time I tried to use the flatpak I had trouble](https://github.com/fwupd/fwupd/issues/2261). But I see from the page I linked that people have made suggestions that I could try.

LinuxOnTheDesktop, 2021-05-09:
I add that the problem (the problem of updates not being applied, not the problem that the Flatpak will run at all) seems to be the following. I reach the stage where the 'capsule' is to be installed, and I see the progress bar - but only for a fraction of a second; for, at that point, the machine restarts.

I note also that when I tried something that I thought might help, I received what seems a fairly incomprehensible error message:

```
$ fwupdmgr clear-offline 
failed to execute prepared statement: attempt to write a readonly database
```

LinuxOnTheDesktop, 2021-05-09:
Perhaps the main problem is (as [this other report](https://github.com/fwupd/firmware-lenovo-thinkpad/issues/56) on your tracker suggests) that I have TPM disabled (for, I think I do). If so: (1) the output from (the snap version of) fwupd misleads; (2) fwupd should never offer me the update in the first place.

LinuxOnTheDesktop, 2021-05-09:
I closed this issue by accident. But I have at least managed to escape from all this stuff. For, I discovered that TPM was *enabled* on my machine. So I _disabled_ it in the BIOS and now I no longer see those uninstallable updates. Obviously though this is not a fix. So I leave the issue open.

LinuxOnTheDesktop, 2021-05-10:
Installing a different distro on the laptop in question is not something I am prepared to do. I am prepared to boot without my kernel switches but I anticipate that so doing would make no difference.

hughsie, 2021-05-10:
The thing is, nobody is testing on Linux Mint. It's not in CI, and the distro package looks woefully out of date. I think updating the packaged version of fwupd (possibly syncing from Ubuntu/Debian) might be the quickest way to rule out flatpak-specific interactions.

LinuxOnTheDesktop, 2021-05-10:
1) _Updating the package version of `fwupd`_: sounds like a good idea, but you cannot expect me, a mere user, to make that so.

2) '_[T]he quickest way to rule out flatpak-specific interactions'_: I am unsure what you mean. At any rate, henceforth I will touch the flatpak and indeed the snap versions only if I have to. For, each of them - including the snap - caused problems for the apt version when I uninstalled it (and [the flatpak was a disaster](https://github.com/fwupd/fwupd/issues/226), at least given the seeming lack of instructions).

3) _Perhaps you **should** test your stuff on Mint_. For, it's a popular distro. Note also that I might not be the only user who has had, or will have, the following not-at-all-good experience.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.1) Lenovo updates fail to install (and as I've said elsewhere, this problem has occurred with _multiple_ updates).

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.2) At that point, fwupd maintainers advise installing the fwupd flatpak but - see 2 - that goes wrong too.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;3.3) The user tries the Snap version (which is rather hidden away and snaps are by default disallowed on Mint), which may or may not solve the original problem. In my case it did not. And (cf. 2) it caused its own problems.

closed 2021-05-10 by hughsie

LinuxOnTheDesktop, 2021-05-10:
I understand. Yet, I am doing my best to help you to squash the bug and I too have limited resources and for the reasons I have given it has been a grim experience for me.