fwupd/fwupd #3236: Desktop detected as being on battery power

ReillyBrogan, 2021-05-09:
**Describe the bug**
I just got 1.6.0 compiling on my system (Solus linux) and I am seeing that most updates will not install with the following message:
```
Update Error:       Cannot install update when system battery is not at least 10%
```
The only problem is that this is a desktop and lacks any sort of internal battery. 

Relevant log message from starting fwupd directly from the shell:
```
08:42:45:0469 FuPlugin             startup(upower)
08:42:45:0471 FuContext            battery threshold now 10
08:42:45:0471 FuContext            battery level now 0
```

Taking a look at the code I see that the uplower plugin [queries](https://github.com/fwupd/fwupd/blob/ba890cde4621aca2b8d52a5862bcaea4b390a921/plugins/upower/fu-plugin-upower.c#L42) for the `Type`, `State`, and `Percentage` properties. Manually querying those on my system show that all three of them return `0` (which appears to be the correct value according to the upower [api](https://upower.freedesktop.org/docs/Device.html#Device:Type)). The code however seems to assume that it's on battery unless at least one of those is null. As far as I can tell the correct approach is only assume we are on battery if the `Type` property returns `2` (for `Battery`) or `3` (for `Ups`). 

**Steps to Reproduce**
Started fwupd after compiling and installing it.

**Expected behavior**
That these updates would be installable.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
client version:	1.6.0
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.6.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Custom built package using a Solus-specific format. 

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
├─CT1000P1SSD8:
│     Device ID:          xxxx
│     Summary:            NVM Express Solid State Drive
│     Current version:    P3CR010
│     Vendor:             Micron/Crucial Technology (NVME:0xC0A9)
│     Update Error:       Cannot install update when system battery is not at least 10%
│     GUIDs:              xxx
│                         xxx
│                         xxx
│                         xxx
│                         xxx
│     Device Flags:       • System requires external power source
│                         • Needs a reboot after installation
│                         • Updatable

```

**Additional questions**
- Operating system and version: Solus Linux 
- Have you tried rebooting? Yes
- Is this a regression? Probably. I have only tested building 1.6.0.

ReillyBrogan, 2021-05-09:
OK, I tested it with this incredibly hacky patch just to verify my hypothesis that `Type`, `State`, and `Percentage` all being `0` was breaking it and it now works as expected:

```diff
diff --git a/plugins/upower/fu-plugin-upower.c b/plugins/upower/fu-plugin-upower.c
index 79525c03..95e34296 100644
--- a/plugins/upower/fu-plugin-upower.c
+++ b/plugins/upower/fu-plugin-upower.c
@@ -62,7 +62,13 @@ fu_plugin_upower_rescan (FuPlugin *plugin)
 		fu_context_set_battery_level (ctx, FU_BATTERY_VALUE_INVALID);
 		return;
 	}
-	fu_context_set_battery_level (ctx, g_variant_get_double (percentage_val));
+
+	if (g_variant_get_uint32 (type_val) == 0 && g_variant_get_uint32 (state_val) == 0 && g_variant_get_double (percentage_val) == 0) {
+		fu_context_set_battery_level (ctx, 100);
+	}
+	else {
+		fu_context_set_battery_level (ctx, g_variant_get_double (percentage_val));
+	}
 }
 
 static void
```

hughsie, 2021-05-10:
@ReillyBrogan can you check with the patch in https://github.com/fwupd/fwupd/pull/3240 please

ReillyBrogan, 2021-05-11:
> @ReillyBrogan can you check with the patch in #3240 please

@hughsie I can confirm that that patch fixes my issue. Thanks!

ReillyBrogan, 2021-05-11:
@hughsie Though now that I look at that patch closer (and I'm just spitballin' here, you have far more domain knowledge than I do) wouldn't a battery percentage of `0` be a valid value that could be returned if a system did indeed have a batter? 

I'm imagining a scenario where a laptop has the battery fully discharged yet is plugged in to charge. In this scenario we would have a type of `2` for `Battery`, a state of `1` for `Charging`, and (probably unlikely unless it was rounding or erroneous) a percentage `0`. 

I suspect that having only the first two checks on `Type` and `State` are sufficient and the latter `Percentage` check can be dropped. Especially given that the API has this to say about the `Percentage` property:
```
 This property is only valid if the property type has the value "battery". 
 ```
 
 In fact, and perhaps there's something I'm not aware of, is there any point in continueing past the first check on `Type` if it's anything other than `2` (aka `Battery`)?