fwupd/fwupd #3238: UEFI capsule updates not available or enabled in firmware setup

developer91234, 2021-05-10:
I'm pretty sure my system supports this feature, it's a t440s laptop. I've updated the BIOS before from windows and I don't remember seeing this warning last year when using another distro.

Is there anything about my installation that could be causing this? My efi paritition is flagged as efi and mounted to /efi. Could this setup cause the problem?

superm1, 2021-05-10:
start out by filing a bug that follows the template.

developer91234, 2021-05-10:
updated

superm1, 2021-05-10:
Can you check the contents of /sys/firmware/efi/esrt? Does it have entries? And if so what are they?

developer91234, 2021-05-10:
There's no such folder. I only have ./efivars and ./runtimemap

superm1, 2021-05-11:
If you don't have this then it means one of three things:
1. It's turned off in your BIOS setup.
2. It's not supported by this model
3. Your kernel has `EFI_ESRT` turned off.

closed 2021-05-20 by superm1

developer91234, 2021-05-23:
@superm1 how do you determine if EFI_ESRT is turned off?

superm1, 2021-05-23:
Check your kernel configuration for that config option.

developer91234, 2021-05-23:
Alright that was easy, it;s turned on for the running kernel:

```
# CONFIG_EFI_VARS is not set
CONFIG_EFI_ESRT=y
CONFIG_EFI_VARS_PSTORE=y
CONFIG_EFI_VARS_PSTORE_DEFAULT_DISABLE=y
CONFIG_EFI_RUNTIME_MAP=y
# CONFIG_EFI_FAKE_MEMMAP is not set
CONFIG_EFI_SOFT_RESERVE=y
CONFIG_EFI_RUNTIME_WRAPPERS=y
CONFIG_EFI_GENERIC_STUB_INITRD_CMDLINE_LOADER=y
CONFIG_EFI_BOOTLOADER_CONTROL=m
CONFIG_EFI_CAPSULE_LOADER=m
# CONFIG_EFI_TEST is not set
CONFIG_APPLE_PROPERTIES=y
# CONFIG_RESET_ATTACK_MITIGATION is not set
CONFIG_EFI_RCI2_TABLE=y
# CONFIG_EFI_DISABLE_PCI_DMA is not set
# end of EFI (Extensible Firmware Interface) Support

```
```
CONFIG_EFI_EMBEDDED_FIRMWARE=y
CONFIG_UEFI_CPER=y
CONFIG_UEFI_CPER_X86=y
CONFIG_EFI_DEV_PATH_PARSER=y
CONFIG_EFI_EARLYCON=y
CONFIG_EFI_CUSTOM_SSDT_OVERLAYS=y
```
Still hard to believe that a Thinkpad t440 doesn't support this because after all, you can update its Bios through windows. I remember doing it a couple years ago. Anything else I could check?

superm1, 2021-05-23:
Updating through windows isn't the same as capsule updates. It is most likely a proprietary vendor interface used in Windows.

developer91234, 2021-05-23:
Ah I see.

petrkutalek, 2021-06-26:
Just FYI: Lenovo T490, there was no `/sys/firmware/efi/esrt` folder available until I enabled `Security > Windows UEFI Firmware Update` option. Then after reboot everything works all right.

hughsie, 2021-06-28:
@superm1 can we use the thinklmi thing to detect `Windows UEFI Firmware Update` being off?

superm1, 2021-06-28:
Yep that's definitely a way that we can go if we want.