fwupd/fwupd #3254: Firmware Capsule Is Not Getting Loaded on a Thinkpad T480

kendoori, 2021-05-16:
**Describe the bug**
A clear and concise description of what the bug is.

When I run fwupdmg I am told: "Upgrade available for UEFI Device Firmware." I then apply the update. When I reboot, it looks like the capsule is being loaded. I see this screen, and then it goes to the normal login screen

I keep getting prompted to reapply this update.

**Steps to Reproduce**
Steps to reproduce the behavior.

sudo fwupdmgr update

`Upgrade available for UEFI Device Firmware from 184.79.3722 to 184.83.3874
20L50011US must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]`

[image]

If I respond Yes and reboot, the screen displays "found update, adding new capsule and then boots into the system

[image]

**Expected behavior**
A clear and concise description of what you expected to happen.

After applying update, the same update shouldn't be presented again.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
`client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
`

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
`fwupdmgr get-devices --show-all-devices
20L50011US
│
├─Thunderbolt Controller:
│     Device ID:           36cf863540d97a0506a091d24cf4f1c5d848183e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Intel AMT [unprovisioned]:
│     Device ID:           e2623122c99d58220498aacbfcfdb1baebbae3c5
│     Summary:             Hardware and firmware technology for remote out-of-band management
│     Current version:     11.8.83.3874
│     Bootloader Version:  11.8.83.3874
│     Vendor:              Intel Corporation
│     GUID:                2800f812-b7b4-2d4b-aca8-46e0ff65814c
[... 464 lines cut ...]
├── initrd.img-5.8.0-53-lowlatency
├── initrd.img.old -> initrd.img-5.4.0-73-generic
├── System.map-5.4.0-26-generic
├── System.map-5.4.0-73-generic
├── System.map-5.4.0-73-lowlatency
├── System.map-5.8.0-50-generic
├── System.map-5.8.0-50-lowlatency
├── System.map-5.8.0-53-lowlatency
├── vmlinuz -> vmlinuz-5.8.0-53-lowlatency
├── vmlinuz-5.4.0-26-generic
├── vmlinuz-5.4.0-73-generic
├── vmlinuz-5.4.0-73-lowlatency
├── vmlinuz-5.8.0-50-generic
├── vmlinuz-5.8.0-50-lowlatency
├── vmlinuz-5.8.0-53-lowlatency
└── vmlinuz.old -> vmlinuz-5.4.0-73-generic
`

**Additional questions**
- Operating system and version: Mint 20.1 Cinnamon
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? No
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

hughsie, 2021-05-17:
If you do that again, then when it boots up into Linux without applying the update what does `fwupdmgr get-devices` say? e.g. you should get an "Update Error" that gives us some hint.

kendoori, 2021-05-17:
I am not seeing any errors.

`kenny@kenny-ThinkPad-T480:~$ fwupdmgr get-devices
20L50011US
│
├─Thunderbolt Controller:
│     Device ID:           36cf863540d97a0506a091d24cf4f1c5d848183e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               7d131734-64ed-5865-a46e-a67cfb9b5a46
│                          e45313fd-8d5c-54e9-b175-1f15ffe6b05c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─SanDisk SDSSDXPS960G:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:             ATA Drive
│     Current version:     X212T0RL
│     GUIDs:               22be2cc6-990f-583a-98d3-f89e332bcb93
│                          bc497ac3-efa7-5d3a-98ca-7777bb59fad1
│                          a511a2d6-df8d-5291-90cc-49dc6ab6fe76
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           48477ba14bd9bd0c97323a07caa54b7766f917c2
│     Current version:     0.1.36
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           5decfff65e3391259bca4003a090e59bf50e3bea
│     Current version:     184.79.3722
│     Minimum Version:     184.79.3722
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                dd31d983-cf3b-4c84-acc9-70caa94e827b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:     0.1.21
│     Minimum Version:     0.1.21
│     Vendor:              DMI:LENOVO
│     GUID:                b87a926d-189c-49a1-b1ff-921099b1de89
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b1825e2b03a9ebdea247a74f1f3ddee2a21ac565
│     Current version:     1.3.1084
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                24aac206-3921-45c0-a79e-0a607b11aef0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           a567cff02396e1038d53d7433ef3dbcd9558aebe
      Current version:     1.1.5
      Minimum Version:     0.0.1
      Vendor:              DMI:LENOVO
      GUID:                d4583ddb-bf0a-4801-aefd-951930545cb9
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
    

`

hughsie, 2021-05-17:
> client version: 1.3.11

Ohh wow, that's kinda old. Can you try with something newer like 1.5.x or 1.6.0?

kendoori, 2021-05-17:
I'm on Mint so 1.3.11 is what they provide. I don't have Docker installed to compile for the instructions on the Wiki. Do I need to go through that process, or is there another way to get the newer version (Snap, Flatpack?) and if I install one of those, do I need to uninstall the Mint 1.3.11 version in order to run the update from the CLI. Alternatively, should I just boot from a LiveUSB of another distro (e.g. Manjaro) and try to update there?

hughsie, 2021-05-17:
Maybe try a Fedora 34 LiveUSB image?

kendoori, 2021-05-17:
Will do and I will report back.

superm1, 2021-05-17:
Will live USB image actually work with the existing ESP?  I guess I've never considered to try this.  Can you, please capture full `efibootmgr -v` output before the reboot to try to apply the update?  I'd like to make sure it made a new boot entry and used the correct binary when it tried.

kendoori, 2021-05-17:
The Fedora update didn't work, or I didn't do it right. I created a Live USB, started the daemon and did the update. I then rebooted first into Fedora (and repeated the process, whereby it offered the update again). I then rebooted back into Mint and was also presented with the update option again.

As requested, this is from Fedora, after the update, but before rebooting:
`An update requires a reboot to complete. Restart now? [y|N]: N
[liveuser@localhost-live ~]$ efibootmgr -v
BootNext: 0001
BootCurrent: 001C
Timeout: 0 seconds
BootOrder: 0017,0019,001A,0018,0002,001D,001E,001F,001B,0000,0004,001C,0024,0001
Boot0000  Windows Boot Manager	HD(1,GPT,cb2f4f3a-6d5d-41d7-bc0b-6a6b0c7dc6b2,0x1000,0xf8fff)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Linux-Firmware-Updater	HD(1,GPT,0d102bc9-bb4e-4d57-8c08-f83b29468410,0x800,0x100000)/File(\EFI\fedora\fwupdx64.efi)
Boot0002* ubuntu	HD(1,GPT,0d102bc9-bb4e-4d57-8c08-f83b29468410,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0004  ubuntu	HD(1,GPT,0d102bc9-bb4e-4d57-8c08-f83b29468410,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019  NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001A  NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B  ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
[liveuser@localhost-live ~]$ sudo fwupdmgr update
`

superm1, 2021-05-18:
```Boot0001* Linux-Firmware-Updater	HD(1,GPT,0d102bc9-bb4e-4d57-8c08-f83b29468410,0x800,0x100000)/File(\EFI\fedora\fwupdx64.efi)
```
After it prompts, let it boot into the Boot option `Boot0002` (`Linux-Firmware-Updater`) which looks like it built to match something looking nominal.

kendoori, 2021-05-18:
So you are saying, boot to Live USB (Fedora), do the update within Fedora,
and then when it prompts to reboot, choose the boot option for the firmware
updater?

In my experience on Thinkpads, I might need to interrupt the boot to choose
the boot option (e.g, F12), at which point one can choose Linux
Fimrware Updater. So would I do that?

superm1, 2021-05-18:
Well it's set as bootnext so I think it should pick on it's own

kendoori, 2021-05-18:
I applied the update in Fedora and rebooted, and let it just go on its own without stopping the sequence from the keyboard. It took a while to boot, but on its own it booted into Mint. When I ran `fwupdmgr update` it offered the update again.

superm1, 2021-05-18:
Seems like it's firmware then