fwupd/fwupd #3255: efivars mounted but "Required efivarfs filesystem was not found"

eoli3n, 2021-05-16:
**Describe the bug**
When i run ``fwupdmgr get-devices`` I get ``Required efivarfs filesystem was not found`` but mounting it manually doesn't solve the warning.

**Steps to Reproduce**
```
$ mount -t efivarfs efivarfs /sys/firmware/efi/efivars
$ fwupdmgr get-devices
# cry
```

**Expected behavior**
fwupdmgr to find efivars

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): xbps

**fwupd device information**
```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: Required efivarfs filesystem was not found
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:efivar-not-mounted for more information.
Latitude 7490
│
├─Dell WD15:
│ │   Device ID:          3086ec033384094e82dff9652509f61e12f66da7
│ │   Résumé:             A USB type-C docking station
│ │   Version actuelle:   0.0.0
│ │   Fournisseur:        Dell Inc. (PCI:0x1028)
│ │   Update Error:       Not updatable as efivarfs was not found
│ │   GUID:               e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:       • System requires external power source
│ │                       • Needs a reboot after installation
│ │ 
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:        09821649fd4c49c71adcaadc9f779db402b46cb2
│ │     Résumé:           A USB type-C docking station
│ │     Version actuelle: 0.3.18
│ │     Fournisseur:      Dell Inc. (PCI:0x1028)
│ │     Update Error:     Not updatable as efivarfs was not found
│ │     GUID:             fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:     • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ ├─Dell WD15 Port Controller 1:
│ │     Device ID:        5da6f3f1074a5fd9e04d426b49b301310515e34f
│ │     Résumé:           A USB type-C docking station
│ │     Version actuelle: 1.1.8
│ │     Fournisseur:      Dell Inc. (PCI:0x1028)
│ │     Update Error:     Not updatable as efivarfs was not found
│ │     GUID:             8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│ │     Device Flags:     • System requires external power source
│ │                       • Needs a reboot after installation
│ │   
│ └─VMM3332 inside Dell WD15/TB16/TB18 wired Dock:
│       Device ID:        d762543f8c20f636e6fff031a000078d3e10c600
│       Résumé:           Multi-Stream Transport Device
│       Version actuelle: 3.12.02
│       Fournisseur:      Synaptics (DRM_DP_AUX_DEV:0x06CB)
│       GUIDs:            87f8cd71-1e6b-5622-94e9-0e9d11bdde33 ← MST-274
│                         c1e81f96-234e-597d-b9cb-6477d0652868 ← MST-wd15-274
│                         78c73924-821a-5e6f-af82-86df6f67b182 ← MST-wd15-vmm3332-274
[... 253 lines cut ...]
```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-a86a3f07-b7d3-4545-9238-0274fc1ba682-0
```

```shell
$ tree /boot
/boot
├── config-5.11.18_1
├── config-5.11.21_1
├── EFI
│   └── gummiboot
├── initramfs-5.11.18_1.img
├── initramfs-5.11.21_1.img
├── vmlinuz-5.11.18_1
└── vmlinuz-5.11.21_1
```

**Additional questions**
- Operating system and version: latest voidlinux
- Have you tried rebooting? yep
- Is this a regression? nop
- Are you using an NVMe disk? yep TOSHIBA_KSG60ZMV256G_M.2_2280_256GB
- Is secure boot enabled? nop
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? nop

hughsie, 2021-05-16:
> $ mount -t efivarfs efivarfs /sys/firmware/efi/efivars
> $ fwupdmgr get-devices

Maybe try restarting the fwupd service before the get-devices?

eoli3n, 2021-05-16:
it seems that it made ``fwupd`` not build properly, because of a test failing
https://github.com/void-linux/void-packages/blob/master/srcpkgs/fwupd/template#L54

``efivar`` was not installed before I wrote that issue, I noticed the miss when running the command you ask in issue template.

The template doesn't contains any service, but i can find "/usr/libexec/fwupd/fwupd" running.
Killing it manually and starting it again solved the problem.

Thanks

closed 2021-05-16 by eoli3n