fwupd/fwupd #3270: Compiling with test support fails

djcampello, 2021-05-20:
data/tests/missing-hwid/meson.build:1:0: ERROR: Unknown variable "gcab".

Adding to data/tests/colorhug/meson.build the following (extracted from meson.build) resolves the issue:
gcab = find_program('gcab', required : get_option('tests'))

The Meson build system
Version: 0.55.3

Probably unrelated but also present in the logs (regardless of test feature):
WARNING: Project targeting '>=0.47.0' but tried to use feature introduced in '0.50.0': install arg in configure_file.

superm1, 2021-05-20:
>Adding to data/tests/colorhug/meson.build the following (extracted from meson.build) resolves the issue:
gcab = find_program('gcab', required : get_option('tests'))

That's a bit confusing that this actually helps.  By it being in top level meson.build, it's supposed to be used elsewhere too.  Can you provide the full configure flags used to cause it so we can repro it?

>WARNING: Project targeting '>=0.47.0' but tried to use feature introduced in '0.50.0': install arg in configure_file.

This is expected; we intentionally check which version is in use and only turn those features on newer versions.

djcampello, 2021-05-20:
Hope the following works:

sys-apps/fwupd-9999::chromiumos was built with the following:
USE="abi_x86_64 bluetooth elibc_glibc flashrom gnutls gusb kernel_linux minimal nvme pkcs7 python_single_target_python3_6 synaptics test thunderbolt userland_GNU -agent -amt -archive -cros_host -cros_workon_tree_ -dell -elogind -gpg -gtk-doc -introspection -man -nls -policykit -python_single_target_python2_7 -python_single_target_python3_7 -systemd -uefi"

https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/master/sys-apps/fwupd/fwupd-9999.ebuild

hughsie, 2021-05-20:
>  By it being in top level meson.build, it's supposed to be used elsewhere too

That was also my conclusion. Weird!

superm1, 2021-05-20:
> Hope the following works:
>
> sys-apps/fwupd-9999::chromiumos was built with the following:
> USE="abi_x86_64 bluetooth elibc_glibc flashrom gnutls gusb kernel_linux minimal nvme pkcs7 [...]

Can you show the meson command that is calculated by those use flags? Hopefully it shows up in a lot or you can add a print statement for it directly.

djcampello, 2021-05-20:
meson --buildtype plain --libdir lib64 --localstatedir /var/lib --prefix /usr --sysconfdir /etc --wrap-mode nodownload --cross-file /build/puff-borealis/tmp/portage/sys-apps/fwupd-9999/temp/meson.x86_64-cros-linux-gnu.amd64 --localstatedir /var -Dbuild=standalone -Dagent=false -Dplugin_amt=false -Dlibarchive=false -Dbluez=true -Dplugin_dell=false -Delogind=false -Dplugin_flashrom=true -Dgnutls=true -Dgtkdoc=false -Dgusb=true -Dplugin_altos=true -Dman=false -Dplugin_nvme=true -Dintrospection=false -Dpolkit=false -Dplugin_synaptics_mst=true -Dplugin_synaptics_rmi=true -Dsystemd=false -Dplugin_dummy=true -Dtests=true -Dplugin_thunderbolt=true -Dplugin_uefi_capsule=false -Dplugin_uefi_pk=false -Dconsolekit=false -Dcurl=true -Dplugin_modem_manager=false -Dplugin_tpm=false /build/puff-borealis/tmp/portage/sys-apps/fwupd-9999/work/fwupd-9999 /build/puff-borealis/tmp/portage/sys-apps/fwupd-9999/work/fwupd-9999-build

hughsie, 2021-05-20:
Hmm, so i tried almost exactly that:

    mkdir build-cros && cd build-cros
    meson --buildtype plain --libdir lib64 --localstatedir /var/lib --prefix /usr --sysconfdir /etc --wrap-mode nodownload --localstatedir /var -Dbuild=standalone -Dagent=false -Dplugin_amt=false -Dlibarchive=false -Dbluez=true -Dplugin_dell=false -Delogind=false -Dplugin_flashrom=true -Dgnutls=true -Dgtkdoc=false -Dgusb=true -Dplugin_altos=true -Dman=false -Dplugin_nvme=true -Dintrospection=false -Dpolkit=false -Dplugin_synaptics_mst=true -Dplugin_synaptics_rmi=true -Dsystemd=false -Dplugin_dummy=true -Dtests=true -Dplugin_thunderbolt=true -Dplugin_uefi_capsule=false -Dplugin_uefi_pk=false -Dconsolekit=false -Dcurl=true -Dplugin_modem_manager=false -Dplugin_tpm=false ../

i.e. just removing the --cross-file and that seemed to work. What branch/commit are you on?

superm1, 2021-05-20:
Can you turn off the cross compiling on your side and see if it helps?  If it does, it sounds like a subtle meson bug related to it that probably is asking for a very small reproducer to submit upstream.

djcampello, 2021-05-20:
Sadly I cannot. I am compiling inside the SDK for a target chrome board. Trying to compile fwupd for the SDK itself is not currently supported as not all the portage dependencies are satisfied

superm1, 2021-05-21:
Tried with both your version of meson from pypi and the latest too:
```
$ meson --version
0.55.3
$ meson --version
0.58.0
```

djcampello, 2021-05-24:
I tried with the following file:

```
[binaries]
c = 'gcc'
cpp = 'cpp'
ar = 'ar'
strip = 'strip'
pkgconfig = '/build/<board>/build/bin/pkg-config'

[host_machine]
system = 'linux'
cpu_family = 'x86_64'
cpu = 'x86_64'
endian = 'little'
```

Contents of the script for pkg-config in Chrome OS (replacing <board> for a real board name):

```
#!/bin/bash

PKG_CONFIG_LIBDIR=$(printf '%s:' "/build/<board>"/usr/*/pkgconfig)
export PKG_CONFIG_LIBDIR

export PKG_CONFIG_SYSROOT_DIR="/build/<board>"

# Portage will get confused and try to "help" us by exporting this.
# Undo that logic.
unset PKG_CONFIG_PATH

# Use full path to bypass automated wrapper checks that block `pkg-config`.
# https://crbug.com/985180
exec /usr/bin/pkg-config "$@"
```

And still fails... 

Just to note: it is not really needed at the moment for me to compile with test USE flag, I only needed the dummy plugin and that is on a different meson option so this is not a pressing matter. On the other hand, I would like to enable the self tests in the future. Do we know if this works on gentoo? (I don't have a gentoo system to test on)

superm1, 2021-05-24:
@polynomial-c can you comment on above?

djcampello, 2021-09-24:
Still an issue

closed 2021-09-24 by hughsie

djcampello, 2022-02-25:
Wow..... I'm so so sorry... I finally found why this was happening... We were modifying the meson.build from the .ebuild during build time for some reason (I don't even recall how this got in).... Fix: crrev.com/c/3491174

hughsie, 2022-02-25:
Well, at least it explains the error message. 🙂

PR #3794 (2021-09-24): trivial: workaround for ERROR: Unknown variable "gcab"
This change enables the use of tests in Chrome OS. Fixes #3270

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [X] Code fix
- [ ] Feature
- [ ] Documentation
commit 4fa2a760 (2021-09-24): trivial: workaround for ERROR: Unknown variable "gcab"

This change enables the use of tests in Chrome OS. Fixes #3270