fwupd/fwupd #3277: 'fwupdmgr get-updates' suggests an update is, and is not, available

LinuxOnTheDesktop, 2021-05-21:
**Describe the bug**

`fwupdmgr get-updates` produces output that seems to say (1) all devices are up to date, and (2) not all devices are up to date. See the, 'Steps to reproduce' section to see what I mean.

**Steps to Reproduce**
$ fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 6 local devices supported
$ fwupdmgr get-updates
• Thunderbolt Controller has no available firmware updates
• LENSE20512GMSP34MEAT2TA has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
• Unifying Receiver has the latest available firmware version
20KH001UKP
│
└─UEFI Device Firmware:
  │   Device ID:           9e329270a7a68d289c82fe77d32d02208ddf0890
  │   Current version:     0.71.12
  │   Minimum Version:     0.71.12
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                cea87551-1701-43fb-afbc-6e8ce9728345
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th System Update:
        New version:       0.73.20
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware
        License:           Proprietary
        Size:              439.6 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20
        
         • Do NOT turn off your computer or remove the AC adaptor while update is in progress

**Expected behavior**
The user should be given to understand clearly whether any update is available. I think that the output given above means to say that the following. The computer - identified in the output as ''20KH001UKP' - has firmware that can be updated. Yet, the output does not identify that firmware in the manner in which that same output identifies the other relevant (i.e. potentially updatable) components, namely, with a bullet point. It is all jolly confusing. Or so it seems to me. (Perhaps though the confusing output is a result of [this problem](https://github.com/fwupd/fwupd/issues/3231).)

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fclient version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20KH001UKP
│
├─Thunderbolt Controller:
│     Device ID:           9e4ed2585a7c7ad2de1e4593bb8b631301584cea
[... 819 lines cut ...]
│       └── zfs.mod
├── initrd.img -> initrd.img-5.8.0-53-generic
├── initrd.img-5.4.0-73-generic
├── initrd.img-5.8.0-50-generic
├── initrd.img-5.8.0-53-generic
├── initrd.img.old -> initrd.img-5.4.0-73-generic
├── System.map-5.4.0-73-generic
├── System.map-5.8.0-50-generic
├── System.map-5.8.0-53-generic
├── vmlinuz -> vmlinuz-5.8.0-53-generic
├── vmlinuz-5.4.0-73-generic
├── vmlinuz-5.8.0-50-generic
├── vmlinuz-5.8.0-53-generic
└── vmlinuz.old -> vmlinuz-5.4.0-73-generic

21 directories, 653 files
```

**Additional questions**
- Operating system and version: Mint 20.1 x64 Cinnamon.
- Have you tried rebooting? Yes.
- Is this a regression? Unknown.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

superm1, 2021-05-24:
So are you meaning to say that instead of

```
• Unifying Receiver has the latest available firmware version
20KH001UKP
```

You would like to see:
```
• Unifying Receiver has the latest available firmware version
• The following devices have firmware updates available:
20KH001UKP
```

LinuxOnTheDesktop, 2021-05-25:
&nbsp;&nbsp;&nbsp;&nbsp;I fear not. Rather I propose that you go with something like the following.

[image]

**Problems with the current output of the `refresh` command**

1) Text of the form '• Unifying Receiver has the latest available firmware version' suggests that each bullet point will comprise a line that mentions a device. However: '• The following devices have firmware updates available' does not mention a device, at least not until its next line, and that next line lacks a bullet-point.

2) It is unclear that the next line even mentions a device, at least in the case of '20KH001UKP'. For, that string is the model number of my _computer_, a computer that _has_ e.g. a BIOS. But, OK, perhaps e.g. the mouse is counted as one device, and the computer itself as another. On the other hand . . the connective symbols (` |` and `└─ `) within the output suggest that, actually,  '20KH001UKP' _is_ being treated as something that has other devices.

3) The following two things seem to be distinguished from each other insufficiently clearly: the current firmware of the updateable device; the update to that firmware.

superm1, 2021-05-26:
Perhaps you would consider working out some patches to what behavior you think makes sense?

> Text of the form '• Unifying Receiver has the latest available firmware version' suggests that each bullet point will comprise a line that mentions a device. However: '• The following devices have [...]

I'm afraid this is going to be a lot of bike shedding to decide which way to do it.
The one thing I'll mention - walls of text are never a useful thing.  

> It is unclear that the next line even mentions a device, at least in the case of '20KH001UKP'. For, that string is the model number of my computer, a computer that has e.g. a BIOS. But, OK, perhaps [...]

You have your OEM to thank for that.  Some manufacturers use useful strings for end users in SMBIOS tables, others don't.  This is `/sys/class/dmi/id/product_name`

The idea here is that it shows the topology of things connected, and matches what you see in `get-devices` output.

LinuxOnTheDesktop, 2021-05-26:
&nbsp;&nbsp;&nbsp;&nbsp;Thanks.

&nbsp;&nbsp;&nbsp;&nbsp;'Perhaps you would consider working out some patches to what behavior you think makes sense?' I would, but in some (most?) ways I'm ill-qualified to do it; indeed I would have to learn a lot before I was even in a position to submit a patch.

&nbsp;&nbsp;&nbsp;&nbsp;However: upon reflection, and helped by your post, I think that what I am proposing is not much of a change. I am proposing that the `refresh` output take the following form. Note the labelling of the computer.

Computer ('20KH001UKP')
└─ Thunderbolt Controller has no available firmware updates
└─ LENSE20512GMSP34MEAT2TA has the latest available firmware version
└─System Firmware is updatable:
&nbsp;&nbsp;&nbsp;&nbsp; └─System Firmware, CURRENT:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Device ID:           9e329270a7a68d289c82fe77d32d02208ddf0890
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Current version:     0.71.12
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Minimum Version:     0.71.12
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Vendor:              DMI:LENOVO
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Update State:        success
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;GUID:                cea87551-1701-43fb-afbc-6e8ce9728345
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Device Flags:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;• Internal device
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;• Updatable
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;• Requires AC power
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; • Supported on remote server
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; • Needs a reboot after installation
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;• Device is usable for the duration of the update
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; └─System Firmware, AVAILABLE UPDATE:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Version:       0.73.20
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Remote ID:         lvfs
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Summary:           Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;License:           Proprietary
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Size:              439.6 kB
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Vendor:            Lenovo Ltd.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Flags:             is-upgrade
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Description:       Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20

&nbsp;&nbsp;&nbsp;&nbsp;Also: the better that updates work, the less one scrutinises output. So: the more other things work, there less the text output will be a problem. I note lastly that I do not think that I should be seeing what I see, in the sense that fwupd is not working as designed here. For: if the following is true

> LENSE20512GMSP34MEAT2TA has the latest available firmware version
> • System Firmware has the latest available firmware version
> • UEFI Device Firmware has the latest available firmware version
> • UEFI [...]

then why am I being offered an update? (Also: why two entries for 'UEFI Device Firmware'?)

superm1, 2021-05-28:
> What I am proposing is that the refresh output take the following form. Note the labelling of the computer.

That's a HUGE change for refresh.  `refresh` output doesn't say anything about the details of the devices, just the number of devices supported and whether it was a pass/fail.  Adding all that extra information is a lot of work.  You can look at `get-updates` output for that information you want.

>then why am I being offered an update? 
You appear to have 3 UEFI device firmware entries and one system firmware entry.  Seems correct to me.  Like I said, as a user you can't tell what these actually are.

>(Also: why two entries for 'UEFI Device Firmware'?)

Because your manufacturer has multiple UEFI devices that can be updated, but we don't know what they actually specify.  The ESRT specification has no detail about this to convey it to the user in any way.  Similarly if you look in Windows device manager you will see several "UEFI device" items listed.

closed 2021-05-28 by superm1