fwupd/fwupd #3290: compile error when config_tests = false

djcampello, 2021-05-25:
I am not sure why this is happening on *some* of our boards but it seems related to https://github.com/fwupd/fwupd/blob/master/plugins/test/meson.build#L49 and -Dtests=false

```
fwupd-1.6.0-r114: meson --buildtype plain --libdir lib64 --localstatedir /var/lib --prefix /usr --sysconfdir /etc --wrap-mode nodownload --cross-file /build/zork/tmp/portage/sys-apps/fwupd-1.6.0-r114/temp/meson.x86_64-cros-linux-gnu.amd64 --localstatedir /var -Dbuild=standalone -Dagent=false -Dplugin_amt=false -Dlibarchive=false -Dbluez=true -Dplugin_dell=false -Delogind=false -Dplugin_flashrom=false -Dgnutls=true -Dgtkdoc=false -Dgusb=true -Dplugin_altos=true -Dman=false -Dplugin_nvme=true -Dintrospection=false -Dpolkit=false -Dplugin_synaptics_mst=true -Dplugin_synaptics_rmi=true -Dsystemd=false -Dtests=false -Dplugin_thunderbolt=true -Dplugin_uefi_capsule=false -Dplugin_uefi_pk=false -Dconsolekit=false -Dcurl=true -Dplugin_modem_manager=false -Dplugin_tpm=false /build/zork/tmp/portage/sys-apps/fwupd-1.6.0-r114/work/fwupd-1.6.0 /build/zork/tmp/portage/sys-apps/fwupd-1.6.0-r114/work/fwupd-1.6.0-build
...
fwupd-1.6.0-r114: [10/329] x86_64-cros-linux-gnu-clang -Iplugins/test/libfu_plugin_invalid.so.p -Iplugins/test -I../fwupd-1.6.0/plugins/test -I. -I../fwupd-1.6.0 -Ilibfwupd -I../fwupd-1.6.0/libfwupd -Ilibfwupdplugin -I../fwupd-1.6.0/libfwupdplugin -I/build/zork/usr/include/libxmlb-1 -I/build/zork/usr/include/libmount -I/build/zork/usr/include/blkid -I/build/zork/usr/lib64/libffi-3.1/include -I/build/zork/usr/include/glib-2.0 -I/build/zork/usr/lib64/glib-2.0/include -I/build/zork/usr/include/gio-unix-2.0 -I/build/zork/usr/include/gudev-1.0 -I/build/zork/usr/include/json-glib-1.0 -I/build/zork/usr/include/gusb-1 -I/build/zork/usr/include/libusb-1.0 -Xclang -fcolor-diagnostics -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -Os -march=znver1 -g -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -ffunction-sections -fdata-sections -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginTest"' -MD -MQ plugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o -MF plugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o.d -o plugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o -c ../fwupd-1.6.0/plugins/test/fu-plugin-invalid.c
fwupd-1.6.0-r114: [31mFAILED: [0mplugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o 
fwupd-1.6.0-r114: x86_64-cros-linux-gnu-clang -Iplugins/test/libfu_plugin_invalid.so.p -Iplugins/test -I../fwupd-1.6.0/plugins/test -I. -I../fwupd-1.6.0 -Ilibfwupd -I../fwupd-1.6.0/libfwupd -Ilibfwupdplugin -I../fwupd-1.6.0/libfwupdplugin -I/build/zork/usr/include/libxmlb-1 -I/build/zork/usr/include/libmount -I/build/zork/usr/include/blkid -I/build/zork/usr/lib64/libffi-3.1/include -I/build/zork/usr/include/glib-2.0 -I/build/zork/usr/lib64/glib-2.0/include -I/build/zork/usr/include/gio-unix-2.0 -I/build/zork/usr/include/gudev-1.0 -I/build/zork/usr/include/json-glib-1.0 -I/build/zork/usr/include/gusb-1 -I/build/zork/usr/include/libusb-1.0 -Xclang -fcolor-diagnostics -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -Os -march=znver1 -g -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -ffunction-sections -fdata-sections -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginTest"' -MD -MQ plugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o -MF plugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o.d -o plugins/test/libfu_plugin_invalid.so.p/fu-plugin-invalid.c.o -c ../fwupd-1.6.0/plugins/test/fu-plugin-invalid.c
fwupd-1.6.0-r114: In file included from ../fwupd-1.6.0/plugins/test/fu-plugin-invalid.c:9:
fwupd-1.6.0-r114: [1m../fwupd-1.6.0/libfwupdplugin/fu-plugin-vfuncs.h:17:10: [0m[0;1;31mfatal error: [0m[1m'fu-hash.h' file not found[0m
fwupd-1.6.0-r114: #include "fu-hash.h"
fwupd-1.6.0-r114: [0;1;32m         ^~~~~~~~~~~
fwupd-1.6.0-r114: [0m1 error generated.
```

djcampello, 2021-05-25:
This also seems like a flake on how meson process the dependencies. It does not reproduce 100%

djcampello, 2021-05-25:
Actually, it does seem that the dependency was processed fine: 

```
fwupd-1.6.0-r114: [7/329] /build/zork/tmp/portage/sys-apps/fwupd-1.6.0-r114/temp/python3.6/bin/python3 /build/zork/tmp/portage/sys-apps/fwupd-1.6.0-r114/work/fwupd-1.6.0/libfwupdplugin/fu-hash.py libfwupdplugin/fu-hash.h ../fwupd-1.6.0/libfwupdplugin/fu-archive.c ../fwupd-1.6.0/libfwupdplugin/fu-bluez-device.c ../fwupd-1.6.0/libfwupdplugin/fu-cabinet.c ../fwupd-1.6.0/libfwupdplugin/fu-chunk.c ../fwupd-1.6.0/libfwupdplugin/fu-common.c ../fwupd-1.6.0/libfwupdplugin/fu-common-cab.c ../fwupd-1.6.0/libfwupdplugin/fu-common-guid.c ../fwupd-1.6.0/libfwupdplugin/fu-common-version.c ../fwupd-1.6.0/libfwupdplugin/fu-context.c ../fwupd-1.6.0/libfwupdplugin/fu-device-locker.c ../fwupd-1.6.0/libfwupdplugin/fu-device.c ../fwupd-1.6.0/libfwupdplugin/fu-dfu-firmware.c ../fwupd-1.6.0/libfwupdplugin/fu-volume.c ../fwupd-1.6.0/libfwupdplugin/fu-firmware.c ../fwupd-1.6.0/libfwupdplugin/fu-firmware-common.c ../fwupd-1.6.0/libfwupdplugin/fu-dfuse-firmware.c ../fwupd-1.6.0/libfwupdplugin/fu-fmap-firmware.c ../fwupd-1.6.0/libfwupdplugin/fu-hwids.c ../fwupd-1.6.0/libfwupdplugin/fu-ihex-firmware.c ../fwupd-1.6.0/libfwupdplugin/fu-io-channel.c ../fwupd-1.6.0/libfwupdplugin/fu-plugin.c ../fwupd-1.6.0/libfwupdplugin/fu-quirks.c ../fwupd-1.6.0/libfwupdplugin/fu-security-attrs.c ../fwupd-1.6.0/libfwupdplugin/fu-smbios.c ../fwupd-1.6.0/libfwupdplugin/fu-srec-firmware.c ../fwupd-1.6.0/libfwupdplugin/fu-efi-signature.c ../fwupd-1.6.0/libfwupdplugin/fu-efi-signature-list.c ../fwupd-1.6.0/libfwupdplugin/fu-efivar.c ../fwupd-1.6.0/libfwupdplugin/fu-udev-device.c ../fwupd-1.6.0/libfwupdplugin/fu-usb-device.c ../fwupd-1.6.0/libfwupdplugin/fu-hid-device.c
```

djcampello, 2021-05-25:
Awesome! Thank you!

closed 2021-05-25 by djcampello

hughsie, 2021-05-25:
I'll do a 1.6.1 at some point this week I think.