fwupd/fwupd #3292: Dell XPS 15 9550 not detected

Gelmo, 2021-05-26:
**Describe the bug**
This issue is the same that was described in issue #2671 

Just wanted to report that I had the same issue today with the same laptop on 1.13.1 (Device ID 2da31270d317b076424992de14a0f08ae373c137 and GUID 124c207d-5db8-4d95-bd31-34fd971b34f9). There still seems to be something interfering that you may want to review for other device owners.

fwupdmgr did not find this as a supported device, even after a forced refresh. I manually installed 1.14.0 via fwupdtool.

After the update to 1.14.0, fwupdmgr still does not see this as a supported device. I submitted a report via get-devices a few hours ago, in case that's helpful.

The Dell site suggests that the latest bios for this is 1.14.0. When searching the GUID on fwupd.org, the results show that the latest version is 1.16.1. Should I attempt to manually install 1.16.1? Is that version not intended for my device? If so, perhaps the listings at fwupd.org need more adjustments. Let me know if there's anything I could provide to assist with this.

**General Info**
```shell
$ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
```

```shell
$ sudo fwupdmgr get-releases
No supported devices
```

**Expected behavior**
The device should be detected and supported, as the GUID is listed at fwupd.org.

**fwupd version information**
```shell
$ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.8
```

Installed via Pop OS's default configuration

**fwupd device information**
```shell
$ sudo fwupdmgr get-devices --show-all-devices
XPS 15 9550
│
├─BCM920703 Bluetooth 4.1:
│     Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│     Current version:    1.12
│     Vendor:             Broadcom Corp (USB:0x0A5C)
│     Serial Number:      441CA8E1EC60
│     GUIDs:              35fd074d-7c78-599e-b37a-88931c63ba01 ← USB\VID_0A5C&PID_6410&REV_0112
│                         f8498646-6724-5f55-9806-57b3f8636ad8 ← USB\VID_0A5C&PID_6410
│     Device Flags:       • Updatable
│
├─GM107M [GeForce GTX 960M] (XPS 15 9550):
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a2
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              5d36a05e-42ad-5b87-8a2d-27b8bd7bfe04 ← PCI\VEN_10DE&DEV_139B&SUBSYS_102806E4&REV_A2
│                         4bbd4dbf-83dd-58ea-a5ad-081ca6f8bca9 ← PCI\VEN_10DE&DEV_139B&SUBSYS_102806E4
│                         995dfa05-5863-56db-8087-221c0cce69a4 ← PCI\VEN_10DE&DEV_139B&REV_A2
│                         2195ad44-2598-565c-819e-16026a8d2f22 ← PCI\VEN_10DE&DEV_139B
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─HD Graphics 530 (XPS 15 9550):
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    06
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              bf3a0b36-1995-5fe3-b81d-9effc018f659 ← PCI\VEN_8086&DEV_191B&SUBSYS_102806E4&REV_06
│                         cc0161dd-f85a-5d3c-8a6f-329aeffd9c55 ← PCI\VEN_8086&DEV_191B&SUBSYS_102806E4
[... 151 lines cut ...]
│       │   └── Recovery-DA8B-2159.conf
│       ├── loader.conf
│       └── random-seed
├── initrd.img -> initrd.img-5.11.0-7614-generic
├── initrd.img-5.11.0-7612-generic
├── initrd.img-5.11.0-7614-generic
├── initrd.img-5.3.0-7642-generic
├── initrd.img.old -> initrd.img-5.11.0-7612-generic
├── System.map-5.11.0-7612-generic
├── System.map-5.11.0-7614-generic
├── vmlinuz -> vmlinuz-5.11.0-7614-generic
├── vmlinuz-5.11.0-7612-generic
├── vmlinuz-5.11.0-7614-generic
└── vmlinuz.old -> vmlinuz-5.11.0-7612-generic

16 directories, 34 files
```

**Additional questions**
- Operating system and version: Pop OS (tested on 20.04 and 20.10)
- Have you tried rebooting? Yes.
- Is this a regression? Perhaps; issue #2671 suggests that this was previously corrected.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

hughsie, 2021-05-26:
I'm a bit confused; the last update for 124c207d-5db8-4d95-bd31-34fd971b34f9 was 3 years ago. The reason it's not showing up as supported is the firmware has CHID requirements of e7eb42e2-06c6-5185-9c5f-33ff1dbb1c0b or d98dc575-9143-5a8f-a620-9440ef640c8d which I'm guessing you don't match. (sudo fwupdtool hwids).

Gelmo, 2021-05-26:
I do seem to meet that requirement:

```shell
$ sudo fwupdtool hwids
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.14.0
BiosMajorRelease: 1
BiosMinorRelease: 14
Manufacturer: Dell Inc.
Family: XPS
ProductName: XPS 15 9550
ProductSku: 6E4
EnclosureKind: 9
BaseboardManufacturer: Dell Inc.
BaseboardProduct: N7TVV

Hardware IDs
------------
{89f94ddb-7b99-5570-b4ae-1df409590915}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{a49d8e03-5b4c-52c4-8de2-2f4423264f3d}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{ba56b8f0-994e-53b0-b667-c18715c16262}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{9c1c3ab8-d0fc-5358-9172-05b9d5391570}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{70ecbe19-7e56-526c-9860-9e7f37716ebb}   <- Manufacturer + Family + ProductName + ProductSku
{a1352d1a-a429-5cb2-9f9b-3fd3aaead809}   <- Manufacturer + Family + ProductName
{2c794b86-60f9-5260-8e5a-2fe1ef7ba2c8}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{6e9d152f-c527-5443-bcc8-ef11c1ecbcf4}   <- Manufacturer + ProductSku
{1aa331da-22d4-5ae2-a6b7-d7d70cb10e5b}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{e7eb42e2-06c6-5185-9c5f-33ff1dbb1c0b}   <- Manufacturer + ProductName
{0ce55c98-8152-5d9b-ae73-8af629e67e3b}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{61178075-a8fd-563c-9045-44227d8c121f}   <- Manufacturer + Family
{29ca4080-4d68-5395-afe2-35fa9a4f13bf}   <- Manufacturer + EnclosureKind
{c9b4af59-c773-5718-85e8-d6149061aae7}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer
```

Gelmo, 2021-05-27:
Sorry for the late reply; went to bad after that post and had a busy work day! I have some extra info to provide that could be related. This particular model has a Nvidia GPU as well; perhaps Dell overlooked this sku or excluded it intentionally.

```shell
$ sudo /usr/libexec/fwupd/fwupd --verbose
00:37:06:0032 FuDebug              Verbose debugging enabled (on console 1)
00:37:06:0032 FuConfig             loading config values from /etc/fwupd/daemon.conf
00:37:06:0032 FuConfig             using autodetected max archive size 18.4 EB
00:37:06:0032 FuRemoteList         loading /usr/share/fwupd/metainfo
00:37:06:0032 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
00:37:06:0033 XbSilo               file: 3e34937e-05a8-54b2-6ce5-f3dad4cf0df7, current:3e34937e-05a8-54b2-6ce5-f3dad4cf0df7, cached: (null)
00:37:06:0033 XbSilo               loading silo with file contents
00:37:06:0033 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
00:37:06:0033 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
00:37:06:0033 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
00:37:06:0033 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
00:37:06:0033 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
00:37:06:0033 FuRemoteList         ordering lvfs-testing=lvfs+1
00:37:06:0033 FuRemoteList         ignoring unfound remote fwupd
00:37:06:0033 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
00:37:06:0033 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
00:37:06:0033 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
00:37:06:0033 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
00:37:06:0033 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
00:37:06:0033 Jcat                 loaded 1 certificates
00:37:06:0033 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
00:37:06:0033 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
00:37:06:0033 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
00:37:06:0034 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
00:37:06:0034 Jcat                 loaded 1 certificates
00:37:06:0034 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
00:37:06:0034 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
00:37:06:0040 FuEngine             client certificate exists and working
00:37:06:0040 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
00:37:06:0040 FuHistory            got schema version of 6
00:37:06:0040 FuIdle               setting timeout to 7200s
00:37:06:0040 FuHwids              smbios property Manufacturer=Dell Inc.
00:37:06:0040 FuHwids              smbios property EnclosureKind=9
00:37:06:0040 FuHwids              smbios property Family=XPS
00:37:06:0040 FuHwids              smbios property ProductName=XPS 15 9550
00:37:06:0040 FuHwids              smbios property ProductSku=06E4
00:37:06:0040 FuHwids              smbios property BiosVendor=Dell Inc.
00:37:06:0040 FuHwids              smbios property BiosVersion=1.14.0
00:37:06:0040 FuHwids              smbios property BiosMajorRelease=01
00:37:06:0040 FuHwids              smbios property BiosMinorRelease=0e
00:37:06:0040 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
00:37:06:0040 FuHwids              smbios property BaseboardProduct=0N7TVV
00:37:06:0040 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
00:37:06:0040 XbSilo               file: c18ff41a-a8b9-63d7-32e0-6085d4d84065, current:c18ff41a-a8b9-63d7-32e0-6085d4d84065, cached: (null)
00:37:06:0040 XbSilo               loading silo with file contents
00:37:06:0041 FuEngine             575 components now in silo
00:37:06:0045 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_dfu_csr.so)
00:37:06:0045 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_dell_esrt.so)
00:37:06:0045 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_linux_sleep.so)
00:37:06:0046 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_dfu.so)
00:37:06:0046 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_system76_launch.so)
00:37:06:0047 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_superio.so)
00:37:06:0047 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_thelio_io.so)
00:37:06:0047 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_pixart_rf.so)
00:37:06:0047 FuPlugin             added udev subsystem watch of hidraw
00:37:06:0048 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_acpi_facp.so)
00:37:06:0048 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_redfish.so)
00:37:06:0049 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-3/libfu_plugin_ata.so)
00:37:06:0049 FuPlugin             added udev subsystem watch of block
[... cut ...]
00:37:09:0138 XbSilo               ignoring for OR statement: indexed string 'c6682ade-b5ec-57c4-b687-676351208742' was unfound
00:37:09:0142 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
00:37:09:0142 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ‘gio-memory-monitor’
00:37:09:0164 FuMain               Daemon ready for requests (locale en_US.UTF-8)
00:37:09:0166 FuMain               acquired name: org.freedesktop.fwupd
```

superm1, 2021-05-27:
I think this *might* be a case that updates keep coming for the Precision version but not the XPS version.

Gelmo, 2021-05-27:
> I think this _might_ be a case that updates keep coming for the Precision version but not the XPS version.

If that were the case, wouldn't it say my device is supported but no update is available?

superm1, 2021-05-27:
Look at https://github.com/fwupd/fwupd/issues/2671 specifically, https://github.com/fwupd/fwupd/issues/2671#issuecomment-743287079

I'm no longer with Dell so I can't confirm if anything has changed on this.

Gelmo, 2021-05-27:
> Look at #2671 specifically, [#2671 (comment)](https://github.com/fwupd/fwupd/issues/2671#issuecomment-743287079)
> 
> I'm no longer with Dell so I can't confirm if anything has changed on this.

Yes I understand this, but fwupdmgr should still list my device as supported and say that there's no update available.

There are 2 entries for XPS 15 9550 at fwupd.org:

https://fwupd.org/lvfs/devices/com.dell.uefi124c207d.firmware - Has 1.14.0 but also the Precision-only 1.16.1

https://fwupd.org/lvfs/devices/com.dell.uefi124c207d_precision5510.firmware - Has 1.14.2 and 1.15.0 for the Precision model

If fwupdmgr reports my device as not supported, that would suggest that there is something on my device that differs from the requirements in the 2 listings on the site. This would mean that Dell's listing is not accurate, or Dell does not support LVFS for this device, or there is something else going wrong in relation to fwupdmgr.

Am I misunderstanding the intended functionality? If my device is listed in the LVFS database and I'm on any firmware (even if it's the latest), should fwupdmgr report my device as supported? My question is not about eligibility for updates.

PS: Regardless of the cause of the issue, it seems likely that the Dell listings are not accurate, as different firmware versions for the Precision model are available at the above links.

superm1, 2021-05-28:
OK I think I found what's going on.  If you look at the stable metadata, you can see that 
```
    <requires>
      <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </requires>
```

Is set for the device in `firmware.xml.gz` _not_ in the **release**.  So it is applied to _all_ releases.

```
      <release version="68354" timestamp="1564495178" urgency="critical">
        <location>https://fwupd.org/downloads/8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab</location>
        <checksum type="sha1" filename="8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab" target="container">767bec5099a9ce9278ee95860fd64018af1617d5</checksum>
        <checksum type="sha256" filename="8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab" target="container">257a690baf81eec1ef98e5e57c31aede3055e5f6547c22629f99795d5ebba6d3</checksum>
        <checksum type="sha1" filename="firmware.bin" target="content">8e335187c2946574c103b0e44dc5d6747416e0ac</checksum>
        <checksum type="sha256" filename="firmware.bin" target="content">3e571e0b81239b23f33f59717541c447aa794bbea0c32f656c8b688fd5422463</checksum>
        <description>
          <p>This stable release fixes the following issues:</p>
          <ul>
            <li>Fixed the issue where the TPM option disappears from the BIOS setup menu intermittently.</li>
          </ul>
        </description>
        <size type="installed">10240280</size>
        <size type="download">10260780</size>
        <artifacts>
          <artifact type="binary">
            <location>https://fwupd.org/downloads/8d16ce33cf73505dfc76ff7609ed2909ffcb79ee-Precision_5510_1.11.2.cab</location>
            <filename>Precision_5510_1.11.2.cab</filename>
            <checksum type="sha1">767bec5099a9ce9278ee95860fd64018af1617d5</checksum>
            <checksum type="sha256">257a690baf81eec1ef98e5e57c31aede3055e5f6547c22629f99795d5ebba6d3</checksum>
            <size type="installed">10240280</size>
            <size type="download">10260780</size>
          </artifact>
        </artifacts>
      </release>
    </releases>
    <requires>
      <hardware>d98dc575-9143-5a8f-a620-9440ef640c8d</hardware>
    </requires>
  </component>
```

superm1, 2021-05-28:
I recall discussing this before. It happens when they share the same "id".

I think what needs to be done is rename the ID for all the old ones or rename the ID for new ones. @dell-client-linux

GoPerry, 2021-05-28:
> I recall discussing this before. It happens when they share the same "id".
> 
> I think what needs to be done is rename the ID for all the old ones or rename the ID for new ones. @dell-client-linux

Mario.
could you show us which ID used for firmware matching?
I can check that with Dell BIOS to make the ID differentiated for different system. @superm1

superm1, 2021-05-28:
Since the BIOS cab is built by automation and presumably the XPS version isn't supposed to have more updates it might be an easier process to rename the ID on all the old ones that apply to both to me.