fwupd/fwupd #3294: Failed to verify signature

djcampello, 2021-05-26:
I am trying to use the fwupd daemon on Chrome OS but I'm having issues with the daemon verifying signatures:

I am trying to update a device with `a5c8f0883f6089b780a25490b53eb8d5d6ba8a1d109ce44f18bd2b2ef3ffe315-Lenovo-ThinkPad-USBCGen2Dock-Firmware-49-0E-14.cab` and I get the following:

```
19:11:37:0186 FuEngine             Emitting PropertyChanged('Status'='decompressing')
19:11:37:0186 FuMain               Emitting PropertyChanged('Status'='decompressing')
19:11:37:0187 Jcat                 Jcat engine kind 'gpg' not supported
19:11:37:0187 Jcat                 Jcat engine kind 'gpg' not supported
19:11:37:0187 Jcat                 got 1 PKCS7 signatures
19:11:37:0187 Jcat                 signature failure: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
19:11:37:0187 FuCabinet            failed to verify firmware.metainfo.xml: signatures were required, but none supplied
19:11:37:0187 FuCabinet            processing file: firmware.metainfo.xml
19:11:37:0187 XbSilo               compiling f8473242e001823f7db30674cfb6a395ad3c1a25:filename=firmware.metainfo.xml:prefix=components?
19:11:37:0187 FuCabinet            processing release: 49-0E-14
19:11:37:0188 Jcat                 Jcat engine kind 'gpg' not supported
19:11:37:0188 Jcat                 Jcat engine kind 'gpg' not supported
19:11:37:0188 Jcat                 got 1 PKCS7 signatures
19:11:37:0188 Jcat                 signature failure: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
19:11:37:0188 FuCabinet            failed to verify payload firmware.bin: signatures were required, but none supplied
19:11:37:0188 FuEngine             Emitting PropertyChanged('Status'='idle')
19:11:37:0188 FuMain               Emitting PropertyChanged('Status'='idle')
```

Any pointers on what could be the problem?

djcampello, 2021-05-26:
I actually recall that this was brought up before and https://github.com/hughsie/libjcat/commit/109399e1f28cec84b43c355b2be77bac38943df7 should address it... I will check that we have that commit...

djcampello, 2021-05-26:
We do have it: crrev.com/c/2495757 So might be something else

hughsie, 2021-05-26:
> Any pointers on what could be the problem

Can you show the top bit of the log that shows the pkcs pubkey being added? Email fine if private.

djcampello, 2021-05-26:
```
19:11:29:0715 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:11:29:0715 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
19:11:29:0715 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
19:11:29:0716 Jcat                 loaded 1 certificates
19:11:29:0716 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
19:11:29:0716 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
19:11:29:0717 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:11:29:0717 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
19:11:29:0717 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
19:11:29:0718 Jcat                 loaded 1 certificates
19:11:29:0718 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
19:11:29:0718 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
```

I also attempted to (not sure if I'm missing some flags):

```
# jcat-tool verify firmware.jcat 
firmware.bin:
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED gpg: Jcat engine kind 'gpg' not supported
    FAILED pkcs7: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
    FAILED: Validation failed
firmware.metainfo.xml:
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED gpg: Jcat engine kind 'gpg' not supported
    FAILED pkcs7: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
    FAILED: Validation failed
Validation failed
```

hughsie, 2021-05-27:
> FAILED pkcs7: failed to verify data

You need to specify the  public key if they're not installed in your keyring, e.g. `jcat-tool verify firmware.jcat  --public-key /etc/pki/fwupd/LVFS-CA.pem`

hughsie, 2021-05-27:
I think we need to handle the "not supported" case much better, e.g. with `-Dgpg=false` I also get:

	firmware.bin:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
	    FAILED gpg: Jcat engine kind 'gpg' not supported
	    FAILED: Validation failed

hughsie, 2021-05-27:
@djcampello https://github.com/hughsie/libjcat/pull/53 makes the jcat-tool debugging more accurate but won't actually fix this issue.

hughsie, 2021-05-27:
Huh, so I can reproduce -- compiling jcat with `-Dgpg=false` and fwupd with `-Dpolkit=false` -- debugging now!

hughsie, 2021-05-27:
> compiling jcat with -Dgpg=false

Actually, that's just a symptom of something else -- on PKCS failure we fall back to GPG which does work.

Testing more, I get the same failure when doing `fwupdmgr get-details` on `Lenovo-ThinkPad-USBCGen2Dock-Firmware-49-0E-14.cab` but not on `hughski-colorhug2-2.0.7.cab` -- but decompressing the latter and then running `jcat-tool verify firmware.jcat  --public-key /etc/pki/fwupd/LVFS-CA.pem` also fails in the same way. Debugging further.

hughsie, 2021-05-27:
Okay, worked it out. New files on the LVFS were created with the PKCS certificate included, and old files were not. This was changed in https://gitlab.com/fwupd/lvfs-website/-/commit/6659b6b23bfb4589e5359490000d34bbde5caa70 -- and you can verify using:

    jcat-tool export firmware.jcat
    certtool --p7-info --infile=./firmware.bin.p7b

...and then the one that fails won't have a `-----BEGIN CERTIFICATE-----` block. The reason we ship the intermediate cert in the jcat file is that we only ship the CA certificate to clients. The CA then signs the LVFS instance cert, and that signs firmware. If we replaced the LVFS or if it was compromised in some way we'd sign a new server key with the presumed safe CA and we wouldn't need to do client updates (but we would of course have to resign all the existing firmware). Of course, without the intermediate cert available to the client then gnutls doesn't know that server cert was signed by the CA cert. The workaround is to install the server cert into trust list, or better, to resign the files that don't have the PKCS#7 cert included in the signature.

I'll ponder how to do the latter as part of a lvfs fsck action later.

hughsie, 2021-05-27:
> I'll ponder how to do the latter as part of a lvfs fsck action later.

https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1006

djcampello, 2021-05-27:
Does this mean that all (with older signatures) of the files in LVFS will be updated? Our mirror does not allow overwrites (because of portage usage) so it will require so manual fixing on my side if so. How often this type of update happens?

hughsie, 2021-05-27:
I'm running it on staging right now, 1h in its found 5 firmwares that need resigning. I'll let you know what the conclusion is.

superm1, 2021-05-28:
Although it's valid I wouldn't say it's necessarily standard practice and encouraged to do.

This of course also breaks people who have an old copy of metadata until their next refresh cycle too.

hughsie, 2021-05-28:
> breaks people who have an old copy of metadata

True indeed, and we've seen this in the wild, so to speak. We could use the sha256 of the generated archive instead of the uploaded archive perhaps? Or add a -v1 as a suffix. The suffix might be better as then you could show the signing epoch.

Maybe we need a FirmwareArtifact object that we GC after say 7 days? That seems like the least hacky, but most work solution.

djcampello, 2021-05-28:
-v1 suffix works pretty well with portage as this is effectively done for src.tar.gz packages. Mainly the issue comes when a portage repository is branched and kept stable. ebuilds will point to .cab files and Manifest files will have a checksum for those .cab. If the LVFS mirror changes the files under portage then the builder that is using the stable portage branch will fail when verifying against the old checksum in the portage Manifest.

superm1, 2021-05-28:
If that's achievable, yeah I think that's a good idea to add a suffix to a newly generated archive and garbage collect the old one.
No one should really break then if you get the GC timeline right (I think at _least_ 2 days, no concerns with longer).

I guess the question I would have though is how does mirroring stuff know to discard the old one when it garbage collects on LVFS?  Does `PULP_MANIFEST` have a way to reflect that?  If not, maybe we need something extra to tell people that?

superm1, 2021-05-28:
> -v1 suffix works pretty well with portage as this is effectively done for src.tar.gz packages. Mainly the issue comes when a portage repository is branched and kept stable. ebuilds will point to [...]

Hopefully there is some automation that can update those manifests, otherwise this might turn into a very tedious non-obvious action.

djcampello, 2021-05-28:
There is none that I'm aware of, what is done for regular src packages is that our Gentoo mirror keeps all versions (since each has a different name)

hughsie, 2021-06-08:
Little tested LVFS patch here: https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1009 if I get some more time tomorrow I'll clean it up a bit.

hughsie, 2021-06-11:
LVFS patch deployed, extended fsck running. It'll take a few (4?) hours to complete:

[image]

hughsie, 2021-06-11:
Ha, so the "_5 firmwares_" was a woeful mis-underestimation. It was 161 firmwares!

Firmwares that needed resigning because they were too old to support PKCS#7: 29, 73, 193

Firmwares that needed resigning because the PKCS#7 cert was invalid: 1975, 2427, 3840, 4104, 4105, 4140, 4159, 4558, 4692, 4983, 5047, 5049, 5050, 5295, 5336, 5337, 5348, 5369, 5640, 5641, 5750, 5751, 5792, 5793, 5797, 5799, 5801, 5804, 5832, 5833, 5834, 5835, 5836, 5839, 5841, 5842, 5843, 5844, 5845, 5846, 5847, 5848, 5853, 5856, 5857, 5858, 5863, 5864, 5865, 5866, 5868, 5869, 5870, 5871, 5872, 5873, 5874, 5876, 5877, 5879, 5880, 5881, 5882, 5883, 5884, 5887, 5888, 5890, 5891, 5893, 5894, 5895, 5898, 5899, 5900, 5901, 5902, 5904, 5906, 5907, 5908, 5909, 5910, 5912, 5913, 5914, 5915, 5917, 5918, 5919, 5920, 5921, 5922, 5923, 5924, 5926, 5927, 5928, 5929, 5930, 5931, 5932, 5933, 5934, 5935, 5937, 5940, 5941, 5942, 5943, 5948, 5951, 5953, 5954, 5958, 5959, 5961, 5962, 5963, 5964, 5965, 5966, 5967, 5968, 5969, 5970, 5971, 5972, 5982, 5983, 5984, 5985, 5986, 5987, 5988, 5989, 5992, 5994, 5996, 5998, 5999, 6000, 6011, 6020, 6021, 6022, 6023, 6024, 6025, 6026, 6027, 6028, 6029, 6033, 6036, 6042, 6044, 6045, 6052

@djcampello can you sync the metadata again and retry please. The USBCGen2Dock-Firmware-49-0E-14.cab was resigned with a new filename (`2e0bf8aaf9c63ca11cfe3444d032277c21ec0d678e5963123a8b33e5dcd37d99-Lenovo-ThinkPad-USBCGen2Dock-Firmware-49-0E-14.cab`), so please use the **new** one, not the **old** one.

djcampello, 2021-06-11:
Was this one resigned? d619a3c051e33df53094e274fbb672bdfa50d19d950ec337de36a68fe682fe18-Lenovo-ThinkPad-USBCGen2Dock-DP-Firmware-5.05.00.cab

hughsie, 2021-06-11:
Yes, the new one you want is ac37f23af002e91df11094b08fd2e076cf9c8cb4f08930be8eefe35850097a60-Lenovo-ThinkPad-USBCGen2Dock-DP-Firmware-5.05.00.cab

djcampello, 2021-06-11:
For some reason the mirroring task did not pick that one yet... I'll wait for the next run

hughsie, 2021-06-11:
what's the difference between those two runs? are you changing the BaseURI and doing a `fwupdmgr refresh` between them? If so, what's the refresh look like for the failing one?

djcampello, 2021-06-11:
I am using fwupdtool with the vendor directory remote... So really the difference between the two runs is which .can I placed under /usr/share/fwupd/remotes.d/vendor/firmware/

Something to note is that if I use fwupdtool install with the new .cab it works

hughsie, 2021-06-11:
Ahh, fwupdtool, so no concerns about restarting the daemon. Are *both* files in that directory now? Could you share a fwupdtool --verbose log so we can see the setup? private email is fine too.

djcampello, 2021-06-11:
```
campello@derch:/tmp$ diff a b
diff a/firmware.bin.asc b/firmware.bin.asc
3,10c3,10
< iQEzBAABCAAdFiEEP8a4BEEO0IQNjy+XSKbYDkU4usIFAl/KAlQACgkQSKbYDkU4
< usJnnggAiOOMRJvg/WW9nGQKpjGa3sk8ypLxPDHnX7zKKIjtMTk+x0z/5rNAZiLE
< TMhL0q9TA14QFYdK/i8vxNpNTOUXYIagSwLJdp9LRqpuxtrnaE8o57+CyBt/+EQA
< 23tO2Tw0Cr9ek19q7hPLyKUL1zb4rwFP/RJBNZNB34oW3FS3iIdvLbNhO8QvCMhu
< CAaLFOTXHBJg5WvJrTHCDrbT01f8TjstUcghVl2E/YHK/bs7H+upvXDxZQN5cyTw
< r00/S39ii5FfEGLuMEkkh0w3X5Nx8xPpYvxUOjYUdT+i2IGeuJRcaFYTjrZP3+YA
< Roww5BMnvknRGeruiG+Ax2BA51hAwQ==
< =dYCq
---
> iQEzBAABCAAdFiEEP8a4BEEO0IQNjy+XSKbYDkU4usIFAmDDOtwACgkQSKbYDkU4
> usJ5gwf/S8Yvw7y1gzb7YTjrdSRudprxr1jmk0ZMhltLCHKk1RuRy0DHwRbFXvf9
> D08elGqWdpKm1TR48+j+jbZXpUfmFrVa0CwaD9f8g4+XLjqW/g/OFil/MfBB2Uf1
> jrBbGkEoskf1sX5wEi2Ez1H1PQh1HFNT08c2sdKKxe+Df1rRU7/IgMd0D4oborKd
> 8cWCGicLlhFa2vS9HSXxn8HjnB83QMm26kHKXDcPDRvTJT5KUCv0FIDKl67XewM6
> iXxg+O50K1W781qgoEy1QWzr5zs9kxD7kmNK5DpPArS8gayPngsmMoQ2r7VDSzeM
> 6T8ZyEDjvdoAxZu6lAccWtFSyTYGQg==
> =HOEr
Binary files a/firmware.jcat and b/firmware.jcat differ
diff a/firmware.metainfo.xml b/firmware.metainfo.xml
30a31,37
>       <artifacts>
>         <artifact type="binary">
>           <filename>firmware.bin</filename>
>           <checksum type="sha1">293561166ad374f69bc8e889ce75dd62c722524a</checksum>
>           <checksum type="sha256">92011bdebb45c904f72339d5496ced93274194e6aee8666aed7ea2565904e2aa</checksum>
>         </artifact>
>       </artifacts>
diff a/LICENSE.txt b/LICENSE.txt
1c1
< Copyright 2020 Lenovo
---
> Copyright 2020-2021 Lenovo
diff a/README.txt b/README.txt
7c7
<     fwupdmgr install a5c8f0883f6089b780a25490b53eb8d5d6ba8a1d109ce44f18bd2b2ef3ffe315-Lenovo-ThinkPad-USBCGen2Dock-Firmware-49-0E-14.cab
---
>     fwupdmgr install Lenovo-ThinkPad-USBCGen2Dock-Firmware-49-0E-14.cab
16c16
< This firmware archive was signed by the LVFS on 2020-12-04T09:33:09 and was made available
---
> This firmware archive was signed by the LVFS on 2021-06-11T10:28:44 and was made available
```

djcampello, 2021-06-11:
Removing the following makes the cab work:

```
diff a/firmware.metainfo.xml b/firmware.metainfo.xml
30a31,37
>       <artifacts>
>         <artifact type="binary">
>           <filename>firmware.bin</filename>
>           <checksum type="sha1">293561166ad374f69bc8e889ce75dd62c722524a</checksum>
>           <checksum type="sha256">92011bdebb45c904f72339d5496ced93274194e6aee8666aed7ea2565904e2aa</checksum>
>         </artifact>
>       </artifacts>
```

djcampello, 2021-06-15:
It seems that the fix worked and I was able to `fwupdmgr update` as well. Thank you!

closed 2021-06-15 by djcampello