fwupd/fwupd #3296: Arch containers have been failing to build

superm1, 2021-05-28:
@ArchangeGabriel I've noticed the nightly build for fwupd arch docker containers has been failing for a while.  It hasn't broken CI (we've just used an older container for a while now), but this is surprising to me.

Full Build log is here:
https://github.com/fwupd/fwupd/runs/2690192596?check_suite_focus=true

But the error is:
```
Step 1/12 : FROM archlinux/base
pull access denied for archlinux/base, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
Error: exit status 1
Usage:
  github-actions build-push [flags]
```

Which seems .... surprising.  Any thoughts?

superm1, 2021-06-08:
@ArchangeGabriel can you help?

ArchangeGabriel, 2021-06-08:
Sorry I’m quite busy lately, but anyway this is not something I can solve my self (I should have had a look earlier, I expected work that I would not have been able to produce), transmitting to our Docker devs. ;)

svenstaro, 2021-06-08:
`archlinux/base` has been deprecated for a long good while now. Simply switch it to `archlinux:latest` or `archlinux:base` and you should be good.

hashworks, 2021-06-08:
Note: There is `library/archlinux:base{,-devel}` in the official DockerHub library (updated weekly) and `archlinux/archlinux:base{,-devel}` in our DockerHub account (updated daily).

superm1, 2021-06-08:
I switched to `archlinux:latest` for now, and we'll see if that works later tonight.

closed 2021-06-08 by superm1

hashworks, 2021-06-08:
I would guess that GitHub provides a cached proxy for DockerHub. We don't push directly to the GitHub registry.

commit 723deed2 (2021-06-08): trivial: Fix dockerfile for Arch (Closes: #3296)