fwupd/fwupd #3304: T14s firmware update failure on boot, secure boot off, boot order lock off, luks encrupted disk

tabroughton, 2021-05-30:
**Describe the bug**
I have stock T14s AMD with firmware 0.1.9 and thought I ought to update to to the latest (0.1.30).  

`fwupdmgr update` runs as expected but when I reboot I get a failure as described here: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot

I have `boot order lock off` in the bios.
Also, I'm running majaro so have `secure boot off` and I also have seen reports on arch wiki about enabling secure boot with own keys will brick the T14s. 
Disk is LUKS encrypted

**Steps to Reproduce**

- Using T14s (2OUH)
- Secure boot off in bios
- Boot order lock off in bios
- fwupdmgr get-updates
- fwupdmgr update (Y, y)
- reboot
- fwupdmgr get-history : 

```
Previous version:   0.1.9
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2021-05-30 17:17
```

**Expected behavior**

fwupdmgr get-history :  success
(previous version 0.1.9, new 0.1.30)

**fwupd version information**

```
client version: 1.6.0
compile-time dependency versions
        gusb:   0.3.6

daemon version: 1.6.0
```
installed as part of distro: pamac (manjaro)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices

20UH001AUK
│
├─Integrated Camera:
│     Device ID:          54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:    58.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUIDs:              08c6b922-0e9b-5283-9e1a-2f5a549abf8e ← USB\VID_04F2&PID_B6CB&REV_5818
│                         95b07a8e-2063-5025-80b5-1fcf4ca8e9e3 ← USB\VID_04F2&PID_B6CB
│     Device Flags:       • Updatable
│   
├─System Firmware:
│ │   Device ID:          b4718caaf1dc79bb34f276c98081532f468822aa
│ │   Current version:    0.1.9
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              428ba8ba-40c8-4446-8d7a-394d6c28b26a
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       660e54f1-43fd-5b2b-a7b5-36422f9c8f2f ← UEFI\RES_{428BA8BA-40C8-4446-8D7A-394D6C28B26A}
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
[... 467 lines cut ...]
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-5.10-x86_64-fallback.img
├── initramfs-5.10-x86_64.img
├── initramfs-5.4-x86_64-fallback.img
├── initramfs-5.4-x86_64.img
├── linux510-x86_64.kver
├── linux510-x86_64.kver.old
├── linux54-x86_64.kver
├── memtest86+
│   └── memtest.bin
├── vmlinuz-5.10-x86_64
└── vmlinuz-5.4-x86_64

11 directories, 327 files

```

**Additional questions**
- Operating system and version: `5.12.2-1-MANJARO #1 SMP PREEMPT Fri May 7 17:53:15 UTC 2021 x86_64 GNU/Linux`
- Have you tried rebooting? yes
- Is this a regression? don't know, first time using
- Are you using an NVMe disk? yes
- Is secure boot enabled? no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no, it is lenovo but boot order lock is off

tabroughton, 2021-05-30:
The output of an update attempt:
```
$ fwupdmgr update     

Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for System Firmware from 0.1.9 to 0.1.30
20UH001AUK must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[***************************************]
Scheduling…              [***************************************]
Successfully installed firmware
Devices with the latest available firmware version:
 • WDC PC SN730 SDBQNTY-512G-1001

An update requires a reboot to complete. Restart now? [y|N]: n

$ efibootmgr -v      
BootNext: 0002
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,001B,0018,0019,0017,0002,001A,001C
Boot0001* Manjaro       HD(4,GPT,999c0fd4-8789-914b-bad9-5e86458f1bb4,0x800,0x100000)/File(\EFI\Manjaro\grubx64.efi)
Boot0002* Linux-Firmware-Updater        HD(4,GPT,999c0fd4-8789-914b-bad9-5e86458f1bb4,0x800,0x100000)/File(\EFI\manjaro\fwupdx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  ThinkShield secure wipe       FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0015  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PXE BOOT      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot001E* Boot Next Boot Option VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,a7ca6d35b2c2684783721826a7404894)
```

superm1, 2021-06-01:
Can you please check if there is any output from 
```
sudo fwupdate -L
```

If there is none, can you try to enable debug and do it one more time.
```
sudo fwupdate -d
fwupdmgr update
<reboot>
sudo fwupdate -L
```

exploide, 2021-06-03:
I have a very similar issue, so I try to add information here instead of opening a new issue. Problem appears on a ThinkPad T490 when applying "System Update" 0.1.73.

The usual system information first:

```
> fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9
```
Installed via dnf on Fedora 34.

Device information:
```
> fwupdmgr get-devices --show-all-devices
20N2CTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          934297aea936899e588ad4bf075cc02ee5b1272b
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              dd092983-daaf-587a-9ed7-dd8d178adbe7
│                         abc4cc28-099d-5220-b45d-129c21cd9da7
│                         52265728-359a-574c-9a6a-a23d3b1f952d
│                         f117e89e-a75f-5f33-9e8e-c4aded9cbadf
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:    0.1.19
│     Minimum Version:    0.1.19
│     Vendor:             DMI:LENOVO
│     GUIDs:              38ea6335-29ca-417b-8cd4-6b4e5e866f92
│                         78c6acda-f389-5111-839d-7f258cef874b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          b1c32d855d85a2124e865f0a03f8c2384dcca167
│     Current version:    192.72.1757
│     Minimum Version:    192.72.1757
│     Vendor:             DMI:LENOVO
│     GUIDs:              bf2d0f8b-f9a9-400c-8914-36c225d16eb4
│                         f8f599da-ffda-570a-9c05-841037bd487a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i7-8565U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000de
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         f6f850fc-448c-5b85-bda7-dc960c2ab162
[... 340 lines cut ...]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[***************************************]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: 
```

`efibootmgr` shows fwupd for BootNext. But it reboots straight to Fedora. 

When I run `get-devices` again, it displays

```
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
```

I checked that page but Boot Order Lock is turned off (while SecureBoot is turned on). And I'm pretty sure I updated successfully in the past and did not change UEFI settings.

I enabled fwupdate debugging and tried again. But no useful output:

```
> sudo fwupdate -L 
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

woozong, 2021-06-06:
I get the same error as @exploide (error opening file /sys/...)
When I do 
```
sudo fwupdate -d
failed: failed to write data to efivarfs: Error writing to file descriptor: Invalid argument```

This error about writing to efivars also occurs when doing `sudo fwupdmgr update`

I'm on Fedora 34 too by the way, same `fwupdmgr` version as above

ardevd, 2021-06-08:
Same issue here on my Thinkpad P1 Gen 3. Boot order lock is disabled.

superm1, 2021-06-08:
> I have a very similar issue, so I try to add information here instead of opening a new issue. Problem appears on a ThinkPad T490 when applying "System Update" 0.1.73.
>
> The usual system information [...]

For now I would suggest trying with secure boot off.  Recent shim has a problem (see https://github.com/rhboot/shim/pull/374 for details) and that might be playing into your problem specifically.

>failed: failed to write data to efivarfs: Error writing to file descriptor: Invalid argument

This looks like https://github.com/fwupd/firmware-lenovo-thinkpad/issues/128, you should probably join that issue.

>I'm really sorry - I have been away on leave and before I went I installed the latest bios using geteltorito to make the img file. I wish I had waited so we could have tried a few more things to help with resolving the issues. If the issue is still open when there is a new update available I will try fwupd again and report back here but of course that may be a while.

OK let's close this issue then.  Anyone new coming to this issue, open your own with your own circumstances.  There can be problems all over, so even if the behavior looks the same it could be lots of things.

closed 2021-06-08 by superm1