fwupd/fwupd #3306: How to get WD19TB to update

jdfoote, 2021-06-01:
**Describe the question**

I have been looking through some of the related issues and it sounds like after updating the firmware for my WD19TB I should just have to unplug the cable from my laptop and plug it back in. I've tried that a few times, and I still get this from
`$ sudo fwupdmgr get-devices`

```
├─WD19TB:                                                             
│ │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:             High performance dock            
│ │   Current version:     01.00.00.05                                                                                                      
│ │   Minimum Version:     01.00.00.00                                                                                                      
│ │   Vendor:              Dell Inc. (USB:0x413C)                                                                                           
│ │   Install Duration:    1 minute                                                                                                         
│ │   Serial Number:       75YFXW2/3184503723470071                                                                                         
│ │   Update Message:      Updates will complete next time the dock USB cable is unplugged.  The dock will be unavailable for approximately 
1 minute.                                                                                                                                   
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
```

Is there another step that I'm missing?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version

client version: 1.3.11
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.11

```

superm1, 2021-06-01:
What firmware are you trying to update to?  Can you show a whole test run of `fwupdmgr get-devices; fwupdmgr update; (unplug); fwupdmgr get-devices`

jdfoote, 2021-06-01:
Thanks so much!

I put the outputs in a few gists:

`get-devices`: https://gist.github.com/jdfoote/46bbef4f516a790b88a52a09ab627696
`update`: https://gist.github.com/jdfoote/1806668a06a457007aa60248487146ec
`get-devices (when unplugged)`: https://gist.github.com/jdfoote/0014b32d233f404528a655f2e4fd9f4f

superm1, 2021-06-01:
Looks like you're fully up to date to me.  Your versions match the latest on https://fwupd.org/lvfs/devices/com.dell.wd198ceeeffd.firmware

jdfoote, 2021-06-01:
Ah - when I saw the "Update Message" in `get-devices`, I thought that was an active message telling me that it wasn't updated yet.

superm1, 2021-06-01:
Ahhhh... If you upgrade to a newer version it will only show on the upgrade.

Specifically that was fixed in fwupd 1.5.0 to make it clearer: https://github.com/fwupd/fwupd/commit/c2721c8695ddb4bfe9d0dd9844d453528a4fbe39

closed 2021-06-01 by superm1