fwupd/fwupd #3307: Can't compile fwupd against efivar on FreeBSD

mkopec, 2021-06-01:
We (3mdeb) are working on a port of fwupd for FreeBSD. I've encountered an issue with linking fwupd against efivar:

```
ld: error: undefined symbol: efi_guid_zero
>>> referenced by fu-efivar-freebsd.c
>>>               libfwupdplugin/libfwupdplugin.so.2.0.0.p/fu-efivar-freebsd.c.o:(efi_guid_is_zero)

ld: error: undefined symbol: efi_variables_supported
>>> referenced by fu-efivar-freebsd.c
>>>               libfwupdplugin/libfwupdplugin.so.2.0.0.p/fu-efivar-freebsd.c.o:(fu_efivar_supported)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
*** Error code 1

Stop.
make: stopped in /usr/ports/sysutils/fwupd

```

The issue is very similar to rhboot/fwupdate#1, however the solution doesn't really work for use since we're using Meson. Any idea what can be done about this?

superm1, 2021-06-01:
Are you missing a library to link against in `library_deps` perhaps?

superm1, 2021-06-01:
Like it needs to be found in toplevel `meson.build` and then added to `library_deps` here: https://github.com/3mdeb/fwupd/pull/6/files#diff-9c6de0ed4ace5c7b1a7cdeb986ac1c64be57bb9dcf1d652fe5cdd7ad40485a26R136

mkopec, 2021-06-02:
I just tried adding the library to the toplevel `meson.build` and to `library_deps` in `libfwupdplugin/meson.build`, as you suggested. Now I'm left with just this error:

```
ld: error: undefined symbol: efi_guid_zero
>>> referenced by fu-efivar-freebsd.c
>>>               libfwupdplugin/libfwupdplugin.so.2.0.0.p/fu-efivar-freebsd.c.o:(efi_guid_is_zero)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
*** Error code 1
```

I've tried specifying which directory to search here: [https://github.com/3mdeb/fwupd/pull/6/commits/27611c928494f4b01663cee5cce47cc7e9f08b56](https://github.com/3mdeb/fwupd/pull/6/commits/27611c928494f4b01663cee5cce47cc7e9f08b56), but it didn't help, even though the `efivar.h` file in `/usr/local/include` `efi_guid_zero`.

mkopec, 2021-06-02:
Fixed by applying https://github.com/rhboot/efivar/commit/4af5b56066b4af66ce8d968948bb2bf4f5a87c79 manually on top of efivar provided by FreeBSD.

superm1, 2021-06-03:
So can this be closed now?  Or you'll have a PR soon?

closed 2021-06-03 by mkopec