fwupd/fwupd #3322: fwupdmgr sometimes fails to find system firmware devices on X1C7

bredelings, 2021-06-08:
**Describe the bug**
I found this bug because gnome-software was reporting that the system firmware needed to be upgraded, but then failing to update it saying that the device could not be found.  Indeed, running "fwupdmgr get-devices" only listed three devices, and nothing about the system firmware.  The weird thing though was that sometimes fwupdmgr _would_ list the system firmware devices, but typing the same command immediately afterware would _not_ list them.

**Steps to Reproduce**
Simply run `fwupdmgr get-devices` twice in a row.  Usually the system firmware would not get listed, but sometimes it would be listed the first time, but not the second time.

**Expected behavior**
The system firmware devices would always show up.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.7
```
I installed with apt.  So,
```shell
$ dpkg -s fwupd
Package: fwupd
Status: install ok installed
Priority: optional
Section: admin
Installed-Size: 5818
Maintainer: Debian EFI <debian-efi@lists.debian.org>
Architecture: amd64
Multi-Arch: foreign
Version: 1.5.7-4
Replaces: fwupdate (<< 12-7), gir1.2-dfu-1.0 (<< 0.9.7-1), libdfu-dev (<< 0.9.7-1), libdfu1 (<< 0.9.7-1)
Provides: fwupdate
Depends: libc6 (>= 2.17), libcurl3-gnutls (>= 7.63.0), libefiboot1 (>= 37), libelf1 (>= 0.142), libflashrom1, libfwupd2 (>= 1.5.6), libfwupdplugin1 (>= 1.5.7), libglib2.0-0 (>= 2.63.3), libgnutls30 (>= 3.7.0), libgudev-1.0-0 (>= 146), libgusb2 (>= 0.3.5), libjcat1 (>= 0.1.3), libjson-glib-1.0-0 (>= 1.5.2), libpolkit-gobject-1-0 (>= 0.99), libsmbios-c2, libsqlite3-0 (>= 3.6.1), libsystemd0, libtss2-esys-3.0.2-0 (>= 2.3.1), libxmlb1 (>= 0.1.13), shared-mime-info
Recommends: python3, bolt, dbus, secureboot-db, udisks2, fwupd-signed
Suggests: gir1.2-fwupd-2.0
Breaks: fwupdate (<< 12-7), gir1.2-dfu-1.0 (<< 0.9.7-1), libdfu-dev (<< 0.9.7-1), libdfu1 (<< 0.9.7-1)
Conflicts: fwupdate-amd64-signed, fwupdate-arm64-signed, fwupdate-armhf-signed, fwupdate-i386-signed
Conffiles:
 /etc/fwupd/daemon.conf 1ffb79beaaa4cf36e4833bdde2d638ad
 /etc/fwupd/redfish.conf 04489225dfb344254ff28fdb22f67026
 /etc/fwupd/remotes.d/dell-esrt.conf 71236e67e6fd095c771f9312cbae0382
 /etc/fwupd/remotes.d/lvfs-testing.conf d5980f0675d9c10937c71e8313463278
 /etc/fwupd/remotes.d/lvfs.conf 03115854971bef4f13b16729e9c107ad
 /etc/fwupd/remotes.d/vendor-directory.conf f65a59cfb87730fb523910f492dbceb6
 /etc/fwupd/remotes.d/vendor.conf b54e960f98c71d49af9421cf8e9cf7b9
 /etc/fwupd/thunderbolt.conf ecd9f08e4e9382b959e8604758ce71b3
 /etc/fwupd/uefi_capsule.conf 34e64ad37737ecebdb75106e9f855b2e
 /etc/fwupd/upower.conf 101e62f86b3fc2947723bbf6cd7f8e66
 /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata 2ace8894994764ae32e391bc0c68e398
 /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service 5abbdc42a8e3d9e60039ab658700ec71
 /etc/pki/fwupd-metadata/LVFS-CA.pem d9dff488bd9051268984da014f0be43d
 /etc/pki/fwupd/GPG-KEY-Hughski-Limited 885ac785ffef40dfcc4f8ea765fa07d4
 /etc/pki/fwupd/GPG-KEY-Linux-Foundation-Firmware de0622638572a2f6a6e29fa83f387eea
 /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service 5abbdc42a8e3d9e60039ab658700ec71
 /etc/pki/fwupd/LVFS-CA.pem d9dff488bd9051268984da014f0be43d
 /etc/update-motd.d/85-fwupd f8d4f2d80b40283ddce1445a5f0ffd35
 /etc/fwupd/ata.conf 20d356e1df3567a59967bfdf5b273521 obsolete
Description: Firmware update daemon
 fwupd is a daemon to allow session software to update device firmware.
 You can either use a GUI software manager like GNOME Software to view and
 apply updates, the command-line tool or the system D-Bus interface directly.
 Firmware updates are supported for a variety of technologies.
 See <https://github.com/fwupd/fwupd> for details
Homepage: https://github.com/fwupd/fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

[... 577 lines cut ...]
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-5.10.0-5-amd64
├── initrd.img-5.10.0-6-amd64
├── initrd.img-5.10.0-7-amd64
├── System.map-5.10.0-5-amd64
├── System.map-5.10.0-6-amd64
├── System.map-5.10.0-7-amd64
├── vmlinuz-5.10.0-5-amd64
├── vmlinuz-5.10.0-6-amd64
└── vmlinuz-5.10.0-7-amd64

```

**Additional questions**
- Operating system and version: Linux transposon 5.10.0-7-amd64 #1 SMP Debian 5.10.40-1 (2021-05-28) x86_64 GNU/Linux
- Have you tried rebooting? Yes.
- Is this a regression? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

bredelings, 2021-06-08:
Interestingly, it seems that running `systemctl restart fwupd.service` makes it work for a bit.  So, I was able to update my firmware when previously that was impossible.  But eventually it starts failing again.

hughsie, 2021-06-08:
Can you get a daemon log `sudo /someplace/on/debian/fwupd --verbose` and try to reproduce the missing devices with `fwupdmgr get-devices` in another terminal tab please? Thanks.

superm1, 2021-06-08:
I feel like we've had this come up recently, but I'm failing to find it.  My memory tells me it was only reproducing from systemd.  So you might be better to modify the systemd unit and add `--verbose` to it, and then look at the journal for that same log Richard mentioned.

joakim-tjernlund, 2021-06-10:
Doesn't systemd have some Idle timer than may shutdown/suspend fwupd? Perhaps the Idle handling has a bug?

bredelings, 2021-07-12:
OK, here are some logs.  I rebooted and ran `fwupdmgr update` 3 times, but the devices were only found the first time.
[fwupd.log](https://github.com/fwupd/fwupd/files/6800982/fwupd.log)

```
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
 • Intel Management Engine
 • System Firmware
 • UEFI Device Firmware
 • WDC PC SN730 SDBQNTY-512G-1001
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • WDC PC SN730 SDBQNTY-512G-1001
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • WDC PC SN730 SDBQNTY-512G-1001
```

superm1, 2021-07-28:
By chance did you unplug something after the first time?  Or change anything physical on the system?

I notice in your logs:
```
12 07:28:36 transposon fwupd[3547]: 11:28:36:0844 FuEngine             auto-removing GUsbDevice
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0844 FuDeviceList         ::removed f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0844 FuPlugin             backend_device_removed(dell_dock)
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0845 FuEngine             auto-removing GUsbDevice
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0845 FuDeviceList         ::removed 0ec6a2d51c2c55a649247d119ffb8e229168d6c9
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0845 FuPlugin             backend_device_removed(dell_dock)
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0845 FuEngine             auto-removing GUsbDevice
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0845 FuDeviceList         ::removed c085a3cbf4dc1d60d110b7616816e2ffd279b64e
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuPlugin             backend_device_removed(dell_dock)
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuEngine             auto-removing GUsbDevice
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuDeviceList         ::removed 9561e1f11a616405008effa52d976f04de1e3c7e
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuPlugin             backend_device_removed(dell_dock)
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuEngine             auto-removing GUsbDevice
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuDeviceList         ::removed f61a138e8240c473a88d703babf7be5bc0e098e6
Jul 12 07:28:36 transposon fwupd[3547]: 11:28:36:0846 FuPlugin             backend_device_removed(dell_dock)
```

Which `f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f` corresponds with your system firmware.  So definitely whatever triggered `auto-removing GUsbDevice` is what at least caused this.

superm1, 2021-07-28:
This code has changed both in 1_5_X and master, so it would be really useful to do the following:
1. Check out /compile 1_5_X (or use the 1.5.11 tarball).
2. Run the daemon with `FWUPD_PROBE_VERBOSE=1` in the environment
3. Reproduce this again
4. Check the logs before/after that time specifically.

superm1, 2021-07-28:
Specifically I think https://github.com/fwupd/fwupd/commit/253e6e4f1 might have fixed it (which is part of 1.5.8).

bredelings, 2021-07-30:
* I was able to compile the code -- I tried 1.5.12 and the master branch, but could use 1.5.11 if it matters
* meson says that you are using a feature from 0.5, but the minimum version required is 0.47 -- just a nit.
* If I run `ninja install`, it asks me for super-user privileges, despite the fact that I'm installing not installing into a system directory.  
* I can run the daemon in-place by doing (cd . ; $HOME/path/to/fwupd/build/src/fwupd --verbose )
* I stopped the fwupd systemd service
* However, the daemon quits with "another service has claimed the dbus name org.freedesktop.fwupd"

Help?

Is there a README or wiki page that has this information?

superm1, 2021-07-31:
>I was able to compile the code -- I tried 1.5.12 and the master branch, but could use 1.5.11 if it matters

1.5.12 is perfectly fine for this purpose.

>meson says that you are using a feature from 0.5, but the minimum version required is 0.47 -- just a nit.

don't worry about that message

>If I run ninja install, it asks me for super-user privileges, despite the fact that I'm installing not installing into a system directory.

I seem to recall this in 1.5.x timeframe, but I don't know if master this still happens.

> However, the daemon quits with "another service has claimed the dbus name org.freedesktop.fwupd"

See https://github.com/fwupd/fwupd/wiki/Compilation (specifically configure dbus to let the daemon start).  It's related to the location that is recognized for dbus services.

closed 2021-09-19 by stale[bot]