fwupd/fwupd #3336: Chrome OS CI failure due to mgi-docgen

djcampello, 2021-06-10:
Related to https://github.com/fwupd/fwupd/commit/1e17457b16c9a05f6f32799ceb3922b76b807994. Our first error caught by the brand new CI:
```
fwupd-9998: Run-time dependency [1mgi-docgen[0m found: [1;31mNO[0m (tried pkgconfig and cmake)
fwupd-9998: Looking for a fallback subproject for the dependency [1mgi-docgen[0m
fwupd-9998: 
fwupd-9998: meson.build:541:2: [1;31mERROR:[0m Automatic wrap-based subproject downloading is disabled
```

I imagine the portage ebuild would have a new dependency for this.

superm1, 2021-06-10:
What did the arguments for the meson setup command use?

djcampello, 2021-06-10:
``fwupd-9998: meson setup --buildtype plain --libdir lib64 --localstatedir /var/lib --prefix /usr --sysconfdir /etc --wrap-mode nodownload --build.pkg-config-path /usr/share/pkgconfig --pkg-config-path /build/amd64-generic/tmp/portage/sys-apps/fwupd-9998/temp/pkgconfig:/build/amd64-generic/tmp/portage/sys-apps/fwupd-9998/temp/python3.6/pkgconfig:/usr/share/pkgconfig --native-file /build/amd64-generic/tmp/portage/sys-apps/fwupd-9998/temp/meson.x86_64-pc-linux-gnu.amd64.ini --cross-file /build/amd64-generic/tmp/portage/sys-apps/fwupd-9998/temp/meson.x86_64-cros-linux-gnu.amd64.ini --localstatedir /var -Dagent=true -Dplugin_amt=false -Dlibarchive=false -Dbluez=true -Dplugin_dell=false -Dplugin_dummy=true -Delogind=false -Dplugin_flashrom=false -Dgnutls=true -Dgtkdoc=false -Dgusb=true -Dplugin_altos=true -Dman=false -Dplugin_nvme=false -Dintrospection=false -Dpolkit=false -Dplugin_synaptics_mst=true -Dplugin_synaptics_rmi=true -Dsystemd=false -Dtests=false -Dplugin_thunderbolt=true -Dplugin_uefi_capsule=false -Dplugin_uefi_pk=false -Dconsolekit=false -Dcurl=true -Dplugin_modem_manager=false -Dplugin_tpm=false /build/amd64-generic/tmp/portage/sys-apps/fwupd-9998/work/fwupd-9998 /build/amd64-generic/tmp/portage/sys-apps/fwupd-9998/work/fwupd-9998-build
``

djcampello, 2021-06-10:
Also here is the ebuild used: https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/HEAD/sys-apps/fwupd/fwupd-9998.ebuild

hughsie, 2021-06-10:
@djcampello can you change `-Dgtkdoc=false` to `-Ddocs=none` please. Sorry about that.

superm1, 2021-06-15:
(@Whissi you might need to update the Gentoo ebuild that ChromeOS bases on here if you haven't already)

djcampello, 2021-06-15:
crrev.com/c/2952952 fixed the issue. Thanks!

closed 2021-06-15 by djcampello