fwupd/fwupd #3354: Missing source dist tarballs

kloczek, 2021-06-14:
On https://fwupd.org/lvfs/docs/developers there is no details from where is possible to download source dist tar balls with integrated subprojets. As well here is onlu (in last tag) fwupd-1.6.1-setup-x86_64.exe.
It would be good to add dist tar balls as github autogenerated from git tag tar ball is useless as it does not contain subprojects.

superm1, 2021-06-14:
Its actually intentional that we only have GitHub generated tarballs there.

Can you elaborate more on why you want them with subprojected content? The only new thing is gidocgen which we're not treating as something in distros yet, should be online documentation only until the first stable release of the tools.

kloczek, 2021-07-12:
OK I just realized that fwupd-efi has been moved to standalone project. After package it I was able to build fwupd.
In other words it was my fault :P

closed 2021-07-12 by kloczek