fwupd/fwupd #3356: fwupd changes to support USB4 retimer firmware update when no devices are attached

RajmohanMani, 2021-06-15:
New kernel changes to support USB4 retimer firmware has merged in Mika's USB4 next branch as below.
https://git.kernel.org/pub/scm/linux/kernel/git/westeri/thunderbolt.git/log/?h=next

fwupd needs to be updated to add support for USB4 retimer fw update using the latest kernel changes.

fwupd should be modified to make use of the new USB4 retimer firmware update kernle changes, when no devices are attached.

More context on how the kernel changes can be used by user space to enable USB4 retimer enumeration (from Mika):

   # echo 1 > 0-0/usb4_port1/offline
   # echo 1 > 0-0/usb4_port1/rescan
   # dd if=new_nvm.bin of=0-0:1.1/nvm_nonactiveX/nvmem
   # echo 2 > 0-0:1.1/nvm_authenticate
   # echo 0 > 0-0/usb4_port1/offline

Now the port is functional again. Whenever user logs out the upgrade app can do this:

   # echo 1 > 0-0/usb4_port1/offline
   # echo 1 > 0-0/usb4_port1/rescan
   # echo 3 > 0-0:1.1/nvm_authenticate
   # echo 0 > 0-0/usb4_port1/offline

and it runs the authentication only."

This feature tracks the work that need to be done in fwupd/fwupdtool and related UI scripts, so that USB4 retimer firmware update is functional with the upstream solution.

superm1, 2021-06-15:
>when no devices are attached

How do you determine no devices attached and this flow is needed?

>echo 1 > 0-0/usb4_port1/offline

Will this take the port offline?  Does that mean the device plugged in during this time would be non-functional?  What if a device was plugged in when calling this?

superm1, 2021-06-15:
>How do you determine no devices attached and this flow is needed?

Seem to me that it's `link attribute returns "none"`.  This sounds relatively straightforward now.  Both during enumeration and writing though the check will need to be performed since a user can start with nothing plugged in, or start with something plugged in and later change it.

RajmohanMani, 2021-06-15:
Yes Mario.
The link sysfs entry should return "none" as the status. Only then, "offline" mode can be entered for that port.
This status needs to be checked, just before trying to do a retimer firmware update on that USB4 port.

superm1, 2021-06-15:
Does kernel automatically do this and check for firmware version when driver loads? Or will it be on demand?  I'm wondering during initial query from fwupd if value is cached by kernel or not.

RajmohanMani, 2021-06-15:
It will be on demand. By default, no sysfs entries will appear, if there are no devices attached.

Per https://git.kernel.org/pub/scm/linux/kernel/git/westeri/thunderbolt.git/commit/?h=next&id=cae5f5151d76635f6b5c08133184c48048346e63, the initial query gives the latest link status that the kernel is aware of.

superm1, 2021-06-15:
Then that means fwupd needs to coldplug too. The daemon/tool will check at startup and cache it so that any client can request upgrades on demand.

RajmohanMani, 2021-06-15:
Caching is fine for coldplug, but it needs to check the link status again when a client requests an upgrade, as the cached status would be stale as time goes (since the user might unplug or unplug and replug another type of device on an USB4 port).

superm1, 2021-06-15:
Right, completely agree.  Will you do the PR for this new stuff?

hughsie, 2021-06-15:
Someone with hardware will have to do the code, we can't really do this from theory alone. I'm also guessing there is a uevent when the link file changes?

bleungatchromium, 2021-06-17:
@djcampello and I have hardware that has these retimers, fyi.

RajmohanMani, 2021-06-22:
/cc @westeri

RajmohanMani, 2021-07-08:
/cc @arao15

hughsie, 2021-08-03:
Is anyone going to propose a PR for this?

arao15, 2021-08-04:
Hi Richard, 

I have some basic question to understand the work and scope, pls help.

As I understand now we tbt driver side changes are present, I want to understand who needs to do the changes in fwupd tool. 
Is there a dependency of HW that is delaying this work ? What is meant by PR :). kindly help to share more details that will help me to understand the work needed from our side

hughsie, 2021-08-04:
> Is there a dependency of HW that is delaying this work

I don't think either Mario or I have any hardware that's capable of doing TBT4. I have a TBT4 dock from Lenovo, but nothing to actually drive it at that speed...

> What is meant by PR

Sorry, a pull request. We already have a https://github.com/fwupd/fwupd/tree/master/plugins/thunderbolt plugin which can update TBT2/3 controllers and devices, and that will need changes for the TBT4 host I think.

superm1, 2021-08-05:
Specifically a host that will support doing the retimer upgrade with this interface. If Intel or Google can please do the code changes and validate we can review it.

arao15, 2021-08-05:
to expedite the work here I can propose. 
1. if you can make changes, I can help to validate the check
or
2. Since this is new to me would require little more hand holding details like; cloning this repo, build and deploy steps.
3. Will this require ubuntu to check or we need to check on Chrome.

superm1, 2021-08-09:
@arao15 

Happy to handhold along the way for development.

Here's some instructions of how to do it from Ubuntu/Debian/Fedora using master:
```
# git clone https://github.com/fwupd/fwupd
# ./contrib/setup
```

[Building](https://github.com/fwupd/fwupd/wiki/Compilation)
# meson build
# ninja -C build
```

If you want to do it from Chrome, I would suggest you follow https://chromium.googlesource.com/chromiumos/docs/+/master/developer_guide.md to work on the fwupd package there and come up with your patch.  When you have it working you can then add a remote target of upstream and cherry-pick your patch.
@bleungatchromium @djcampello can probably help you a lot more closely in this regard.

RajmohanMani, 2021-08-09:
/cc @krankun

superm1, 2021-08-12:
@djcampello can help you with build environment for chrome - that command is specific to getting build environment for Ubuntu ready.

Can you show the log of all 3 commands on Ubuntu?
```
./contrib/ci/generate_dependencies.py -o ubuntu | sudo xargs apt install -y
./contrib/setup
meson build
```

krankun, 2021-08-12:
[log1.txt](https://github.com/fwupd/fwupd/files/6978341/log1.txt)
[log2.txt](https://github.com/fwupd/fwupd/files/6978342/log2.txt)
[log3.txt](https://github.com/fwupd/fwupd/files/6978343/log3.txt)
logs from ubuntu

superm1, 2021-08-12:
> meson.build:554:4: ERROR: Problem encountered: python3-markdown version 3.3.3 required for gi-docgen

So you need to either install a newer `python3-markdown` from pip, or turn off docs.  I would just say turn off docs.
Try this instead:
```
meson build -Ddocs=none
```

krankun, 2021-08-12:
turn off docs is helping , Thanks!

krankun, 2021-08-17:
on chrome env compiling fwupd gives below error
"meson.build:205:2: ERROR: Dependency "gudev-1.0" not found"

ubuntu seems to have the package :
sudo ldconfig -p | grep -i gudev
	libgudev-1.0.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libgudev-1.0.so.0
	libgudev-1.0.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libgudev-1.0.so

@djcampello  have you seen this issue on chrome env  , how to install this package on cros_sdk for chrome ? Thank you!

djcampello, 2021-08-17:
To compile fwupd in chrome you want to follow the steps in https://chromium.googlesource.com/chromiumos/docs/+/master/developer_guide.md 

At some point you would have created a board and then you can do `emerge-${BOARD} fwupd` to compile the package with its appropriate dependencies handled by portage

djcampello, 2021-08-17:
Specifically around this point in the guide: https://chromium.googlesource.com/chromiumos/docs/+/master/developer_guide.md#initialize-the-build-for-a-board

krankun, 2021-08-17:
@djcampello  I was able to successfully build "emerge-${BOARD} fwupd" on chrome.
is there a way to just install fwupdtool from my local build with out building OS ? 
or do I need to build OS image with below every time to test changes in tool ? thank you!
./build_packages --board=${BOARD} 
and 
./build_image --board=${BOARD} --noenable_rootfs_verification test

djcampello, 2021-08-17:
`cros deploy <ip_address> fwupd`

djcampello, 2021-09-24:
> Then that means fwupd needs to coldplug too. The daemon/tool will check at startup and cache it so that any client can request upgrades on demand.

Could you expand how do we handle this a coldplug vs the current udev hotplug of retimers?

superm1, 2021-09-24:
>Could you expand how do we handle this a coldplug vs the current udev hotplug of retimers?

It doesn't need to be a "real" coldplug, but the event of a parent coming in is enough of an event to do it.  Something along the lines of:
1. Parent event shows up.
2. Create parent
2. Sees the sysfs node that lets it do a NDA routine.
3. Add children, set some flags so they don't disappear and need to be "woken" up during updates next time
4. Cleanup NDA routine.

djcampello, 2021-09-24:
> 1. Parent event shows up.
This can be the host controller showing up in udev?
> 2. Create parent
> 3. Sees the sysfs node that lets it do a NDA routine.
> 4. Add children, set some flags so they don't disappear and need to be "woken" up during updates next time
How do we create this children? Are these the retimer devices themselves? While processing the parent the udev event for the retimers showing up wont be processed. How do we set up this flag? Udev remove event trigger the removal of the fu-device.
> 5. Cleanup NDA routine.

superm1, 2021-09-24:
>This can be the host controller showing up in udev?

Yeah I think so.

>How do we create this children? Are these the retimer devices themselves? While processing the parent the udev event for the retimers showing up wont be processed. How do we set up this flag? Udev remove event trigger the removal of the fu-device.

I think that means you'll have to go in NDA mode and let the parent device "finish" probing.
They retimers up as events right after that?  If so I think set a flag in the plugin that it's in NDA mode, and then those can get picked up from their own uevents.

The only uncertainty to me is how do you know it's time to cleanup?

djcampello, 2021-09-24:
Worse case scenario, we do it on a timer after a while to give time to the retimers to show up?

Is there any example of this "flag" to avoid removal when the retimers de-enumerate on cleanup?

superm1, 2021-09-24:
We used to do this a while back with force power: https://github.com/fwupd/fwupd/commit/6358e23490acde05f1ca65c714d141bb8f6497ae but we've modernized and that's not really the design I would suggest now.

I think what I would say you should do is set a bit in global plugin data.  When the remove events come in for the retimers, check if that bit was set and just don't remove them.  Instead apply a private device flag saying they need to be woken up manually.

>Worse case scenario, we do it on a timer after a while to give time to the retimers to show up?

I'm not crazy about a timer since it will break things like using a debugger to find and reproduce problems.  Is there any other event we can look for?  Or maybe we can have a quirk to indicate the number of retimers we should expect to see?

djcampello, 2021-09-30:
It looks to me that https://github.com/fwupd/fwupd/commit/6358e23490acde05f1ca65c714d141bb8f6497ae is exactly what we want to implement this feature but instead of being Dell specific it will be generic using the new sysfs kernel interface. The part that I did not get from that commit is how the retimers are coldplug I suppose that was handled by the thunderbolt coldplug and got removed in a separate change

djcampello, 2021-09-30:
It seems it was https://github.com/fwupd/fwupd/commit/ec44dd335333402087a780741a0ee1c4b88ff3c0 who left the handling of the udev removal to the framework... We would have to alter the framework to skip the removal of the device based on a device flag

superm1, 2021-09-30:
>It looks to me that 6358e23 is exactly what we want to implement this feature but instead of being Dell specific it will be generic using the new sysfs kernel interface

Well it was specific to how TBT3 worked in the UEFI ecosystem before runtime PM was supported.

>The part that I did not get from that commit is how the retimers are coldplug I suppose that was handled by the thunderbolt coldplug and got removed in a separate change

Those designs didn't have dedicated retimers.  They were discrete TBT controllers physically located in a certain proximity to the port.

>It seems it was ec44dd3 who left the handling of the udev removal to the framework... We would have to alter the framework to skip the removal of the device based on a device flag

Yup agree.

hughsie, 2021-10-01:
> We would have to alter the framework to skip the removal of the device based on a device flag

Sounds sane to me.

krankun, 2021-10-06:
@superm1  @djcampello 
Could you please outline the steps to reproduce the issue , probably that will help me in understanding the example patches
( 6358e23 )posted. Thank you!

superm1, 2021-10-13:
Can you please submit them upstream? They should be reviewed here.

krankun, 2021-10-18:
auto update usecase that is the fw update triggered at the boot time is still not working with above patches:

After following all the steps I can trigger auto update with below 
localhost ~ # echo 1 > /sys/bus/thunderbolt/devices/0-0/usb4_port1/offline 
localhost ~ # echo 1 > /sys/bus/thunderbolt/devices/0-0/usb4_port1/rescan  
localhost ~ # reboot

I can see a splash screen with FW update however the NVM is not updated and still showing the old nvm version 

The thunderbolt kernel driver logs have this latest info:
[    7.842171] thunderbolt 0000:00:0d.2: 0:1: ACPI: calling _DSM to power on retimers
[    9.755630] thunderbolt 0000:00:0d.2: 0:1: enter offline mode

how to get log on the fwupd side to check where this is failing at the boot time ?

CC: @superm1  @djcampello

superm1, 2022-07-15:
@CragW pointed out this should have been closed when https://github.com/fwupd/fwupd/pull/4046 was merged.