fwupd/fwupd #3357: fwupdmgr-1.6.1: fwupd update: /usr/libexec/fwupd/efi/fwupdx64.efi cannot be found

m1027, 2021-06-15:
I believe this must be something trivial...

To reproduce:

- use fwupd-1.6.1 (on Gentoo here)
- fwupdmgr refresh
- fwupdmgr update

Result:
```
Devices with no available firmware updates:
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.80.3746 to 184.83.3874?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2020.2 HF2 Release                                 ║
║ Version 11.8.83.3874 (LVFS: 184.83.3874)                                     ║
║ Problem Fixes                                                                ║
║ • Mitigated several security vulnerabilities.  Please see details under      ║
║ "Fixed Issues".                                                              ║
║ 20KFCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
/usr/libexec/fwupd/efi/fwupdx64.efi cannot be found
```

Expected behaviour:

- update Intel ME

Some notes:

- This error message is new since 1.6.1.
- The file /usr/libexec/fwupd/efi/fwupdx64.efi is indeed nowhere on the system.
- A fresh reinstall, after clearing all related files and caches did not help.
- gcap: 1.4
- In the past I had success using fwupdmgr, using older versions <1.5.9.

hughsie, 2021-06-15:
Is fwupd-efi installed? Did you also disable the EFI binary subproject when building fwupd?

m1027, 2021-06-15:
Thanks for the hint. Indeed I believe it is not the case yet.

Here is the according bug report: https://bugs.gentoo.org/791307.

If you are sure that's the reason feel free to close this issue.

closed 2021-06-15 by hughsie