fwupd/fwupd #3361: FAILED: plugins/uefi-capsule/uefi-capsule-ux.tar.xz: Pango:ERROR:../pango-1.48.5/pango/pangofc-fontmap.c:2750:pango_fc_font_description_from_pattern: assertion failed: (res == FcResultMatch)

Whissi, 2021-06-15:
**Describe the bug**
Sometimes building will fail with
```
[248/248] /var/tmp/portage/sys-apps/fwupd-1.6.1/temp/python3.9/bin/python3 /var/tmp/portage/sys-apps/fwupd-1.6.1/work/fwupd-1.6.1/plugins/uefi-capsule/make-images.py --podir /var/tmp/portage/sys-apps/fwupd-1.6.1/work/fwupd-1.6.1/po --label 'Installing firmware update…' --out plugins/uefi-capsule/uefi-capsule-ux.tar.xz
FAILED: plugins/uefi-capsule/uefi-capsule-ux.tar.xz
/var/tmp/portage/sys-apps/fwupd-1.6.1/temp/python3.9/bin/python3 /var/tmp/portage/sys-apps/fwupd-1.6.1/work/fwupd-1.6.1/plugins/uefi-capsule/make-images.py --podir /var/tmp/portage/sys-apps/fwupd-1.6.1/work/fwupd-1.6.1/po --label 'Installing firmware update…' --out plugins/uefi-capsule/uefi-capsule-ux.tar.xz
**
Pango:ERROR:../pango-1.48.5/pango/pangofc-fontmap.c:2750:pango_fc_font_description_from_pattern: assertion failed: (res == FcResultMatch)
Bail out! Pango:ERROR:../pango-1.48.5/pango/pangofc-fontmap.c:2750:pango_fc_font_description_from_pattern: assertion failed: (res == FcResultMatch)
ninja: build stopped: subcommand failed.
```

When I re-run build it will pass. Looks like a race/parallel build issue?

[build.log](https://github.com/fwupd/fwupd/files/6655132/build.log)

hughsie, 2021-06-15:
> Looks like a race/parallel build issue

If you build with `ninja -j1` do you see the same thing? If so, I would guess it's because we can't find a suitable font.

Whissi, 2021-06-15:
Yes, I just hit this with `-j1`.

superm1, 2021-06-15:
Was this a parallel emerge though perhaps?  Is it possible that fonts or pango weren't fully set up yet by the time fwupd was built?

Whissi, 2021-06-15:
No, the depgraph is complete, all requirements are fully set up. Only fwupd build process is running. 10 attempts, 6x successful, 4x I am hitting this error.

hughsie, 2021-06-15:
Could you try installing more fonts?

Whissi, 2021-06-15:
Are you really asking to install font packages like [media-fonts/open-sans](https://packages.gentoo.org/packages/media-fonts/open-sans) or [media-fonts/roboto](https://packages.gentoo.org/packages/media-fonts/roboto) which are just installing files below `/usr/share/fonts/`?

hughsie, 2021-06-16:
I think so, yes. This isn't an error we've ever seen before, so it might be a wild goose chase.

Whissi, 2021-06-16:
OK, I tested a few things today:

- With just [media-fonts/liberation-fonts](https://packages.gentoo.org/packages/media-fonts/liberation-fonts) which were already installed, it's failing a lot.
- When I add open-sans, failure rate is not really changing.
- When I add roboto in addition, I didn't hit an error in 50 attempts
- Just roboto and liberation failed within 20 attempts

I am unsure what to do with these results. Don't forget that it is even passing 50% with just liberation font.

closed 2021-07-28 by stale[bot]