fwupd/fwupd #3364: Synaptics prometheus can't ensure ID

superm1, 2021-06-15:
**Describe the bug**
Lenovo P14s (AMD) w/ synaptics prometheus emits the following errors:

```
15:35:41:0110 FuDevice             cannot change (null) physical ID from (null) to usb:06:00:03 as FuDevice->setup() has already completed
15:35:41:0110 FuDevice             failed to ensure child: cannot ensure ID: FuSynapromConfig:
Prometheus IOTA Config
  CompositeId:          50efef912094e784a03628350441f6927c4920b0
  Guid:                 7c5a1e62-38fa-5859-9337-09dbac6377e4 <- USB\VID_06CB&PID_00BD-cfg
  Guid:                 21a62213-7012-552c-9f1a-a51d07c87242 <- USB\VID_06CB&PID_00BD&CFG1_3462&CFG2_0
  Protocol:             com.synaptics.prometheus.config
  Flags:                updatable
  Vendor:               Synaptics
  VendorId:             USB:0x06CB
  Version:              0022
  VersionLowest:        0022
  VersionFormat:        plain
  Icon:                 touchpad-disabled
  LogicalId:            cfg
```

**Steps to Reproduce**
w/ master
```
sudo fwupdtool get-devices --plugins synaptics-prometheus
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
1.6.1-12-gbda57e8c

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source
[promtheus.txt](https://github.com/fwupd/fwupd/files/6656901/promtheus.txt)

superm1, 2021-06-15:
CC @mrhpearson to this one.

hughsie, 2021-06-15:
I can reproduce on the device-test hardware I have here. Debugging.

closed 2021-06-15 by superm1