fwupd/fwupd #3366: Lenovo TBT3 Gen2 dock probe errors

superm1, 2021-06-15:
**Describe the bug**
Notice the following emitted from daemon.  This dock is connected to non-TBT3 host (Lenovo P14s AMD).
```
16:33:31:0138 FuPluginVliUsbhub    cannot create MSP430 I?C device: failed to read versions: failed to read I2C: endpoint stalled or request not supported
16:33:31:0183 FuEngine             failed to add device usb:06:00:01:05: failed to get version: endpoint stalled or request not supported
```

**Steps to Reproduce**
```
sudo fwupdtool get-devices --plugins vli
```

**Expected behavior**
No errors

**fwupd version information**
```
1.6.1-g12
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):source

[vli.txt](https://github.com/fwupd/fwupd/files/6657329/vli.txt)

hughsie, 2021-06-15:
I think that's by design; IIUC the MSP device enumerates find when you do it using the USB3 side of the VLI chip, but not on the USB-2 side.

superm1, 2021-06-15:
Then shouldn't the errors downgrade? They're loud and error level..

hughsie, 2021-06-15:
Hmm good question. Will investigate whether we can use a more specific GUID in the morning.

hughsie, 2021-06-17:
We can, but there are a lot of other things to fix before it works :)

closed 2021-06-17 by hughsie