fwupd/fwupd #3384: purpose of upower.conf

binaryNewts, 2021-06-17:
In the most recent version of upower.conf, line 5, `#BatteryThreshold=10`, is commented out. I was wondering what the purpose of this file is without this line. What does this file do in the current version, or should it be removed?

( @bleungatchromium @g2pop @djcampello )

superm1, 2021-06-17:
Since you're CC chrome people it's worth mentioning - chrome doesn't use upower though.

bleungatchromium, 2021-06-17:
> Since you're CC chrome people it's worth mentioning - chrome doesn't use upower though.

We know. :)

I'd like to introduce my interns for the summer, @binaryNewts  and @g2pop . They're going to add a new plugin to support the power manager daemon we use on Chrome OS.

hughsie, 2021-06-17:
In that context, I think it makes sense to move the data in upower.quirk into some kind of common place. We don't need to duplicate that.

superm1, 2021-06-17:
Nice to meet y'all.

Agree Richard.

Maybe `daemon.conf`?

hughsie, 2021-06-17:
.quirk, but yes.

commit 931f6388 (2021-06-17): trivial: drop support for upower.conf (Fixes: #3384)

All overrides should be performed using `power.quirk`
commit 0ed3efa7 (2021-06-18): trivial: drop support for upower.conf (Fixes: #3384)

All overrides should be performed using `power.quirk`
commit 7dd2b4e7 (2021-06-18): trivial: drop support for upower.conf (Fixes: #3384)

All overrides should be performed using `power.quirk`
commit 6f831903 (2021-06-18): trivial: drop support for upower.conf (Fixes: #3384)

All overrides should be performed using `power.quirk`
closed 2021-06-18 by hughsie