fwupd/fwupd #3386: upower plugin / FWUPD_DEVICE_FLAG_REQUIRE_AC functionality : Support explicit Mains, or USB* power supply types as an indication of AC present

bleungatchromium, 2021-06-17:
**Is your feature request related to a problem? Please describe.**
The upower plugin uses the system's battery state to determine whether the system is on AC power. If the battery state is FU_BATTERY_STATE_DISCHARGING, it sets an error that we are not on AC power.

`
        /* not charging */                                                      
        if (fu_context_get_battery_state (ctx) == FU_BATTERY_STATE_DISCHARGING ||
            fu_context_get_battery_state (ctx) == FU_BATTERY_STATE_EMPTY) {     
                g_set_error_literal (error,                                     
                                     FWUPD_ERROR,                               
                                     FWUPD_ERROR_AC_POWER_REQUIRED,             
                                     "Cannot install update "                   
                                     "when not on AC power unless forced");     
                return FALSE;                                                   
        }        
`

There are some systems that, even while attached to some sort of AC mains, will decide to allow the battery to go into a DISCHARGING state (either for battery health, or some other sort of system policy, or as a transient condition when ac is attached, and the battery object hasn't been polled for the new state yet, or possibly some combination of a weaker charger + high system consumption could result in DISCHARGING state).

This could lead to some cases where the AC power error is set while the AC is actually plugged in. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The power supply subsystem has different device types, including non-battery ones Mains, USB, etc. Instead of determining whether the system is on AC power by inference (ie, the battery is not discharging) we should use those more reliable power supplies, which gives a binary online = 1 or online = 0.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

See the comment in chromiumos's power supply properties.proto which addresses this possibility: https://chromium.googlesource.com/chromiumos/platform/system_api/+/HEAD/dbus/power_manager/power_supply_properties.proto#13

`
// For any of these power sources, the system may be consuming power at a high
// enough rate that the battery is discharging rather than charging; see
// BatteryState. See the PowerSource submessage's \|type\| field for more
// details about the active power source.
`

I'm unfamiliar with how upower presents other power supply information through dbus, so I don't know how easy it would be to add this in the upower plugin, but for the new chromiumos power manager plugin being worked on by @binaryNewts and @g2pop, this is something to consider. 

@djcampello as a fyi.

superm1, 2021-06-17:
> There are some systems that, even while attached to some sort of AC mains, will decide to allow the battery to go into a DISCHARGING state (either for battery health, or some other sort of system [...]

You know it's interesting you mention this. I remember raising the same point at the time we added this code, and we ended up with the attitude nahhhhh that's a major corner case.

Something I'm thinking perhaps we've overloaded `FWUPD_DEVICE_FLAG_REQUIRE_AC` and it should really split into multiple flags.  Right now when that is set it translates that we need "AC" power or we need a minimum battery threshold.  Perhaps we should instead explicitly make a flag for minimum battery threshold.

hughsie, 2021-08-03:
I think we fixed this with 1.6.2.

bleungatchromium, 2021-08-03:
Hey @hughsie,

I think this issue still remains. For both the upower and powerd  plugins, we are still using not "DISCHARGING" of a system's battery as a proxy for the AC status.

For 1.6.2, we moved around some of the logic for this into the fu-engine (since we have two plugins that do this job now, and wanted to consolidate), but the logic is still the same.

The proposal here is to allow one or both of these plugins to actually describe the "online" status of the Mains (or USB*) object, which tells us with a binary whether we're plugged in or not. We still haven't done that plumbing yet, as it would require additional changes in upower and powerd daemons.

hughsie, 2022-01-19:
If it helps, I think you'd want to copy the API design of `FuLidState` i.e. `fu_context_set_lid_state`

ccgauvin94, 2022-07-24:
I just ran into this issue with a USB-PD charger as well. Had to drop the battery to 95% and then install went fine. That's not a big deal, but figuring out it was the issue was the problem.

closed 2023-02-21 by hughsie