fwupd/fwupd #3389: fwupd and Ubuntu 20.04 guest on Hyper-V 2016

kapela86, 2021-06-18:
I don't know if this is related only to Ubuntu, but we have Cisco UCSC-C220-M5SX with Windows Server 2016 and Hyper-V role. I just installed Ubuntu 20.04 as it's guest and I noticed this in syslog
```
Jun 18 01:47:11 srvprod fwupd[1793]: 23:47:11:0967 FuPluginUefi         SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: System does not support UEFI mode
Jun 18 01:47:11 srvprod fwupd[1793]: 23:47:11:0975 FuPluginUefi         Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory
```
Guest is Gen 2 virtual machine installed in UEFI mode
```
[ -d /sys/firmware/efi ] && echo "Installed in UEFI mode" || echo "Installed in Legacy mode"
Installed in UEFI mode
```
First issue here is that it doesn't recognize that it's running in UEFI mode
Second issue, well, not really an issue but a question. Is fwupd needed in this situation and can I safely remove it since it will be running only as Hyper-V guest.

**Steps to Reproduce**
Install Ubuntu 20.04 as guest on Hyper-V 2016 Windows Server

**fwupd version information**
```
client version: 1.3.11
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed with Ubuntu 20.04

**fwupd device information**
```
Virtual Machine
│
└─System Firmware:
      Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
      Current version:     Hyper-V UEFI Release v1.0
      Vendor:              Microsoft Corporation
      Update Error:        UEFI Capsule updates not available or enabled
      GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
      Device Flags:        • Internal device
                           • Requires AC power
                           • Needs a reboot after installation

```

**System UEFI configuration**
```
BootCurrent: 0005
BootOrder: 0004,0005,0001,0002,0003,0000
Boot0000* EFI Network   AcpiEx(VMBus,,)/VenHw(9b17e5a2-0891-42dd-b653-80b5c22809ba,635161f83edfc546913ff2d2f965ed0e5b76ee35e3c92843aab7f32df6e4bbab)/MAC(000000000000,0)
Boot0001* EFI SCSI Device       AcpiEx(VMBus,,)/VenHw(9b17e5a2-0891-42dd-b653-80b5c22809ba,d96361baa104294db60572e2ffb1dc7f916074e90e548a4b86b7b94272b81daf)/SCSI(0,0)
Boot0002* EFI SCSI Device       AcpiEx(VMBus,,)/VenHw(9b17e5a2-0891-42dd-b653-80b5c22809ba,d96361baa104294db60572e2ffb1dc7f916074e90e548a4b86b7b94272b81daf)/SCSI(0,1)
Boot0003* EFI SCSI Device       AcpiEx(VMBus,,)/VenHw(9b17e5a2-0891-42dd-b653-80b5c22809ba,d96361baa104294db60572e2ffb1dc7f916074e90e548a4b86b7b94272b81daf)/SCSI(0,2)
Boot0004* EFI SCSI Device       AcpiEx(VMBus,,)/VenHw(9b17e5a2-0891-42dd-b653-80b5c22809ba,d96361baa104294db60572e2ffb1dc7f916074e90e548a4b86b7b94272b81daf)/SCSI(0,3)
Boot0005* ubuntu        HD(1,GPT,8534a9e3-7d2f-4f51-a02b-1e79f3bd63eb,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
```

efivar -l | grep fw
Doesn't return anything, without grep it's this:
```
fab7e9e1-39dd-4f2b-8408-e20e906cb6de-HDDP
f0a30bc7-af08-4556-99c4-001009c93a44-SecureBootEnable
eb704011-1402-11d3-8e77-00a0c969723b-MTC
e20939be-32d4-41be-a150-897f85d49829-MemoryOverwriteRequestControl
d719b2cb-3d3a-4596-a3bc-dad00e67656f-dbx
d719b2cb-3d3a-4596-a3bc-dad00e67656f-db
8be4df61-93ca-11d2-aa0d-00e098032b8c-SetupMode
8be4df61-93ca-11d2-aa0d-00e098032b8c-SecureBoot
8be4df61-93ca-11d2-aa0d-00e098032b8c-PlatformLang
8be4df61-93ca-11d2-aa0d-00e098032b8c-PK
8be4df61-93ca-11d2-aa0d-00e098032b8c-Lang
[... 317 lines cut ...]
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-5.4.0-73-generic
├── initrd.img-5.4.0-73-generic
├── initrd.img.old -> initrd.img-5.4.0-73-generic
├── System.map-5.4.0-73-generic
├── vmlinuz -> vmlinuz-5.4.0-73-generic
├── vmlinuz-5.4.0-73-generic
└── vmlinuz.old -> vmlinuz-5.4.0-73-generic

7 directories, 297 files
```

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried rebooting? Yes
- Is this a regression? Don't know
- Are you using an NVMe disk? No
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2021-06-18:
> can I safely remove it since it will be running only as Hyper-V guest

Yes, I think it's safe to remove.

superm1, 2021-06-18:
It could be https://github.com/fwupd/fwupd/commit/4558e5f317fa16577eb646f7012aaf52e4bbf59b is not in your local fwupd version and that fixes the uefi plugin from running.

But as mentioned, this doesn't seem to be important for a VM use case anyway.

closed 2021-06-18 by superm1