fwupd/fwupd #3395: Lenovo T490, Ubuntu 21.04, Boot Order Lock disabled: "failed to run update on reboot"

kees, 2021-06-18:
**Describe the bug**
It seems UEFI updates are never running. Even if I manually select Linux-Firmware-Updater, normal boot continues. I'm at a loss for where to get further diagnostic details.

**Steps to Reproduce**
```shell
$ sudo fwupdmgr upgrade
$ sudo shutdown -r now
```
**Expected behavior**
Firmware updates run.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.5.8
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.8
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt:
```sh
$ dpkg -l fwupd | grep ^.i
ii  fwupd          1.5.8-0ubuntu1 amd64        Firmware update daemon
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```sh
$ fwupdmgr get-devices --show-all-devices
20N2001YUS
│
├─Thunderbolt host controller:
│     Device ID:          a804dbcfab80c29f688d74aa71a45052be0d2d73
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              dd092983-daaf-587a-9ed7-dd8d178adbe7
│                         abc4cc28-099d-5220-b45d-129c21cd9da7
│                         52265728-359a-574c-9a6a-a23d3b1f952d
│                         f117e89e-a75f-5f33-9e8e-c4aded9cbadf
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:    0.1.23
│     Minimum Version:    0.1.23
│     Vendor:             DMI:LENOVO
│     GUIDs:              38ea6335-29ca-417b-8cd4-6b4e5e866f92
│                         78c6acda-f389-5111-839d-7f258cef874b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          b1c32d855d85a2124e865f0a03f8c2384dcca167
│     Current version:    192.72.1757
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
[... 255 lines cut ...]

**Additional questions**
- Operating system and version:
```sh
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	21.04
Codename:	hirsute
```
- Have you tried rebooting?
A lot. ;)

- Is this a regression?
Yes, but I'm not sure when it started. I'm at least two firmware versions behind currently.

- Are you using an NVMe disk?
Yes.

- Is secure boot enabled?
Yes

- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
It is a Lenovo, but "Boot Order Lock" is disabled.

superm1, 2021-06-18:
It is a shim bug in newer distros (like Ubuntu 21.04): https://bugs.launchpad.net/ubuntu/+source/shim/+bug/1929471

You can turn off secure boot to do the update and it should help the situation.

superm1, 2021-06-18:
duplicate of https://github.com/fwupd/firmware-lenovo-thinkpad/issues/123

closed 2021-06-18 by superm1

kees, 2021-06-18:
Ah! Thanks for the pointer. Yes, if I add a manual UEFI entry for fwupdx64.efi and turn off secure boot, things worked correctly. Hopefully https://github.com/rhboot/shim/pull/379 will fix it?

superm1, 2021-06-28:
Hopefully, that or https://github.com/rhboot/shim/pull/381