fwupd/fwupd #3402: Uprev libxmlb from 0.1.15 to 0.3.2 causes trust validation to fail

djcampello, 2021-06-21:
In CrOS we recently uprev libxmlb from 0.1.15 to 0.3.2. Before the uprev you can see this in the logs:

```
18:13:06:0710 FuEngine             Emitting PropertyChanged('Status'='decompressing')
18:13:06:0710 FuMain               Emitting PropertyChanged('Status'='decompressing')
18:13:06:0711 Jcat                 Jcat engine kind 'gpg' not supported
18:13:06:0711 Jcat                 got 1 PKCS7 signatures
18:13:06:0712 Jcat                 verified: JcatResult:
  Timestamp:             2021-06-16T18:08:46Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature

18:13:06:0712 Jcat                 Jcat engine kind 'gpg' not supported
18:13:06:0712 FuCabinet            verified metadata fakedevice124.metainfo.xml: 3
18:13:06:0712 FuCabinet            processing file: fakedevice124.metainfo.xml
18:13:06:0712 XbSilo               compiling 05144279b5e615a8b1e5fd0240ab6db71d9b410e:filename=fakedevice124.metainfo.xml:prefix=components?
18:13:06:0713 FuCabinet            processing release: 1.2.4
18:13:06:0713 Jcat                 Jcat engine kind 'gpg' not supported
18:13:06:0713 Jcat                 got 1 PKCS7 signatures
18:13:06:0714 Jcat                 verified: JcatResult:
  Timestamp:             2021-06-16T18:08:45Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature

18:13:06:0714 Jcat                 Jcat engine kind 'gpg' not supported
18:13:06:0714 FuCabinet            verified payload fakedevice124.bin: 3
18:13:06:0714 FuEngine             Emitting PropertyChanged('Status'='idle')
18:13:06:0714 FuMain               Emitting PropertyChanged('Status'='idle')
....
```

and then update continues as expected...

After the uprev you would get the exact same output on the daemon as above but only up to that point and the update did not proceed...
On the client side:

```
localhost ~ # fwupdmgr update b585990a-003e-5270-89d5-3705a17f9a43
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
WARNING: This package has not been validated, it may not work properly.
Selected device: Integrated Webcam?
Integrated Webcam? and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading?             [***************************************]
Decompressing?           [***************************************]
archive signature missing or not trusted
```

@g2pop @bleungatchromium

hughsie, 2021-06-21:
Unexpected!

			g_debug ("verified payload %s: %u",
				 basename, results->len);
			release_flags |= FWUPD_RELEASE_FLAG_TRUSTED_PAYLOAD;

...so that bits working fine. I think the problem might be you need to recompile fwupd against the new version due to things like this:

	#if LIBXMLB_CHECK_VERSION(0,2,0)
		query = xb_query_new_full (xb_node_get_silo (component),
					   "releases/release",
					   XB_QUERY_FLAG_FORCE_NODE_CACHE,
					   error);
		if (query == NULL)
			return FALSE;
		rel_newest = xb_node_query_first_full (component, query, &error_local);
	#else
		rel_newest = xb_node_query_first (component, "releases/release", &error_local);
	#endif

Specifically I think you need XB_QUERY_FLAG_FORCE_NODE_CACHE to be set for this new libxmlb version.

djcampello, 2021-06-21:
I did recompile... :(

djcampello, 2021-06-21:
I actually found that snippet too and tried to replace with an #if 0 to see it made any difference.

hughsie, 2021-06-21:
@djcampello does wip/hughsie/node-cache fix things?

closed 2021-06-21 by djcampello