fwupd/fwupd #3411: meson test failed at 'synaptics-mst-self-test'

CragW, 2021-06-23:
**Describe the bug**
Attempt to build ubuntu deb package from source aaa77c6f by following this [readme](https://github.com/fwupd/fwupd/blob/master/contrib/README.md#deb-packages), don't know why failed during meson test.

**Steps to Reproduce**
```
fwupd/contrib/ci$ sudo OS=ubuntu-x86_64 ./generate_docker.py build
fwupd$ sudo docker run --privileged -t -v `pwd`:/github/workspace fwupd-ubuntu-x86_64 

```
**Expected behavior**
DEB build successfully

**docker build console output**
[docker-ubuntu-build-failed.log](https://github.com/fwupd/fwupd/files/6701970/docker-ubuntu-build-failed.log)

superm1, 2021-06-23:
Can you try removing `dist` directory from your builder?  I think that might be causing the problems.

CragW, 2021-06-24:
That worked, thank you!

closed 2021-06-24 by CragW