fwupd/fwupd #3418: Fresh install fwupd-1.6.1-setup-x86_64.exe on Windows 10 gave g_usb_version_string error

breisig, 2021-06-24:
I am interested in the project and downloaded and installed the latest fwupd-1.6.1-setup-x86_64.exe on my Windows 10 Pro 64bit and the following error happens when I try to run fwupdtool.exe in the folder.

```
The procedure entry point g_usb_version_string could not be located in the dynamic link library C:\Program Files(x86)\fwupd\bin\fwupdtool.exe
```
Any ideas?

hughsie, 2021-06-25:
I've just built a new libgusb mingw build for Fedora; will test later: https://koji.fedoraproject.org/koji/buildinfo?buildID=1775516

superm1, 2021-06-25:
I'm confused though, do we subproject libgusb in win32?  If so, is the problem just that we're supposed to use the subproject generated DLL?

breisig, 2021-06-25:
Also, does this mean I will have to wait until version 1.6.2 for this to be fixed?

hughsie, 2021-06-25:
> do we subproject libgusb in win32

I think we do (for building) but from looking at this error we appear to be using the system version at runtime.

superm1, 2021-06-25:
So wouldn't it be better to just put the subprojected one into the binary for runtime?  Like this: https://github.com/fwupd/fwupd/compare/wip/superm1/fix-win32-gusb?expand=1

superm1, 2021-06-25:
> Also, does this mean I will have to wait until version 1.6.2 for this to be fixed?

I think you can grab the DLL from that fedora build and replace it locally and potentially fix it.

commit 74e0c6ce (2021-06-25): trivial: fix win32 build (Fixes: #3418)

Use the gusb that we build rather than the system one.
commit f5839e0d (2021-06-25): trivial: fix win32 build (Fixes: #3418)

Use the gusb that we build rather than the system one.
commit a962d471 (2021-06-25): trivial: fix win32 build (Fixes: #3418)

Use the gusb that we build rather than the system one.
closed 2021-06-25 by superm1