fwupd/fwupd #3428: Dell OptiPlex 5055 AMD Ryzen doesn't run firmware update on reboot

maxamillion, 2021-06-26:
**Describe the bug**
I have a Dell OptiPlex 5055 with an AMD Ryzen 5 PRO 2400G CPU and instead of performing a firmware update on reboot after running `fwupdmgr update` it runs a memory diagnostics check.

What I think is relevant dmidecode information:
```
Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: Dell Inc.
        Version: 1.2.7
        Release Date: 05/31/2019
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                ACPI is supported
                USB legacy is supported
                LS-120 boot is supported
                ATAPI Zip drive boot is supported
                BIOS boot specification is supported
                Function key-initiated network boot is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.2
        Firmware Revision: 99.99

Handle 0x0100, DMI type 1, 27 bytes
System Information
        Manufacturer: Dell Inc.
        Product Name: OptiPlex 5055 Ryzen APU
        Version: Not Specified
        Serial Number: XXXXXX
        UUID: XXXXXXXXXXXXX
        Wake-up Type: Power Switch
        SKU Number: 07F2
        Family: OptiPlex

Handle 0x0200, DMI type 2, 17 bytes
Base Board Information
        Manufacturer: Dell Inc.
        Product Name: 07HXY6
        Version: A01
        Serial Number: XXXXXXXXXXXXXX
        Asset Tag: Not Specified
        Features:
                Board is a hosting board
        Location In Chassis: Not Specified
        Chassis Handle: 0x0000
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0300, DMI type 3, 24 bytes
Chassis Information
        Manufacturer: Dell Inc.
        Type: Desktop
        Lock: Not Present
        Version: Not Specified
        Serial Number: FS8T903
        Asset Tag: Not Specified
        Boot-up State: Safe
        Power Supply State: Safe
        Thermal State: Safe
        Security Status: None
        OEM Information: 0x00000000
        Height: Unspecified
        Number Of Power Cords: 1
        Contained Elements: 0
        SKU Number: Desktop

```

**Steps to Reproduce**
Steps to reproduce the behavior.
```shell
$ sudo fwupdmgr update
Devices with no available firmware updates:
 • Samsung SSD 970 EVO 500GB
 • UEFI dbx
Upgrade available for System Firmware from 1.2.7 to 1.2.11
OptiPlex 5055 Ryzen APU must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[***************************************]
Scheduling…              [***************************************]
Successfully installed firmware
Devices with the latest available firmware version:
 • Unifying Receiver

An update requires a reboot to complete. Restart now? [y|N]: y
```

Upon reboot, the boot is interrupted as expected but a memory diagnostic check runs instead of a firmware update.

**Expected behavior**
Firmware update to be applied.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version: 1.5.10
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.5.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): It's pre-installed in Fedora Silverblue 34

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version: Fedora Silverblue 34
- Have you tried rebooting? Yes
- Is this a regression? Yes, I've successfully performed firmware updates running Silverblue via fwupd on this machine before (and on Silverblue on my Dell Latitude 7390 as well)

superm1, 2021-06-26:
Do you have secure boot on?

GoPerry, 2021-06-28:
@maxamillion 
You can boot into the BIOS setting to see if the Secure Boot checked.

maxamillion, 2021-06-28:
Secure Boot is on, yes.

superm1, 2021-06-28:
There's a bug in the latest shim in Fedora with loading anything but the primary application it's complied for (IE grub).

Can you please try to disable secure boot and try again? This will turn off using shim for fwupd updates.

jwflory, 2021-06-28:
Hi, I was able to replicate this on a Lenovo ThinkPad X1 Carbon Gen 8.

`fwupdmgr` redirected me to [this wiki page](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot) which advised turning off Secure Boot Order. I tried this without success. I saw this issue and disabled Secure Boot. My ThinkPad successfully triggered the firmware update after disabling Secure Boot. After installing the firmware update, I rebooted and turned Secure Boot back on. All is working for me now.

Is this a bug that will be fixed in time, or is this a permanent workaround? I hang out in a few Fedora Linux support chats, so it will be good to know to pass along to other users who mentioned hitting this bug.

superm1, 2021-06-28:
I know Ubuntu has: https://bugs.launchpad.net/ubuntu/+source/shim/+bug/1929471
And this PR is in progress to fix things upstream: rhboot/shim#381

Don't know about any Fedora bug trackers opened for it right now.

superm1, 2021-06-28:
Since it's truly a shim bug, and it will need to be fixed in shim upstream and then each of the distros need to pick it up.
It's really unfortunate that this has effectively broken all UEFI firmware updates w/ secure boot.

We are discussing in https://github.com/fwupd/fwupd/pull/3421 a possibility of a GRUB based fwupd launcher that would work with secure boot on, but it's early discussions.

maxamillion, 2021-06-28:
Unfortunately disabling secure boot didn't work for me, I'm still being dropped into the Memory Check without secure boot enabled.

superm1, 2021-06-28:
Did you re-run the update after turning off secure boot?  
Please do the following:
1. Turn on fwupdate debugging (`fwupdate -d`)
2. Re-run the updates but don't reboot. 
3. Provide `efibootmgr` output after running the update from `fwupdmgr` but before rebooting?  Share this.
3. Run the update
4. Provide `fwupdate -L` output to show debugging logs

michel-slm, 2021-06-29:
> X1

> Hi, I was able to replicate this on a Lenovo ThinkPad X1 Carbon Gen 8.

I also reproduced this on a ThinkPad X1 Nano Gen 1. Updates fail to install until I disable SecureBoot, then they install just fine.

maxamillion, 2021-06-29:
@superm1 
```
BootNext: 0005
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0008,0007,0006,0005,0000
Boot0000* UEFI NVMe Storage
Boot0001* UEFI SanDisk Ultra USB 3.0 4C530000040220108191 2
Boot0002* Fedora
Boot0003* UEFI NVMe Storage
Boot0004* Windows Boot Manager
Boot0005* Linux Firmware Updater
Boot0006* CentOS Linux
Boot0007* Red Hat Enterprise Linux
Boot0008* CentOS Linux
```

maxamillion, 2021-06-29:
@superm1 
```
$ sudo fwupdate -L
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

superm1, 2021-06-29:
So it seems it didn't launch. Can you show the verbose output for efibootmgr instead? Hopefully it's obvious if an entry isn't being built right.

maxamillion, 2021-07-12:
@superm1 
```
$ sudo efibootmgr -v
BootNext: 0005
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0008,0007,0006,0005,0000
Boot0000* UEFI NVMe Storage     PciRoot(0x0)/Pci(0x1,0x6)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-55-01-41-81-D8)/HD(1,GPT,24628d7e-e694-42b8-a372-02535dae1906,0x800,0x12c000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0001* UEFI SanDisk Ultra USB 3.0 4C530000040220108191 2     PciRoot(0x0)/Pci(0x1,0x2)/Pci(0x0,0x0)/USB(2,0)/HD(1,MBR,0x329d2b14,0xa4,0x4f98)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0002* Fedora        HD(1,GPT,24628d7e-e694-42b8-a372-02535dae1906,0x800,0x12c000)/File(\EFI\fedora\shimx64.efi)
Boot0003* UEFI NVMe Storage     PciRoot(0x0)/Pci(0x1,0x6)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-55-01-41-81-D8)/HD(1,GPT,3faaec95-59dc-41fd-94bc-9c3698596cab,0x800,0x12c000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0004* Windows Boot Manager  HD(1,GPT,f925a87f-443d-48a3-8579-2459f0653e1e,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0005* Linux Firmware Updater        HD(1,GPT,24628d7e-e694-42b8-a372-02535dae1906,0x800,0x12c000)/File(\EFI\fedora\fwupdx64.efi)
Boot0006* CentOS Linux  HD(1,GPT,6adacf38-8d60-4cab-b405-ad23a2a6a315,0x800,0x12c000)/File(\EFI\centos\shimx64.efi)
Boot0007* Red Hat Enterprise Linux      HD(1,GPT,3faaec95-59dc-41fd-94bc-9c3698596cab,0x800,0x12c000)/File(\EFI\redhat\shimx64.efi)
Boot0008* CentOS Linux  HD(1,GPT,606bdde6-f003-4ea6-a0e7-9bff0754f9bc,0x800,0x12c000)/File(\EFI\centos\shimx64.efi)
```

superm1, 2021-07-12:
That appears to be with secure boot on. Can you please try with it off.

maxamillion, 2021-07-20:
I've confirmed secure boot is off and this is the output I get:
```
$ sudo efibootmgr -v
BootNext: 0005
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0008,0007,0006,0005,0000
Boot0000* UEFI NVMe Storage     PciRoot(0x0)/Pci(0x1,0x6)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-55-01-41-81-D8)/HD(1,GPT,24628d7e-e694-42b8-a372-02535dae1906,0x800,0x12c000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0001* UEFI SanDisk Ultra USB 3.0 4C530000040220108191 2     PciRoot(0x0)/Pci(0x1,0x2)/Pci(0x0,0x0)/USB(2,0)/HD(1,MBR,0x329d2b14,0xa4,0x4f98)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0002* Fedora        HD(1,GPT,24628d7e-e694-42b8-a372-02535dae1906,0x800,0x12c000)/File(\EFI\fedora\shimx64.efi)
Boot0003* UEFI NVMe Storage     PciRoot(0x0)/Pci(0x1,0x6)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-55-01-41-81-D8)/HD(1,GPT,3faaec95-59dc-41fd-94bc-9c3698596cab,0x800,0x12c000)/File(\EFI\Boot\BootX64.efi)N.....YM....R,Y.
Boot0004* Windows Boot Manager  HD(1,GPT,f925a87f-443d-48a3-8579-2459f0653e1e,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0005* Linux Firmware Updater        HD(1,GPT,24628d7e-e694-42b8-a372-02535dae1906,0x800,0x12c000)/File(\EFI\fedora\fwupdx64.efi)
Boot0006* CentOS Linux  HD(1,GPT,6adacf38-8d60-4cab-b405-ad23a2a6a315,0x800,0x12c000)/File(\EFI\centos\shimx64.efi)
Boot0007* Red Hat Enterprise Linux      HD(1,GPT,3faaec95-59dc-41fd-94bc-9c3698596cab,0x800,0x12c000)/File(\EFI\redhat\shimx64.efi)
Boot0008* CentOS Linux  HD(1,GPT,606bdde6-f003-4ea6-a0e7-9bff0754f9bc,0x800,0x12c000)/File(\EFI\centos\shimx64.efi)
```

maxamillion, 2021-07-20:
Also for clarity, it's still not working and instead launching the RAM test.

superm1, 2021-07-20:
This looks like a BIOS bug at this point. @hughsie  you agree?

To work around it, you could potentially launch fwupd from grub. We do have a branch with this that we were discussing merging. It was mostly because of shim issues with secure boot but I think it would also help your situation most likely. Would you be open to testing that branch to check?

maxamillion, 2021-07-21:
@superm1 happy to test, how do I go about that?

superm1, 2021-07-22:
Can you check out the branch from this PR? https://github.com/fwupd/fwupd/pull/3421
You can generate some RPMS to use from there.  Modify `/etc/fwupd/uefi_capsule.conf` to set it to enable grub, and then restart the daemon, and re-run the update.

hughsie, 2021-08-03:
You can use grub using the recently released 1.6.2

closed 2021-08-03 by hughsie

maxamillion, 2021-08-09:
Apologies I never got around to testing in time, thank you!