fwupd/fwupd #3434: New FW seems published twice ?

joakim-tjernlund, 2021-06-29:
Looking at https://fwupd.org/lvfs/firmware/new is looks like most(every?) FW release is published twice.

hughsie, 2021-06-30:
Hmm, interesting. Investigating now.

hughsie, 2021-06-30:
Probably fixed in https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1035 -- will test with staging first.

hughsie, 2021-06-30:
Okay mostly fixed. There's one issue I need to fix lower prio where some FirmwareEvent objects are duplicated for some reason.

closed 2021-06-30 by hughsie

joakim-tjernlund, 2021-06-30:
I can see that, thanks :)

joakim-tjernlund, 2021-07-04:
FYI, the latest  ```Lenovo ThinkStation P620 Embedded Controller Update``` also looks duplicated. 
I guess this is due to the  ```FirmwareEvent objects``` you mentioned above.

hughsie, 2021-07-04:
Hmm, this is kinda legitimate:

[image]

I think I need a `distinct` or something in the query.

hughsie, 2021-07-05:
Fixed properly in https://gitlab.com/fwupd/lvfs-website/-/commit/68ada3d1bbde630cfc37a5c973e09320601757a3 -- will deploy that sometime the end of this week.

joakim-tjernlund, 2021-07-05:
Does that fix the latest dell fw with same name but different versions? Details for both these point to the same version.

Skaffa Outlook för Android<https://aka.ms/ghei36>