fwupd/fwupd #3441: Please add a null-update UEFI capsule update option

steve-mcintyre, 2021-07-01:
Testing that shim calls fwupd ok is hard, with no way to do this on a VM (for example). It would be Really Nice (TM) if thrre was a way to provoke fwupd to create a null/stubbed-out EFI update so that we can validate this works.

I think we don't need to actually call UpdateCapsule(), just start the fwupd.efi program and say "Congratulations! Here is how you called me: <stuff>".

AFAICS fwupd is about the only user of the load options handling in shim, and it's in a fragile code path inside the shim binary. It would be lovely to be able to validate that at least the basics are working here, for reliable CI and testing.

superm1, 2021-07-01:
There is already a message that shows up, but it's not there for a long time:
https://github.com/fwupd/fwupd-efi/blob/main/efi/fwupdate.c#L564

There is also already the ability to fake an ESRT via an environment variable.  How about a flow like this:

1. dd if=/dev/zero of=foo.bin bs=1
2. FWUPD_SYSFSFWDIR=directory fwupdtool install-blob foo.bin  $GUID
3. Reboot, and check that you see that message.

All we need to do then is give some clearer directions on all of that, and add a longer delay in fwupd-efi in the no capsules case.

BTW if you're not already aware, we're also considering this: https://github.com/fwupd/fwupd/pull/3421 which will take shim arguments junk entirely out of the equation if you use GRUB...

hughsie, 2021-07-01:
> fwupdtool install-blob foo.bin $GUID

Don't we need something to add a fake device with that GUID? e.g. like a fake ESRTR entry?

superm1, 2021-07-01:
Yeah, we use a fake esrt entry by `FWUPD_SYSFSFWDIR`, no?

steve-mcintyre, 2021-07-02:
> Yeah, we use a fake esrt entry by `FWUPD_SYSFSFWDIR`, no?

So what I'm asking for is already available then? If so, can you give me a 100% spelled-out example of how to use it please? :-)

I'm feeling dumb - I have no idea what $GUID might work, etc.

hughsie, 2021-07-02:
> I'm feeling dumb - I have no idea what $GUID might work, etc.

You're certainly not dumb, this is much too hard to test. What about something like https://github.com/fwupd/fwupd/pull/3447

steve-mcintyre, 2021-07-02:
looks bang-on, thanks!

closed 2021-11-02 by hughsie