fwupd/fwupd #3443: usage of gdbus

binaryNewts, 2021-07-01:
I have a couple questions about the implementation of gdbus:

I saw that gdbus is written in C, but can be used in other languages (I’m aiming for C++) with introspection, but I don’t know what exactly that entails on my end.

I would like to use gdbus to send a signal back for my method call without affecting the method calls already in place. If there’s a way to send back a gobj over another dbus service (I can’t figure out the name of the one that’s currently implemented but I think it might not be a common api), that might be fine as well, assuming that fwupd can pick up the signal when it’s sent over this other service.

I’d also like to generally know the process for sending something over dbus through glib. I tried to follow a Hello World I saw but I wasn’t sure which functions were necessary, especially given that I have something in place already.

If you’d prefer to answer these over a call rather than chat, that works for me just let me know and we can find a time.

Thank you!

( @bleungatchromium )

hughsie, 2021-07-01:
So, GDBus is just a binding for the DBus protocol. DBus is like the IPC between the server and the client. In fwupd, the server is fwupd, and the client is something like gnome-software. The client calls methods on the server using DBus, and also listens for global signals from the service.

I guess the questions are:

 * what is your client and server in your question?
 * what interface does the server export?
 * can you do a diagram showing what you imagining the communication would look like?

binaryNewts, 2021-07-01:
I think my server would be powerd and I'm not sure about the client. I think the client that we're using might not have a name. My problem is that I want to use both the client already implemented in powerd and GDBus so that I can send back a g-object to fwupd.

The server is exporting the powerd power manager interface.

I think the basic communication should look something like fwupd -> powerd -> fwupd
All powerd needs to do is receive a method call and send back a signal.

closed 2021-07-28 by superm1