fwupd/fwupd #3448: today's firmware.xml.gz signature appears to be broken

morgwai, 2021-07-02:
**Describe the bug**
After downloading `https://cdn.fwupd.org/downloads/firmware.xml.gz` and `firmware.xml.gz.asc`, attempt to verify the signature fails.

**Steps to Reproduce**
```bash
$ wget https://cdn.fwupd.org/downloads/firmware.xml.gz
--2021-07-02 18:06:35--  https://cdn.fwupd.org/downloads/firmware.xml.gz
Resolving cdn.fwupd.org (cdn.fwupd.org)... 151.101.194.49, 151.101.2.49, 151.101.66.49, ...
Connecting to cdn.fwupd.org (cdn.fwupd.org)|151.101.194.49|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 629484 (615K) [text/xml]
Saving to: ‘firmware.xml.gz’

firmware.xml.gz              100%[==============================================>] 614.73K  1.69MB/s    in 0.4s    

2021-07-02 18:06:36 (1.69 MB/s) - ‘firmware.xml.gz’ saved [629484/629484]

$ wget https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
--2021-07-02 18:06:39--  https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
Resolving cdn.fwupd.org (cdn.fwupd.org)... 151.101.194.49, 151.101.2.49, 151.101.66.49, ...
Connecting to cdn.fwupd.org (cdn.fwupd.org)|151.101.194.49|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 488 [application/octet-stream]
Saving to: ‘firmware.xml.gz.asc’

firmware.xml.gz.asc          100%[==============================================>]     488  --.-KB/s    in 0s      

2021-07-02 18:06:39 (5.88 MB/s) - ‘firmware.xml.gz.asc’ saved [488/488]

$ gpg --verbose --verify firmware.xml.gz.asc firmware.xml.gz
gpg: Signature made Fri 02 Jul 2021 14:42:33 +07
gpg:                using RSA key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
gpg: using pgp trust model
gpg: BAD signature from "Linux Vendor Firmware Service <sign@fwupd.org>" [unknown]
gpg: binary signature, digest algorithm SHA256, key algorithm rsa2048
```
this in turn prevents from upgrading firmware:
```bash
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: '48A6D80E4538BAC2' is not a valid signature
```

**Expected behavior**
the signature should get verified properly and `sudo fwupdmgr refresh` should succeed. 

**Additional questions**
ubuntu-18.04
gpg 2.2.4

hughsie, 2021-07-02:
This is normally a temporary issue that goes away. If you can update to fwupd 1.4.0 that's been designed to fix this issue with a new metadata file called jcat.

morgwai, 2021-07-02:
> This is normally a temporary issue that goes away. If you can update to fwupd 1.4.0 that's been designed to fix this issue with a new metadata file called jcat.

I don't see how updating fwupd can fix the fact that downloading files manually and verifying signature manually with gpg fails: am I missing something?

Thanks!

hughsie, 2021-07-02:
> I don't see how updating fwupd can fix the fact that downloading files manually and verifying signature manually with gpg fails: am I missing something?

We literally designed JCat to cope with the decoupled CDN problem. The first release of fwupd to use libcat was 1.4.0 and I'm guessing you're using a very old unsupported version of fwupd.

morgwai, 2021-07-02:
ah, I see: so it's cdn's lag problem and JCat works around it, correct? in such case I'm closing the issue.
Thanks!

closed 2021-07-02 by morgwai