fwupd/fwupd #3454: fu_device_inhibit() doesn't inhibit child devices

hardboprobot, 2021-07-06:
**Describe the bug**
Removing the `FWUPD_DEVICE_FLAG_UPDATABLE` flag on a device also causes their child devices to become non-updatable. When using `fu_device_inhibit()` instead, I'm seeing a different effect. After calling `fu_device_inhibit()` on a device, the daemon still shows the device and its child devices as updatable. Example after calling `fu_device_inhibit()` on "MX Master 3 B":

```
|─Unifying Receiver:
│ │   Device ID:          10aa9ef6f15c92170b40763d3e48800760483ba4
│ │   Summary:            A miniaturised USB wireless receiver
│ │   Current version:    MPR05.00_B0008
│ │   Bootloader Version: BOT30.00_B0005
│ │   Vendor:             Logitech (HIDRAW:0x046D, USB:0x046D)
│ │   Install Duration:   30 seconds
│ │   GUIDs:              02d049be-c3f5-51c4-a312-b2284b43415a
│ │                       af1404c4-f038-5b3e-92b0-09bf4aa84f1c ← HIDRAW\VEN_046D&DEV_C548&REV_00
│ │                       f70d9241-2656-546b-a8e4-211bac348667 ← HIDRAW\VEN_046D&DEV_C548
│ │   Device Flags:       • Updatable
│ │ 
│ ├─MX Master 3 B:
│ │ │   Device ID:        a2e14e10536ba84dc3b8c04bd921a71f53cef041
│ │ │   Summary:          Unifying Mouse
│ │ │   Current version:  RBM14.00_B0008
│ │ │   Bootloader Version:BL141.00_B0008
│ │ │   Vendor:           Logitech (HIDRAW:0x046D, USB:0x046D)
│ │ │   Install Duration: 30 seconds
│ │ │   Update Error:     unreachable
│ │ │   GUIDs:            02d049be-c3f5-51c4-a312-b2284b43415a
│ │ │                     af1404c4-f038-5b3e-92b0-09bf4aa84f1c ← HIDRAW\VEN_046D&DEV_C548&REV_00
│ │ │                     f70d9241-2656-546b-a8e4-211bac348667 ← HIDRAW\VEN_046D&DEV_C548
│ │ │                     b2b50d12-c3df-5980-b5e9-6cc6c34b3037 ← HIDRAW\VEN_046D&DEV_C548&MOD_B0280000000000
│ │ │   Device Flags:     • Updatable
│ │ │ 
│ │ └─SoftDevice:
│ │       Device ID:      e7f146cdf7b5fa31ed543fd04beaa436662cef89
│ │       Current version:00.00_B0119
│ │       Vendor:         Logitech (HIDRAW:0x046D, USB:0x046D)
│ │       GUID:           fab2306a-2414-5fac-8d0a-6547eb066657 ← HIDRAW\VEN_046D&DEV_C548&MOD_B0280000000000_SOFTDEVICE
│ │       Device Flags:   • Updatable
```

**Steps to Reproduce**
Inhibit a device with `fu_device_inhibit()` and then run `fwupdmgr get-devices`.

**Expected behavior**
I'd expect both the device and its child devices to become non-updatable, or at least I'd expect the child device to show the same "Update Error" as the inhibited parent.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
client version:	1.6.1-122-gc3b43c59
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.6.1-122-gc3b43c59
```
(Private repo, rebased on top of https://github.com/fwupd/fwupd/commit/dfaae2e837a5fc58629aaa02c02003780e6bbef1)
Installed from source.

**Additional questions**
- Operating system and version: Ubuntu Linux 20.04, 5.8.0-59-generic kernel, x86_64

superm1, 2021-07-06:
I don't actually agree that child devices should explicitly become inhibited if parent is. It entirely depends on the circumstances of the inhibition.

If this behavior that makes sense for the device and situation I think the plugin can detect the children and inhibit them.

hughsie, 2021-07-07:
> I don't actually agree that child devices should explicitly become inhibited if parent is

Yes, agreed. I'd argue that inhibiting the child is probably what most plugins want, but I think an internal flag to control this behaviour is exactly what we want. Patch coming up.

closed 2021-07-07 by hughsie