fwupd/fwupd #3460: ThinkPad Thunderbolt 3 Dock; firmware incompatible, got VL820Q8, expected VL820Q7

josiah47, 2021-07-08:
**Describe the bug**
sudo fwupdmgr update
results in error firmware incompatible, got VL820Q8, expected VL820Q7

**Steps to Reproduce**
sudo fwupdmgr get-updates
sudo fwupdmgr update
Restarting device…       [-                                      ]Devices with no available firmware updates: 
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.0 Hub
Upgrade available for ThinkPad Thunderbolt 3 Dock from 56.00 to 62.00
ThinkPad Thunderbolt 3 Dock and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating ThinkPad Thunderbolt 3 Dock USB Audio…******************]
Restarting device…       [***************************************]
Updating USB3.1 Hub…     [************************************** ]
Restarting device…       [************************************** ]firmware incompatible, got VL820Q8, expected VL820Q7

**Expected behavior**
Firmware to be applied

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
fwupdmgr --version
client version:	1.6.0
compile-time dependency versions
	gusb:	0.3.6

Restarting device…       [-                                      ]daemon version:	1.6.0

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`sudo fwupdmgr update

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
fwupdmgr get-devices --show-all-devices
Restarting device…       [-                                      ]20SUS34700
│
├─ThinkPad Thunderbolt 3 Dock:
│ │   Device ID:          11ef1614a9d17196a5e780378194750b11296379
│ │   Current version:    56.00
│ │   Vendor:             Lenovo (THUNDERBOLT:0x0108|TBT:0x0108)
│ │   GUIDs:              45c1814e-eb83-5b03-8997-ef8d4d8ade29
│ │                       03b97e38-f4b8-5a19-917c-28f95ca29768
│ │                       75039175-c354-5df9-a798-5da41461c007
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │ 
│ ├─USB3.1 Hub:
│ │ │   Device ID:        48b0e21e27b6bab9c8208a603e5b65d8a42344e5
│ │ │   Summary:          USB 3.x Hub
│ │ │   Current version:  51.53
│ │ │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration: 15 seconds
[... 242 lines cut ...]
│                         • Cryptographic hash verification is available
│   
└─Unifying Receiver:
      Device ID:          506ad032bedb5227b5d11040e02f4f3058bfd2d6
      Summary:            A miniaturised USB wireless receiver
      Current version:    RQR12.01_B0019
      Bootloader Version: BOT01.02_B0014
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                          279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:       • Updatable
                          • Supported on remote server
    
________________________________________________

Devices that were not updated correctly:

 • USB3.1 Hub (51.53 → 51.63)

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2021-07-09:
@josiah47 just to confirm, this is the "Thunderbolt 3 Dock **2**" -- right?

josiah47, 2021-07-12:
Yes it is a Dock Gen 2

josiah47, 2021-07-13:
Original Dock when Issue first submitted
[image]

Secondary Dock: But this is a first gen

also same error 

sudo fwupdmgr update
Restarting device…       [-                                      ]Devices with no available firmware updates: 
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.0 Hub
Upgrade available for ThinkPad Thunderbolt 3 Dock from 39.00 to 62.00
ThinkPad Thunderbolt 3 Dock and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating ThinkPad Thunderbolt 3 Dock USB Audio…******************]
Restarting device…       [***************************************]
Updating USB3.1 Hub…     [************************************** ]
Restarting device…       [************************************** ]firmware incompatible, got VL820Q8, expected VL820Q7

[image]

hughsie, 2021-07-16:
Okay, I can reproduce here. Debugging.

closed 2021-07-17 by hughsie

amigadave, 2021-11-02:
I tried updating the VIA USB hubs on the Thunderbolt 3 Gen 2 Dock just now, and it failed with 1.7.0 and 1.7.1 with the fix in https://github.com/fwupd/fwupd/pull/3504 with the same error message (firmware incompatible, got VL820Q8, expected VL820Q7). Downgrading to 1.5.9 I was able to upgrade without problems.