fwupd/fwupd #3498: dfu-tool write fail at fwupd 1.6.0

charles0821, 2021-07-15:
I run dfu-tool -v write xxx.dfu at [fwupd 1.6.0] , and I got the fail message =no images in firmware file.
But it is all fine at [fwupd 1.5.5].
Maybe I didn't notice something, would you kindly help me on this. Appreciate a lot.

**_Below is the  dfu-tool -v write xxx.dfu , detailed log got at  at [fwupd 1.6.0]_**

localhost ~ # dfu-tool -v write /media/removable/USB\ Drive/ALC5686_VID0x0BDA_PID0x4BB6_V0003.dfu
(dfu-tool:24498): FuCommon-DEBUG: 19:03:21.277: reading /media/removable/USB Drive/ALC5686_VID0x0BDA_PID0x4BB6_V0003.dfu with 65552 bytes
(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.302: DFU v1.1
(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.302: using DFU transfer size 0x0040 bytes
(dfu-tool:24498): FuDevice-DEBUG: 19:03:21.302: using 23ec719b6aabc2d2dac5176c232f0da7a21881b0 for usb:01:00:04
(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.305: no UM0424 sector description in Realtek USB2.0 Audio
(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.305: no dnload-timeout, using default of 5ms
(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.306: refreshed status=OK and state=dfuIDLE (dnload=5)
decompressing:  0%
(dfu-tool:24498): FuDevice-DEBUG: 19:03:21.309: installing onto 23ec719b6aabc2d2dac5176c232f0da7a21881b0:
<firmware gtype="FuDfuFirmware">
  <flags>has-checksum|has-vid-pid</flags>
  <data size="0x10000">..;.g.A.9</data>
  <vendor>0xbda</vendor>
  <product>0x5685</product>
  <release>0x12</release>
  <dfu_version>0x100</dfu_version>
</firmware>

(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.309: no dnload-timeout, using default of 5ms
(dfu-tool:24498): FuPluginDfu-DEBUG: 19:03:21.309: refreshed status=OK and state=dfuIDLE (dnload=5)
no images in firmware file

**_Below is the  dfu-tool -v write xxx.dfu , detailed log got at  at [fwupd 1.5.5]_** 

localhost /usr/share/fwupd/remotes.d/vendor/firmware # dfu-tool -v write /media/removable/USB\ Drive/ALC5686_VID0x0BDA_PID0x4BB6_V0003.dfu
(dfu-tool:24281): FuCommon-DEBUG: 19:06:01.016: reading /media/removable/USB Drive/ALC5686_VID0x0BDA_PID0x4BB6_V0003.dfu with 65552 bytes
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.290: DFU v1.1
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.291: using DFU transfer size 0x0040 bytes
(dfu-tool:24281): FuDevice-DEBUG: 19:06:01.291: using 8f6aa1dd6787b3d80540f37f0c1072cdd86107f1 for usb:01:00:03:01:02
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.294: no UM0424 sector description in (null)
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.294: no dnload-timeout, using default of 5ms
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.294: refreshed status=OK and state=appIDLE (dnload=5)
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.295: no dnload-timeout, using default of 5ms
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.295: refreshed status=OK and state=appIDLE (dnload=5)
(dfu-tool:24281): libgusb-DEBUG: 19:06:01.363: 1:34 does not exist
(dfu-tool:24281): libgusb-DEBUG: 19:06:01.751: usb:01:00:03:01:02 is in replug, ignoring add
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.774: DFU v1.1
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.775: using DFU transfer size 0x0040 bytes
(dfu-tool:24281): FuDevice-DEBUG: 19:06:01.775: using 8f6aa1dd6787b3d80540f37f0c1072cdd86107f1 for usb:01:00:03:01:02
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.777: no UM0424 sector description in Realtek USB2.0 Audio
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.777: no dnload-timeout, using default of 5ms
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.777: refreshed status=OK and state=dfuIDLE (dnload=5)
(dfu-tool:24281): FuDevice-DEBUG: 19:06:01.778: installing onto 8f6aa1dd6787b3d80540f37f0c1072cdd86107f1:
FuFirmware:
  FuFirmwareImage:
  Data:                 0x10010

(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.779: no dnload-timeout, using default of 5ms
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.779: refreshed status=OK and state=dfuIDLE (dnload=5)
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.782: downloading to target: Realtek USB2.0 Audio
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.783: downloading element at 0x0000
(dfu-tool:24281): FuPluginDfu-DEBUG: 19:06:01.783: setting action device-write

hughsie, 2021-07-15:
This was fixed in 584f1d4c6a8c873c93ace299359b2cfabd7f2f13 which is present in 1.6.1 -- can you re-open if it still happens with a newer release than 1.6.0 please.

closed 2021-07-15 by hughsie

charles0821, 2021-07-15:
I appreciate your help, help a lot. thanks