fwupd/fwupd #3501: Adding "FWUPD_DEVICE_FLAG_REQUIRE_AC" Flag to Fake Webcam Device

g2pop, 2021-07-15:
For the purpose of testing a new plugin, we would like to add the the "FWUPD_DEVICE_FLAG_REQUIRE_AC" to the fake webcam device at plugins/test/fu-plugin-test.c. Since the plugin is related to power states, this feature would help make sure the plugin functions correctly by setting another case to be tested against.

hughsie, 2021-07-15:
How would you change the upower/powerd response at runtime in the test harness?

bleungatchromium, 2021-07-15:
> How would you change the upower/powerd response at runtime in the test harness?

Our test harness is running on an actual Chromebook, and our test environment can send commands to the Chromebook's Embedded Controller (EC) to disable the charging port to trigger the discharging battery case, though it remains physically connected.

bleungatchromium, 2021-07-22:
#3516 fixes this. Thanks!

closed 2021-07-22 by bleungatchromium