fwupd/fwupd #3505: Add a clang-format specification file

djcampello, 2021-07-16:
I have been playing around with clang-format --style and it seems that this is close to what we want:

```
clang-format --style='{IndentWidth: 8, UseTab: 'Always', SpaceBeforeParens: 'Always', BinPackArguments: false, BinPackParameters: false, AlwaysBreakAfterReturnType: 'All', BreakBeforeBraces: Custom, BraceWrapping: { AfterFunction: true } }'
```

We may be able to add this to the commit hooks

@bleungatchromium @g2pop @binaryNewts

hughsie, 2021-07-16:
`g_autoptr (GObject)` needs to have some kind of override. I also like the structs defined in blocks, e.g. 

	 typedef struct {
	-       gboolean         ret;
	-       gchar           *str;
	-       GError          *error;
	-       GPtrArray       *array;
	-       GMainContext    *context;
	-       GMainLoop       *loop;
	-       GVariant        *val;
	-       GHashTable      *hash;
	-       GBytes          *bytes;
	-       FwupdDevice     *device;
	+       gboolean ret;
	+       gchar *str;
	+       GError *error;
	+       GPtrArray *array;
	+       GMainContext *context;
	+       GMainLoop *loop;
	+       GVariant *val;
	+       GHashTable *hash;
	+       GBytes *bytes;
	+       FwupdDevice *device;
	 } FwupdClientHelper;

djcampello, 2021-07-16:
AlignConsecutiveDeclarations: 'AcrossEmptyLinesAndComments' for the second issue

Unfortunately I don't think we can treat g_autoptr differently and it is kind of arbitrary to me. Why do we treat it differently?

djcampello, 2021-07-16:
Here it is in file format:

```
$ cat .clang-format 
IndentWidth: 8
UseTab: 'Always'
SpaceBeforeParens: 'Always'
BinPackArguments: false
BinPackParameters: false
AlwaysBreakAfterReturnType: 'AllDefinitions'
AlignConsecutiveDeclarations: 'AcrossEmptyLinesAndComments'
BreakBeforeBraces: Custom
BraceWrapping:
 AfterFunction: true
 AfterStruct: true
```

hughsie, 2021-07-16:
>Why do we treat it differently

g_autoptr(Type) is essentially a type in itself. It's certainly not a macro or function.

djcampello, 2021-07-16:
maybe ugly but we could treat the exception outside clang-format in a follow up script

```
clang-format $1 | sed 's/g_autoptr (/g_autoptr(/g'
```

superm1, 2021-07-16:
So FYI I did try to come up with something before but eventually gave up:
https://github.com/fwupd/fwupd/pull/568

djcampello, 2021-07-19:
https://github.com/fwupd/fwupd/pull/568 looks good... Why was it abandoned?

superm1, 2021-07-19:
> https://github.com/fwupd/fwupd/pull/568 looks good... Why was it abandoned?

I was busy and it it was feeling like more and more bandaids on top of band-aid. If you want to take it as a starting point or parts of it please feel free.

closed 2021-07-22 by hughsie