fwupd/fwupd #3506: Incomplete recovery of bricked Synaptics VMM2322 MST hub from backup eeprom dump

bazookaben, 2021-07-17:
I eventually fixed my problem by using a combination of fwupd followed by vmmDPtool and vmmUpdater in Windows, but I though I should give a synopsis of what did and didn't work.

thanks to @joevt for helping me use fwupd

**Describe the bug**
Used `--install-blob` and `--ignore-vid-pid`  to attempt to install the original factory .eeprom file to a bricked Sunix DPU3000. Received the following error: 

`failed to update-cleanup after failed update: failed to wait for detach replug: device 7c2dd454e6f29453896a4824e8234317d237f8c5 did not come back
Firmware update failed: checksum 0 mismatched 57c1d9`

**Steps to Reproduce**

I'll just give you the full history of events.

I was trying to use vmmUpdater in Windows to install different firmwares on  my Sunix DPU3000 to see if I could solve some if its problems with certain resolutions and refresh rates.

The firmware weren't intended for this hub, but they were designed for device that used the Synaptics vmm2322, like Lenovo and HP docks. I thought maybe installing something newer.

Before I attempted to install anything, I used VmmDPTool, found in the Windows Store, to make a backup of my original eeprom.

Here is the original firmware and info dump:  [Sunix DPU3000 orignal firmware.zip](https://github.com/fwupd/fwupd/files/6833843/Sunix.DPU3000.orignal.firmware.zip)

I tried various firmware, and eventually got a significant brick where the hub's power light wouldn't turn on, and the hub could no longer be detected by my graphics card or the Synaptics software.

Here is the firmware responsible:  
[Lenovo vmm2322 2.24 firmware.zip](https://github.com/fwupd/fwupd/files/6833854/Lenovo.vmm2322.2.24.firmware.zip)

So then I tried to use fwupd to restore the orignal .eeprom file. 

Just using 'install-blob' I would get a board mismatch error:

`Installing on VMM2322…   [ -                                     ]

Decompressing…           [ -                                     ]board ID mismatch, got 0x0011, expected 0x0203`

I had to use `--ignore-vid-pid` to get around this, and the install started but ultimately failed with this error:

`failed to update-cleanup after failed update: failed to wait for detach replug: device 7c2dd454e6f29453896a4824e8234317d237f8c5 did not come back
Firmware update failed: checksum 0 mismatched 57c1d9`

However, this failed install _did_ put the Sunix hub in a state where it could now be detected by GPU's and Synaptics software again.

At this point, I made another dump via VmmDPTool to see what went wrong during the install with fwupd, and got these files:

[Bugged install from fwupd.zip](https://github.com/fwupd/fwupd/files/6833870/Bugged.install.from.fwupd.zip)

Notice that in the information dump, the firmware name and version are blank.

At this point, I was able to use vmmUpdater in Windows to correctly install the orignal firmware, and the MST hub was fully functional again.

So in the end, fwupd did save the Sunix by at least getting it to a point where it would communicate with Synaptics Windows app again

**fwupd version information**
1.5.8 with Ubuntu 21.04

joevt, 2021-07-17:
Did you use Intel GPU for the connection? Which CPU? Any other GPU probably won't work with a bricked MST Hub because the dp aux device for those other GPUs won't appear for the port that has the MST hub unless a display is connected to the MST hub - and you can't do that with your bricked MST Hub. In the case of an Intel GPU, the three DP aux devices are always present even if nothing is connected to them. I guess it's a GPU driver choice?

I see you used 1.5.8. I wonder if compiling 1.6.1 or whatever the current version is would change anything.

The bugged rom is just 64K of 0xFF bytes.

The two other roms appear to be valid. At least they have a similar format, with a EDID at the beginning. I don't have any info about what a firmware file should look like and I can't say if they're compatible with each other. I only have the firmware files for the newer MST hubs I have (VMM5xxx). They are 512K in size with 128K being mostly duplicated in each. Your 64K roms don't appear to have a duplicated part, but at least the first couple K bytes are similar.

The Tesla firmware does not have the 0xFF padding at the end to make the file an even 64K like the original rom does. I don't know if it matters.

I found the VmmDPTool.exe at a https://kb.cablematters.com/index.php?View=entry&EntryID=147
I used it to read firmwares for my VMM5210,5330,5333 MST hubs (supported by fwupd) and for the Cable Matters DisplayPort 1.4 to HDMI 2.1 VMM6100 adapter (not supported by fwupd maybe because it's not an MST Hub?). I'll try it with my Dual Link DVI adapters (VMM2320 from Club-3D and VMM2326 from StartTech).

bazookaben, 2021-07-17:
Yeah, Intel graphics on an i5 2400.

Couldn't install 1.6.1 because I couldn't download some of the necessary python programs to build it.

The two firmware I had attempted to install came with version history notes that included the number of the fimrware on my Sunix (2.15.002), so that's why I thought they _might_ be compatible and would be worth a shot. But I guess the second number, `configuration file version` might have something to do with why these firmware for laptop docks weren't compatible.

closed 2021-07-28 by superm1