fwupd/fwupd #3509: Lenovo ThinkPad X1 Carbon 5th (20HRCTO1WW) fails to install updates on Debian 10.10

donalhunt, 2021-07-18:
**Describe the bug**

Seems very similar to other X1 carbon issues over the past 2 years - annoyingly none of the bugs have useful summaries indicating what the correct course of action is.

Updates that are pending install:

- • 8 Intel Platform Update 2020.2 HF2 Release for com.lenovo.ThinkPadN1MRN.firmware (v. 0.1.50)
- Lenovo System Firmware Version 1.50 for com.lenovo.ThinkPadN1MET.firmware (v. 184.83.3874)

Behaviour is that upon reboot to apply the update, the machine boot straight into the grub menu. No attempt appears to be made to boot the fwupd.efi firmware updater

**Steps to Reproduce**

1. `$ fwupdmgr update`
2. reboot
3. verify if update applied (it is not)

OR

1. Open Software Centre on desktop
2. Install one or both updates
3. reboot
4. Get prompted to apply updates again

**Expected behavior**

firmware updates are applied during boot process.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$  fwupdmgr --version 
client version:	1.2.14
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.14
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
$ sudo apt-cache policy fwupd
fwupd:
  Installed: 1.2.14-1~deb10u1
  Candidate: 1.2.14-1~deb10u1
  Version table:
     1.5.7-4 500
        500 http://deb.debian.org/debian testing/main amd64 Packages
         50 http://deb.debian.org/debian unstable/main amd64 Packages
 *** 1.2.14-1~deb10u1 900
        900 http://deb.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20HRCTO1WW System Firmware
  DeviceId:             1cbe298fc17877b6883e85560778ac0812f0a385
  Guid:                 798ffd60-f10e-4ac4-8939-c8beabfe55b4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(cb80680a36dc443bd7231b56f4216c55d525e0ee)
  Checksum:             SHA256(88fd39353f45f4178c9a831d2c2bf506fb7fb0e457329b23986c4f79d1239d68)
  Vendor:               LENOVO
  Version:              0.1.49
  VersionLowest:        0.1.14
[... 416 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.19.0-16-amd64
├── initrd.img-4.19.0-17-amd64
├── lost+found [error opening dir]
├── System.map-4.19.0-16-amd64
├── System.map-4.19.0-17-amd64
├── vmlinuz-4.19.0-16-amd64
└── vmlinuz-4.19.0-17-amd64

6 directories, 318 files
```

**Additional questions**
- Operating system and version: *Debian 10.10*
- Have you tried rebooting? *Yes*
- Is this a regression? *Unknown*
- Are you using an NVMe disk? *Yes*
- Is secure boot enabled? *Yes* (also tried with it turned off)
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? *No*

donalhunt, 2021-07-18:
since this was asked previously, here's what happens when I update (`efibootmgr` output)...

```shell
$ fwupdmgr update
Downloading 0.1.50 for 20HRCTO1WW System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating 20HRCTO1WW System Firmware…                             ]
Scheduling…              [***************************************]
Downloading 184.83.3874 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…  \                                ]
Scheduling…              [***************************************]
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.20=same, 0.1.19=older
No upgrades for UEFI Device Firmware, current is 0.71.16: 0.71.16=same
No upgrades for SAMSUNG MZVLB512HAJQ-000L7, current is 5L2QEXA7: 5L2QEXA7=same

An update requires a reboot to complete. Restart now? [Y|n]: n

$ efibootmgr -v
BootNext: 0000
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0001,0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,4e08491c-2383-4959-96a8-387d5b840568,0x800,0x100000)/File(\EFI\debian\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* debian	HD(1,GPT,4e08491c-2383-4959-96a8-387d5b840568,0x800,0x100000)/File(\EFI\debian\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

donalhunt, 2021-07-18:
in case it's relevant...

```shell
$ apt policy shim-signed
shim-signed:
  Installed: 1.36~1+deb10u1+15.4-5~deb10u1
  Candidate: 1.36~1+deb10u1+15.4-5~deb10u1
  Version table:
     1.38+15.4-7 50
         50 http://deb.debian.org/debian unstable/main amd64 Packages
     1.37+15.4-6 500
        500 http://deb.debian.org/debian testing/main amd64 Packages
     1.36~1+deb10u2+15.4-5~deb10u1 500
        500 http://deb.debian.org/debian buster-updates/main amd64 Packages
 *** 1.36~1+deb10u1+15.4-5~deb10u1 900
        900 http://deb.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status
```

donalhunt, 2021-07-18:
Relevant bugs in other projects:
- debian-efi: [Bug#990447: fwupdmgr: Unable to install new updates](https://lists.debian.org/debian-efi/2021/07/msg00000.html)
- fwupd/firmware-lenovo-thinkpad: [A firmware update will not install on my Lenovo Thinkpad X1 Carbon 7th running Ubuntu 21.04](https://github.com/fwupd/firmware-lenovo-thinkpad/issues/123)

donalhunt, 2021-07-18:
Relevant for others stuck with pending updates:

I was able to boot the `fwupdx64.efi` image by turning off secure boot (not sure if that is relevant) and doing the following at the grub commandline on boot:

```shell
insmod chain
set root=(hd0,gpt1)
chainloader ($root)/efi/debian/fwupdx64.efi
boot
```

This resulted in the firmware updater running and the updates being applied.

lorenzhs, 2021-07-21:
> Relevant bugs in other projects:
>
>     * debian-efi: [Bug#990447: fwupdmgr: Unable to install new updates](https://lists.debian.org/debian-efi/2021/07/msg00000.html)
>
>     * [...]

This seems to make it pretty clear that it's a shim issue and not fwupd's fault, right? And that this has happened before, got fixed, and then got broken some other way again? Ugh. Anyway, fwiw, I can confirm the issue on a 20HQ model X1C5 with Debian unstable and Secure Boot enabled.

superm1, 2021-07-23:
> This seems to make it pretty clear that it's a shim issue and not fwupd's fault, right? And that this has happened before, got fixed, and then got broken some other way again? Ugh. Anyway, fwiw, I [...]

Yup

donalhunt, 2021-07-23:
I'm going to close the issue since it's not a fwupd bug. Do the shim folk have a project page / issue tracker?

closed 2021-07-23 by donalhunt

lorenzhs, 2021-08-01:
They do: https://github.com/rhboot/shim-review/issues/165, apparently https://github.com/rhboot/shim/pull/379 fixes this? But there's no release containing the fix afaik.