fwupd/fwupd #3511: Reading SMBIOS information requires root

tari, 2021-07-19:
When fwupd is not running as root on Linux, it cannot read Hwid values from SMBIOS because Linux restricts access to the SMBIOS to the root user. This gracefully fails, but devices that need to key quirks on Hwid values will not work.
Concretely, the `parade_lspcon` plugin needs some Hwid values to know how to talk to a device and on Chromium OS the daemon does not run as root so it can't complete probing the device.

Linux exposes a subset of parsed values from SMBIOS in file in `/sys/class/dmi/id` which are mostly world-readable (some device-specific IDs are still restricted), so I'd like to add support for reading values from there if direct access to the SMBIOS tables fails.

tari, 2021-07-19:
I think the best solution here is to add support in `FuSmbios` for reading the kernel-parsed values from `/sys/class/dmi/id` and convert them back to the raw-table-like representation used in `FuSmbios`. This avoids needing to make any changes further up the stack of extracting Hwids, but is kind of awkward in that it converts values that the kernel has already parsed back into SMBIOS tables, only for fwupd to parse them again. On balance, I think the parsing dance is better than adding another code path further up.

I've begun implementing this, and it mostly involves processing a mapping from sysfs filename to SMBIOS property type and offset that's of similar complexity to the existing `kenv` support for BSD.

hughsie, 2021-07-19:
Back a bit; why would you be installing firmware on devices without root permission? Doesn't that break a few other assumptions?

> and convert them back to the raw-table-like representation used in FuSmbios

I guess that works; I believe the kernel does some fixups and so it's not a 1:1 mapping all the time. For instance, I believe the DeviceInstanceIDs use the raw data which includes things like trailing spaces, and I think the kernel strips those out. For your  use case that might be fine, but caveat emptor.

tari, 2021-07-19:
Chromium OS [runs the daemon in a sandbox as nonroot plus some capabilities](https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/refs/heads/main/sys-apps/fwupd/files/fwupd.conf#36):

```
exec minijail0 \
  ... -u fwupd -g fwupd -c cap_sys_rawio+e -- /usr/libexec/fwupd/fwupd
```

For needs to date this seems to have been fine, though I'm not familiar with the process that led to the current sandbox settings.

hughsie, 2021-07-19:
Would it be possible to open up the minijail a little to be able to read smbios data? It seems quite safe to me...

tari, 2021-07-19:
I think the only hammer available to do that is `CAP_DAC_OVERRIDE` (ignoring *all* file permissions for the process), which is too blunt a tool.

hughsie, 2021-07-19:
> which is too blunt a tool

Is there anywhere else we can get the smbios blob from on ChromeOS? If not, then perhaps your fallback is a good idea.

superm1, 2021-07-19:
>Is there anywhere else we can get the smbios blob from on ChromeOS? If not, then perhaps your fallback is a good idea.

How about copying the blob into the jail before running and then set an override path to access it?

tari, 2021-07-21:
I suppose we could make a copy of the blob somewhere and make it readable by the daemon, but it seems rather messy- I think looking elsewhere if direct access fails is neater.

There was also an idea about using a udev rule to set permissions on the raw tables, but that doesn't work because udev doesn't actually operate on anything under `/sys/firmware`.

superm1, 2021-07-21:
>I suppose we could make a copy of the blob somewhere and make it readable by the daemon, but it seems rather messy- I think looking elsewhere if direct access fails is neater.

I think we have to keep in mind this is a reduced feature set, and any plugins that want to use SMBIOS functionality besides hwids may run into problems.  The ones that use it:
* redfish
* uefi-capsule
* flashrom
* dell

I think some sort of flag will need to be set preventing those plugins from running if this fallback path is introduced.

> There was also an idea about using a udev rule to set permissions on the raw tables, but that doesn't work because udev doesn't actually operate on anything under /sys/firmware.

What about proposing a permissions change upstream or introducing capabilities for reading firmware tables?

tari, 2021-07-21:
I realize that since we have a custom initscript anyway, I can set appropriate ownership during daemon startup which should be a suitable fix.

tari, 2021-07-22:
> I think some sort of flag will need to be set preventing those plugins from running if this fallback path is introduced.

They should be fine, at least based on my inspection of the redfish plugin- they should act like they do now (fail gracefully) if the fallback path isn't able to get the data they need. The fallback should only allow some features to work where they previously wouldn't if run without sufficient permissions.

PR #3517 (2021-07-23): SMBIOS: try reading from /sys/class/dmi if direct access fails
The raw SMBIOS tables that Linux exposes in /sys/firmware/dmi
are restricted to being readable by root only. If running as
non-root access is still permitted by fields that have been
pre-parsed by the kernel in /sys/class/dmi, most of which are
world-readable. This allows the daemon to load most HWIDs even
if running as a non-root user, as is done on Chromium OS.

Fixes #3511.

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ [... cut]
closed 2021-07-23 by hughsie