fwupd/fwupd #3537: Dell OptiPlex 5080: `fwupdmgr install OptiPlex_5080_1.4.0.exe` returns `The input is not of cabinet format`

paulmenzel, 2021-07-23:
With the Dell OptiPlex 5080, [firmware version 1.4.0 is not in LVFS](https://github.com/fwupd/firmware-dell/issues/77). Downloading the file with

    $ wget https://dl.dell.com/FOLDER07406342M/1/OptiPlex_5080_1.4.0.exe
    […]
    $ md5sum OptiPlex_5080_1.4.0.exe
    f8e3ce6031a0139bffc9957cf9c70efa  OptiPlex_5080_1.4.0.exe
    $ file OptiPlex_5080_1.4.0.exe
    OptiPlex_5080_1.4.0.exe: PE32 executable (GUI) Intel 80386, for MS Windows

wanting to install it with `fwupdmgr`, it fails.

    $ fwupdmgr install OptiPlex_5080_1.4.0.exe 
    WARNING: This package has not been validated, it may not work properly.
    Decompressing…           [ -                                     ]The input is not of cabinet format

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
client version:	1.6.1-1-g39b3140e
compile-time dependency versions
	gusb:	0.3.7

Decompressing…           [-                                      ]daemon version:	1.6.1-1-g39b3140e
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): sudo snap install fwupd --classic

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Decompressing…           [-                                      ]WARNING: This package has not been validated, it may not work properly.
OptiPlex 5080
│
├─Intel(R) Core™ i7-10700 CPU @ 2.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         a45b0522-5722-54bd-b802-86cd044262df
│                         538440cb-7cac-5185-bc01-d93a1e474058
│     Device Flags:       • Internal device
│   
├─Oland [Radeon HD 8570 / R7 240/340 / Radeon 520 OEM]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    87
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              f5d1d372-19cf-5dd4-b40f-4e50a914d984
│                         bc139a44-ca12-507d-a0ab-8025f0703a4f
│                         95ac7d23-435e-5904-8ef2-98b0fee31de5
│                         10dbbf9d-35b8-5950-8b06-e8287451eca0
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Current version:    1.1.7
│ │   Minimum Version:    1.1.7
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              459b329f-c828-4000-b4af-73347b3bcd10
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       018f8884-2400-53ad-abe7-268a25b0525e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─Intel AMT [unprovisioned]:
│ │     Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
│ │     Summary:          Hardware and firmware technology for remote out-of-band management
│ │     Current version:  14.0.39.1339
│ │     Bootloader Version:14.0.39.1339
│ │     Vendor:           Intel Corporation
│ │     GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
│ │     Device Flags:     • Internal device
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI Revocation Database
│       Current version:  238
│       Minimum Version:  238
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│     
├─TOSHIBA MQ01ACF050:
│     Device ID:          5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:            ATA Drive
│     Current version:    AV006D
│     Vendor:             Toshiba (ATA:0x1179|OUI:000039)
│     GUIDs:              4f8abbe6-d9b1-54af-817d-908d625827d3
│                         669d8858-49da-5e9a-901d-50eb9d0e515b
│                         21933c3c-6a43-5fe5-995c-99edd026d87d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:    7.2.1.0
│ │   Vendor:             Nuvoton Technology (TPM:NTC|PCI:0x1028)
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       fac1c8f3-73c8-5cd6-8330-07a3690b5140
│ │                       8d94eb9a-78aa-513d-af50-319e3b9f5210
│ │                       e9ccc1dc-960a-5e09-afe9-e59a904b776d
│ │                       f1c4013d-3f4c-5623-a40b-81ad96e23860
│ │   Device Flags:       • Internal device
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Internal device
│     
└─UEFI Device Firmware:
      Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
      Current version:    200
      Minimum Version:    200
      Vendor:             DMI:Dell Inc.
      GUIDs:              ffd6eef5-4372-4adc-8eeb-3dc0b7338375
                          fbfd77fb-abcd-56bc-bb4b-27bcd6c22fb4
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
```

[…]

**Additional questions**

- Operating system and version: 20.04.1 LTS (Focal Fossa)

hughsie, 2021-07-23:
I don't think you can just deploy the .exe like that, fwupd is expecting a .cab file. Have a look at ./contrib/firmware_packager/install_dell_bios_exe.py

paulmenzel, 2021-07-23:
Thank you. Now I remember that script. (Maybe it could be integrated into fwupdmgr.)

Anyway, despite the script looked like it was working, rebooting the system, the firmware update did not start. No idea why.

I now updated the device using the UEFI firmware option.

closed 2021-07-23 by paulmenzel

superm1, 2021-07-23:
> Anyway, despite the script looked like it was working, rebooting the system, the firmware update did not start. No idea why.

secure boot enabled?  There is a shim bug.  More details at the end of here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=990447