fwupd/fwupd #3543: Lenovo TPM update, via Flatpak: missing or invalid embedded capsule header

LinuxOnTheDesktop, 2021-07-24:
**Describe the bug**
When I try to install a Lenovo TPM update, via Flatpak, I get an error message that says: 'missing or invalid embedded capsule header'.

**Steps to Reproduce**
```
sudo flatpak install flathub org.freedesktop.fwupd
sudo -i flatpak run org.freedesktop.fwupd get-devices
sudo -i flatpak run org.freedesktop.fwupd refresh
sudo -i flatpak run org.freedesktop.fwupd get-updates
sudo -i flatpak run org.freedesktop.fwupd update
```

**Expected behavior**

No error message.

**fwupd version information**
```
$ sudo -i flatpak run org.freedesktop.fwupd --version
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7
```

Please note how you installed it: flatpak

**fwupd device information**
```shell
fwupdmgr get-devices --show-all-devices
20KH001UKP
│
├─Thunderbolt Controller:
│     Device ID:           9e4ed2585a7c7ad2de1e4593bb8b631301584cea
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.8.8341
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               963522d7-bc73-51fa-a70c-cf615f42bcf6
│                          b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                          ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.50
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update State:        success
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
[... 103 lines cut ...]
Installing on System Firmware…                                   ]
Scheduling…              [***************************************]

An update requires a reboot to complete. Restart now? [y|N]: y
No supported backend compiled in to perform the operation.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI Device Firmware from 0.71.12 to 0.73.20?                        ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20                 ║
║ • Do NOT turn off your computer or remove the AC adaptor while update is     ║
║ in progress                                                                  ║
║ 20KH001UKP must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
WARNING: This package has not been validated, it may not work properly.
Decompressing…           [***************************************]
Installing on UEFI Device Firmware…                              ]
Scheduling…              [   -                                   ]21:59:13:0886 FuPluginUefiCapsule  missing or invalid embedded capsule header
Scheduling…              [***************************************]

An update requires a reboot to complete. Restart now? [y|N]:
[Being worried by this point, I said 'n'.]
```

LinuxOnTheDesktop, 2021-07-24:
I find also that after performing the steps listed above, and having aborted the update (see above), one of my CPU cores was pegged and my CPU temp his 94 degrees Celsius. (I have changed the title of this issue so as to reflect that fact.) `sudo killall fwupdtool` returned the temperature to a reasonable level.

superm1, 2021-07-28:
> missing or invalid embedded capsule header

This means that the capsule needs to have the header applied.  Some vendors don't seem to do it by default.   To you, I don't know it has any impact. 

>sudo killall fwupdtool returned the temperature to a reasonable level.

So you're saying after you hit "n" to reboot that the process kept running?  That sounds like a bug.  Can you reproduce it?

LinuxOnTheDesktop, 2021-07-28:
> To you, I don't know it has any impact.

Well, updates are not applied - even when I tell fwup to apply them. Please see below.

> So you're saying after you hit "n" to reboot that the process kept running? That sounds like a bug. Can you reproduce it?

Yes, I believe that that is what happened. However, I am unwilling to spend more time risking my system with firmware updates, so I will not try to reproduce the bug. I will wait for Ubuntu - or whoever the relevant upstream party is - to start making updates work again; and I mean work via my distro's repository version of fwupd. I tried the flatpak to try to circumvent that bug. As I have said, pretty much, elsewhere:  the present inability of fwup to apply firmware updates - at least on certain Linux distributions and at least on some machines - is a security disaster, especially since it has gone on so long.

closed 2021-08-03 by hughsie