fwupd/fwupd #3546: Thinkpad P51 does not update firmware anymore

brandsimon, 2021-07-25:
After running 'fwupdmgr update` on my Arch Linux P51, it tells me that secure boot is on. So I use sbsign to sign the fwupdx64.efi. After rebooting and starting the efi file, my device reboots and it seams like it starts the upgrade process, but there is only a black screen forever. The device is not responding to the power button.

Version:
```
client version:	1.6.1
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.1
```

I installed it via pacman (fwupd: 1.6.1-2, fwupd-efi: 1.1-1).

Devices:
```shell
20HHCTO1WW
│
├─GM107GLM [Quadro M1200 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a2
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              34e93021-0a6a-5366-b15d-9004bf884caf ← PCI\VEN_10DE&DEV_13B6&SUBSYS_17AA224D&REV_A2
│                         0045d13b-445b-5d7b-bd58-ab14fe439454 ← PCI\VEN_10DE&DEV_13B6&SUBSYS_17AA224D
│                         5a37b326-73f6-5b5c-b5dc-14cd16efef2f ← PCI\VEN_10DE&DEV_13B6&REV_A2
│                         adf52255-8e5e-50f3-86ff-3efc1210840a ← PCI\VEN_10DE&DEV_13B6
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HD Graphics 630:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              4b4eb650-e2d4-505f-a050-356e8cc6ba1e ← PCI\VEN_8086&DEV_591B&SUBSYS_17AA224D&REV_04
│                         649131f9-bf7d-5e06-8157-04d9c8a958b2 ← PCI\VEN_8086&DEV_591B&SUBSYS_17AA224D
│                         27f3a118-c823-5f12-a6c3-7b5da55444bf ← PCI\VEN_8086&DEV_591B&REV_04
│                         f744dc01-a247-5543-b9db-2adba3091233 ← PCI\VEN_8086&DEV_591B
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Current version:    184.83.3874
│     Minimum Version:    184.83.3874
│     Vendor:             DMI:LENOVO
│     GUIDs:              bec0a796-5688-499f-b208-9fc1d1e734a6
│                         1e8ba52c-a2f1-56f6-99c1-ca1797a2801a ← UEFI\RES_{BEC0A796-5688-499F-B208-9FC1D1E734A6}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i7-7700HQ CPU @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         d0f754d5-1395-5573-bc83-85ba955da70a ← CPUID\PRO_0&FAM_06&MOD_9E&STP_9
│     Device Flags:       • Internal device
│   
├─SAMSUNG MZVKV512HAJH-000L1:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express Solid State Drive
│     Current version:    7L0QBXX7
│     Vendor:             Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:      S2CYNX0H906960
│     GUIDs:              b586caf2-ac7e-5a07-8630-dd289d8fd3f4 ← NVME\VEN_144D&DEV_A802&SUBSYS_144DA801&REV_01
[... 106 lines cut ...]
/boot/efi/EFI
/boot/efi/EFI/arch
/boot/efi/EFI/arch/linux-fallback-signed.efi
/boot/efi/EFI/arch/linux-lts-signed.efi
/boot/efi/EFI/arch/linux-lts-fallback-signed.efi
/boot/efi/EFI/arch/linux-signed.efi
/boot/efi/EFI/arch/fw
/boot/efi/EFI/arch/fw/fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
/boot/efi/EFI/arch/fw/fwupd-bec0a796-5688-499f-b208-9fc1d1e734a6.cap
/boot/efi/EFI/arch/fwupdx64.efi
/boot/efi/EFI/systemd
/boot/efi/EFI/systemd/systemd-bootx64.efi
/boot/efi/EFI/BOOT
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/Linux
```

**Additional questions**

- Operating system and version: Arch Linux (uptodate)
- Have you tried rebooting? Yes
- Is this a regression? Yes, it works with 1.5.9
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2021-07-25:
Does the device come back if you do the pinhole trick to reset? Or if you remove both batteries (including the internal one) for 10 minutes? If yes, then we need to demote the update out of stable, if not I have some angry emails to write.

brandsimon, 2021-07-25:
@hughsie 
Sorry, I forgot to mention that my device is still working. I removed the battery and AC to turn it off and it works perfectly fine after.
I downgraded to fwupd 1.5.9 to upgrade my device, which worked without any flaws.

hughsie, 2021-07-27:
@brandsimon Ohh, that's both good and bad at the same time. Any chance you could try fwupd git master?

brandsimon, 2021-08-01:
@hughsie 
Sorry, since this is my only production machine atm, I dont want to experiment with its firmware a lot.

matthewwardrop, 2021-08-03:
I used to have this problem with all of my Lenovo devices on 1.6.1... But 1.6.2 works again for me.

closed 2021-08-03 by hughsie