fwupd/fwupd #3548: BIOS update v1.42 fails on Lenovo Thinkpad X1 Carbon Gen 9

codethief, 2021-07-26:
**Describe the bug**
I tried to install Lenovo's [recent BIOS update (v1.42)](https://support.lenovo.com/de/de/downloads/ds548819) for the Thinkpad X1 Carbon Gen 9. (My laptop had been running v1.39.) After passing the cabinet files to `fwupdmgr`, I rebooted the device, the screen flickered a few times and then I was greeted by my usual LUKS password prompt. BIOS and `fwupdmgr` output indicate that the BIOS did not get updated.

In the end, after several failed attempts using `fwupdmgr`, I tried the "Bootable CD" image Lenovo provides and with that I managed to successfully carry out the update. (I was somewhat desperate to do the update, hoping that it would fix my throttling issues – unfortunately, it didn't…)

**Steps to Reproduce**

```
$ unzip n32ul08w.zip
Archive:  n32ul08w.zip
  inflating: N32ET66W.cab
 extracting: N32HT46W.cab

$ fwupdmgr install N32ET66W.cab
Do you want to reboot (y/n): n

$ fwupdmgr install N32HT46W.cab
Do you want to reboot (y/n): n

$ efibootmgr -v
BootNext: 0002
[…]
Boot0002* Linux-Firmware-Updater    HD(3,GPT,0f12d37a-3fa8-4196-a6ae-2f72cb0616ba,0x181800,0x40000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
[…]

$ reboot

$ fwupdmgr get-history
[…]
├─Embedded Controller:
│ │   Device ID:          0dcf00f0d9fd0bb13798a121c27a2832d24e005e
│ │   Previous version:   0.1.25
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot <-------------------------------------
│ │   Last modified:      2021-07-26 19:54
│ │   GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.26
│       License:          Unknown
│
└─System Firmware:
  │   Device ID:          dcd4118aa968110eaae58bf95432c9736be3a74e
  │   Previous version:   0.1.39
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot <-------------------------------------
  │   Last modified:      2021-07-26 19:57
  │   GUID:               14f3350e-cf63-4e68-a0d9-0af1d5389a17
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─  New version:      0.1.42
        License:          Unknown

$ sudo fwupdate -L 
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: File not found
```

**Expected behavior**
The BIOS should have been updated.

**fwupd version information**
[... 642 lines cut ...]
├── initrd.img-5.11.0-22-generic
├── initrd.img-5.11.0-25-generic
├── initrd.img.old -> initrd.img-5.11.0-22-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.11.0-22-generic
├── System.map-5.11.0-25-generic
├── vmlinuz -> vmlinuz-5.11.0-25-generic
├── vmlinuz-5.11.0-22-generic
├── vmlinuz-5.11.0-25-generic
└── vmlinuz.old -> vmlinuz-5.11.0-22-generic

5 directories, 297 files
```

**Additional questions**

- Operating system and version: Ubuntu 21.04
- Have you tried rebooting? Yes
- Is this a regression? I don't know
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Yes, no

superm1, 2021-07-28:
Can you please try with secure boot disabled?  See https://bugs.launchpad.net/ubuntu/+source/shim/+bug/1929471 for more context

hughsie, 2021-08-03:
Also, I think you might have more success with 1.6.2 - we uncovered a problem with the UX capsule and Lenovo firmware.

AidanGG, 2021-08-04:
I was able to perform the EC and System upgrades with fwupd 1.5.12. However, I have not been able to have 1.6.2 launch `fwupd.service` at all (Arch Linux testing repo). There is some error message when using `linux-hardened`:

```
FuEngine     failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to open /dev/mem: No such file or directory
```

where as on the `linux` kernel, I get a similar message but with `failed to mmap SPIBAR`. In tried with secure boot on and off without any luck. Under `lspci`, that device corresponds to

```
00:1f.5 Serial bus controller [0c80]: Intel Corporation Tiger Lake-LP SPI Controller (rev 20)
```

hughsie, 2021-08-04:
> failed to open /dev/mem

that should be harmless, right?

> failed to mmap SPIBAR

Also harmless. Can you post the entire daemon log please?

AidanGG, 2021-08-04:
Here's a link to the gist: https://gist.github.com/AidanGG/4d0c87b4a67e4f2cc55a0c94c503a4e6

hughsie, 2021-08-05:
> Aug 05 07:00:18 graphite systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV

A segfault! It's been a long time since one of those was reported!

Can you do this please:

    $ sudo gdb --args /usr/libexec/fwupd/fwupd --verbose
    (gdb) r

then wait for it to segfault, then:

    (gdb) bt

and then pastebin/gist the entire backtrace please.

AidanGG, 2021-08-05:
Hopefully this is what you want:

```
(gdb) bt
#0  0x0000000000000000 in  ()
#1  0x00007ffff41c408c in  () at /usr/lib/fwupd-plugins-3/libfu_plugin_elantp.so
#2  0x00007ffff7f1ea51 in fu_device_setup () at /usr/lib/libfwupdplugin.so.2
#3  0x00007ffff7f1eba8 in  () at /usr/lib/libfwupdplugin.so.2
#4  0x00007ffff7f16082 in fu_device_locker_new_full () at /usr/lib/libfwupdplugin.so.2
#5  0x00007ffff7f2c524 in fu_plugin_runner_backend_device_added () at /usr/lib/libfwupdplugin.so.2
#6  0x0000000100019bd2 in  ()
#7  0x00007ffff7c14d8f in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#8  0x00007ffff7c30588 in  () at /usr/lib/libgobject-2.0.so.0
#9  0x00007ffff7c31c49 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#10 0x00007ffff7c321a0 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#11 0x0000000100031c97 in  ()
#12 0x00000001000320eb in  ()
#13 0x000000010002964f in fu_engine_load ()
#14 0x000000010001031c in main ()
```

hughsie, 2021-08-05:
Interesting! Can you install the fwupd debug symbols for your distro and do the gdb again please? I'll start looking for a potential problem in the meantime. Thanks!

AidanGG, 2021-08-05:
Here you go:

```
(gdb) bt
#0  0x0000000000000000 in  ()
#1  0x00007ffff555e08c in fu_elantp_hid_device_setup (device=0x102582e00, error=0x7fffffffdc50) at ../fwupd-1.6.2/plugins/elantp/fu-elantp-hid-device.c:165
#2  0x00007ffff7f1ea51 in fu_device_setup (self=self@entry=0x102582e00, error=error@entry=0x7fffffffdc50) at ../fwupd-1.6.2/libfwupdplugin/fu-device.c:4201
#3  0x00007ffff7f1eba8 in fu_device_open_internal (self=0x102582e00, error=0x7fffffffdc50) at ../fwupd-1.6.2/libfwupdplugin/fu-device.c:3939
#4  0x00007ffff7f16082 in fu_device_locker_new_full (device=0x102582e00, open_func=0x7ffff7f1ec10 <fu_device_open>, close_func=0x7ffff7f1dc90 <fu_device_close>, error=0x7fffffffdc50) at ../fwupd-1.6.2/libfwupdplugin/fu-device-locker.c:203
#5  0x00007ffff7f2c524 in fu_plugin_backend_device_added (error=0x7fffffffdc50, device=0x102538f20, self=0x10222b7a0) at ../fwupd-1.6.2/libfwupdplugin/fu-plugin.c:1480
#6  fu_plugin_runner_backend_device_added (self=self@entry=0x10222b7a0, device=device@entry=0x102538f20, error=error@entry=0x7fffffffdc50) at ../fwupd-1.6.2/libfwupdplugin/fu-plugin.c:1524
#7  0x0000000100019bd2 in fu_engine_backend_device_added_cb (backend=<optimized out>, device=0x102538f20, self=0x10009a820) at ../fwupd-1.6.2/src/fu-engine.c:6254
#8  0x00007ffff7c14d8f in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#9  0x00007ffff7c30588 in  () at /usr/lib/libgobject-2.0.so.0
#10 0x00007ffff7c31c49 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#11 0x00007ffff7c321a0 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#12 0x0000000100031c97 in fu_udev_backend_device_add (self=0x1000a2020, udev_device=<optimized out>) at ../fwupd-1.6.2/src/fu-udev-backend.c:31
#13 0x00000001000320eb in fu_udev_backend_coldplug (backend=<optimized out>, error=<optimized out>) at ../fwupd-1.6.2/src/fu-udev-backend.c:158
#14 0x000000010002964f in fu_engine_load (self=<optimized out>, flags=flags@entry=14, error=error@entry=0x7fffffffe238) at ../fwupd-1.6.2/src/fu-engine.c:6747
#15 0x000000010001031c in main (argc=<optimized out>, argv=<optimized out>) at ../fwupd-1.6.2/src/fu-main.c:1971
```

(For future reference to anyone that needs to do this on Arch, add `options=(debug !strip)` to the PKGBUILD and `--buildtype debugoptimized` to the call to `arch-meson`)

closed 2021-08-05 by hughsie