fwupd/fwupd #3554: No Prometheus IOTA Config on HP laptop with version 1.5.11

andch-nn, 2021-07-28:
**Describe the bug**
Use fwupdmgr get-devices to list all devices on HP laptop, Promethues IOTA config will miss in 1.5.11-0ubuntu1~21.04.1.

**Steps to Reproduce**
1. Enable proposed package in Ubuntu Hirsute
2. sudo apt install fwupd
3. fwupdmgr get-devices

**Expected behavior**
├─Prometheus:
│ │ Device ID: 0fef0a0c55f6442bffaebd774ae771341c89571b
│ │ Summary: Fingerprint reader
│ │ Current version: 10.01.3273255
│ │ Vendor: Synaptics (USB:0x06CB)
│ │ Install Duration: 2 seconds
│ │ GUIDs: 816af83b-646c-5033-ae19-26391eeb675d
│ │ 8b06355d-849b-5125-85ec-6d02a20cf1ac
│ │ Device Flags: • Updatable
│ │ • Supported on remote server
│ │ • Cryptographic hash verification is available
│ │
│ └─Prometheus IOTA Config:
│ Device ID: 40b77e58384a02768e3ce90ed74c93ca2311327d
│ Current version: 0021
│ Minimum Version: 0021
│ Vendor: Synaptics (USB:0x06CB)
│ GUIDs: 0e58a9cb-f338-567b-a7ca-2a821eb8f3c2
│ 7b93dd19-18d1-5a0c-a32d-79a5d27e2b38
│ Device Flags: • Updatable
│ • Supported on remote server

**fwupd version information**
1.5.11.

```shell
fwupdmgr --version
```
client version: 1.5.11
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.11
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
HP ZBook Fury 15 G7 Mobile Workstation
│
├─Thunderbolt host controller:
│     Device ID:          22ec5d304a254d9e72059a7661ad2e09dbb49d2f
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    62.00
│     Vendor:             HP Inc. (THUNDERBOLT:0x00F0, TBT:0x00F0)
│     GUIDs:              e0eb3006-c0f8-5868-8e2f-547c2c9e9904
│                         012e4cf2-92de-5067-96b2-7269ba095b54
│                         2a296088-9b04-556c-bf2c-5a47d66f5760
│                         59d64b52-7286-56f7-a15a-f2504c6e6cd6
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─ELAN0720:00 04F3:313A:
│     Device ID:          37127274da0461232269feecebfc638f30488887
│     Summary:            Elan Touchpad
│     Current version:    0x000c
[... 143 lines cut ...]
│                         3777783a-3f83-56a5-95f4-533eb6a2bd19
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─WDC PC SN730 SDBPNTY-512G-1006:
      Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
      Summary:            NVM Express Solid State Drive
      Current version:    HPS2
      Vendor:             Sandisk Corp (NVME:0x15B7)
      GUIDs:              7b974856-317d-5538-a915-de0a6353750f
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0
                          a39943dd-3afb-54f8-b110-c5a21f071200
                          fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                          430b96d5-8289-5250-a02c-2f0ec84e7f09
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
**Additional questions**

- Operating system and version: Ubuntu Hirsute
- Have you tried rebooting? Yes
- Is this a regression?

superm1, 2021-07-28:
Can you please share verbose daemon log with 1.5.11?

andch-nn, 2021-07-28:
Hi @superm1 ,
The attached file is daemon log.
[fwupd-202007282139.log](https://github.com/fwupd/fwupd/files/6893515/fwupd-202007282139.log)

superm1, 2021-07-28:
```
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]: 13:13:39:0671 FuDevice             cannot change (null) physical ID from (null) to usb:01:00:08 as FuDevice->setup() has already completed
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]: 13:13:39:0671 FuDevice             failed to ensure child: cannot ensure ID: FuSynapromConfig:
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]: Prometheus IOTA Config
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   ParentDeviceId:       0fef0a0c55f6442bffaebd774ae771341c89571b
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Guid:                 0e58a9cb-f338-567b-a7ca-2a821eb8f3c2 <- USB\VID_06CB&PID_00DF-cfg
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Guid:                 7b93dd19-18d1-5a0c-a32d-79a5d27e2b38 <- USB\VID_06CB&PID_00DF&CFG1_3483&CFG2_500
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Protocol:             com.synaptics.prometheus.config
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Flags:                updatable
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Vendor:               Synaptics
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   VendorId:             USB:0x06CB
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Version:              0026
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   VersionLowest:        0026
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   VersionFormat:        plain
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Icon:                 touchpad-disabled
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   LogicalId:            cfg
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]: 13:13:39:0673 FuPlugin             ignoring add: cannot ensure ID: FuSynapromConfig:
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]: Prometheus IOTA Config
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   ParentDeviceId:       0fef0a0c55f6442bffaebd774ae771341c89571b
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Guid:                 0e58a9cb-f338-567b-a7ca-2a821eb8f3c2 <- USB\VID_06CB&PID_00DF-cfg
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Guid:                 7b93dd19-18d1-5a0c-a32d-79a5d27e2b38 <- USB\VID_06CB&PID_00DF&CFG1_3483&CFG2_500
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Protocol:             com.synaptics.prometheus.config
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Flags:                updatable
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Vendor:               Synaptics
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   VendorId:             USB:0x06CB
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Version:              0026
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   VersionLowest:        0026
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   VersionFormat:        plain
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   Icon:                 touchpad-disabled
 七  28 21:13:39 ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation fwupd[6113]:   LogicalId:            cfg
```

I feel like we saw this on master as well, @hughsie do you remember?  Probably something we need  to backport to 1.5.x.

superm1, 2021-07-28:
Can you please try applying https://github.com/fwupd/fwupd/commit/832231da9c1c3cc8a40ea9145a1925f85f237a98 to your build?

andch-nn, 2021-07-29:
Hi @superm1 ,
After I applied this change on Ubuntu, IOTA config pops up in the list.
Thanks.

andch-nn, 2021-07-30:
Hi,
I tried to use `fwupdmgr reinstall [Device id]`, and I got failure. Is it expected?
```
ubuntu@ubuntu-HP-ZBook-Fury-15-G7-Mobile-Workstation:~$ fwupdmgr reinstall 40b77e58384a02768e3ce90ed74c93ca2311327d
Prometheus IOTA Config and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
==== AUTHENTICATING FOR org.freedesktop.fwupd.update-hotplug-trusted ===
Authentication is required to update the firmware on a removable device
Authenticating as: ubuntu,,, (ubuntu)
Password: 
==== AUTHENTICATION COMPLETE ===
Authenticating…          [***************************************]
Installing on Prometheus…[***************************************]
Installing on Prometheus IOTA Config…****************************]
Installing on Prometheus…[***                                    ]
Restarting device…       [***************************************]
error status: 0x62a
```
[fwupd-2021-07301927.log](https://github.com/fwupd/fwupd/files/6907193/fwupd-2021-07301927.log)

hughsie, 2021-07-30:
I don't think you can reinstall the config on the Prometheus device; it's strictly upgrade only.

superm1, 2021-07-30:
> I don't think you can reinstall the config on the Prometheus device; it's strictly upgrade only.

Really?  Maybe we need to prevent that then.

hughsie, 2021-07-30:
It's like the dbx that can only ever be updated. Maybe something like FWUPD_DEVICE_FLAG_ONLY_UPGRADE?

superm1, 2021-07-30:
Can it downgrade to an old config though?  It just can't reinstall?

hughsie, 2021-07-30:
> Can it downgrade to an old config though

Nope, just update. It's a PITA to test as you only get "one try".

superm1, 2021-07-30:
>Nope, just update. It's a PITA to test as you only get "one try".

Well that's too bad.  But if you get it right you only needed one try ;)

>It's like the dbx that can only ever be updated. Maybe something like FWUPD_DEVICE_FLAG_ONLY_UPGRADE?

I don't know how important an extra flag is for this - UX will never show reinstall and downgrade in a list of automatic actions.
Maybe just the plugin's write vfunc looks at `FWUPD_INSTALL_FLAG_ALLOW_REINSTALL` and `FWUPD_INSTALL_FLAG_ALLOW_DOWNGRADE` and returns an error not supported.

superm1, 2021-07-30:
>Maybe just the plugin's write vfunc looks at FWUPD_INSTALL_FLAG_ALLOW_REINSTALL and FWUPD_INSTALL_FLAG_ALLOW_DOWNGRADE and returns an error not supported.

Oh, but I looked now and the plugin doesn't parse the version.  So yeah this probably does need a daemon flag.

hughsie, 2021-07-31:
> Oh, but I looked now and the plugin doesn't parse the version

This is kinda deliberate, I don't really want plugins doing version comparison logic at the point where they're supposed to be writing.

I've added a flag here: https://github.com/fwupd/fwupd/pull/3580

andch-nn, 2021-08-03:
I think this issue can be closed. If it's ok, I'll close it.

closed 2021-08-03 by hughsie