fwupd/fwupd #3563: Intel Management Engine fails on multiple ThinkPads

MountainX, 2021-07-28:
**Describe the bug**

I am unable to update Intel Management Engine on mutliple ThinkPad X270 devices as well as on the current T470s I'm working on.

**Steps to Reproduce**

On the ThinkPad T470s I run:

    # fwupdmgr update
    Devices with the latest available firmware version:
    • Embedded Controller
    ╔══════════════════════════════════════════════════════════════════════════════╗
    ║ Upgrade Intel Management Engine from 184.77.3664 to 184.83.3874?             ║
    ╠══════════════════════════════════════════════════════════════════════════════╣
    ║ Version 11.8.83.3874 (LVFS: 184.83.3874)                                     ║
    ║ Problem Fixes                                                                ║
    ║ • Mitigated the following security vulnerabilities                           ║
    ║ 20HGS6R400 must remain plugged into a power source for the duration of the   ║
    ║ update to avoid damage.                                                      ║
    ╚══════════════════════════════════════════════════════════════════════════════╝

    Perform operation? [Y|n]: y
    Downloading…             [***************************************]
    Decompressing…           [***************************************]
    Authenticating…          [***************************************]
    Authenticating…          [***************************************]
    Updating Intel Management Engine…********************************]
    Scheduling…              [***************************************]
    Successfully installed firmware
    • SAMSUNG MZVLB512HAJQ-000L7
    ╔══════════════════════════════════════════════════════════════════════════════╗
    ║ Upgrade System Firmware from 0.1.38 to 0.1.41?                               ║
    ╠══════════════════════════════════════════════════════════════════════════════╣
    ║ Some new functionality has also been added:                                  ║
    ║ *Updated YD1 skylake MCU to MC0406E3_000000ED_000000EA and KBL MCU to        ║
    ║ MC0806E9_000000E9_000000EA                                                   ║
    ║ *Version 04.17.000 code of FIT's InROM diagnostics.                          ║
    ║ 20HGS6R400 must remain plugged into a power source for the duration of the   ║
    ║ update to avoid damage.                                                      ║
    ╚══════════════════════════════════════════════════════════════════════════════╝

    Perform operation? [Y|n]: y
    Downloading…             [***************************************]
    Decompressing…           [***************************************]
    Authenticating…          [***************************************]
    Authenticating…          [***************************************]
    Updating System Firmware…[***************************************]
    Scheduling…              [***************************************]
    Successfully installed firmware

    An update requires a reboot to complete. Restart now? [y|N]: y

Rebooting results in the system failing to boot. After the Lenovo splash screen a text word briefly appears in the upper left of the screen, the screen goes black and in a few minutes the system fan comes on as the system heats up.

No progress is made after several hours. (I let an X270 with the same issue run for 3 days without any success.) The power button will not turn the device off. The only option is to use the system reset hole on the T470s. Fortunately, the system is not bricked. But the firmware is not installed. I have repeated these steps twice. I don't wish to do it again because my system may not recover next time (?).

**Expected behavior**

The system should reboot and the firmware should be installed.

**fwupd version information**

I'm running a fully updated Arch Linux install. I ran `pacman -Syu` immediately before running `fwupdmgr update`

    # fwupdmgr --version
    client version: 1.6.1
    compile-time dependency versions
            gusb:   0.3.7
    daemon version: 1.6.1

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

pacman

**fwupd device information**

I can share that info privately.

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? Not sure
- Are you using an NVMe disk? Yes, but I don't boot from it.
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Boot order lock is disabled.

MountainX, 2021-07-29:
I found a solution on the T470s. I changed the firmware settings from allowing "UEFI only" boot to "both" (i.e., legacy and UEFI). To do this I had to disable secure boot too.

superm1, 2021-07-29:
Good sleuth work. To me this sounds like a bug with the firmware, the update should be working in UEFI secure boot mode (csm disabled).

smrqdt, 2021-07-29:
I have exactly the same problem with [the 184.86.3909 ME update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24RN.firmware) on my T480 since the update was released. Tried again today, I also tried disabling Secure Boot (I use a self-signed EFI binary), black screen before the „Load Capsule“ screen with the Lenovo logo even appears. Only way to get out is a complete power reset using the secret battery interrupt button hole (that’s a bit scary in a firmware update process).

If it’s helpful I can fill out the bug template too as a comment or open an additional bug. Also on Arch Linux, same fwupd version.